ned-kelly/docker-voltronic-homeassistant #3: Errors being logged in Home Assistant

md7automation, 2019-07-28:
ERROR (MainThread) [homeassistant.components.mqtt.sensor] Exception in async_discover_sensor when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'voltronic_Load_watt', 'unit_of_measurement': 'W', 'state_topic': 'homeassistant/sensor/voltronic_Load_watt', 'icon': 'mdi:chart-bell-curve', 'retain': True, 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/src/app/homeassistant/components/mqtt/sensor.py", line 61, in async_discover_sensor
    config = PLATFORM_SCHEMA(discovery_payload)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 267, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 589, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 427, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: extra keys not allowed @ data['retain']

Please could you update your code or advise ?

hestnes88, 2019-08-09:
Hello, I have the same problem as decribed by md7automation

Log Details (ERROR)
Fri Aug 09 2019 23:12:29 GMT+0200 (sentraleuropeisk sommertid)
Exception in async_discover_sensor when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'voltronic_Inverter_mode', 'unit_of_measurement': '', 'state_topic': 'homeassistant/sensor/voltronic_Inverter_mode', 'icon': 'mdi:solar-power', 'retain': True, 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 81, in async_discover_sensor
    config = PLATFORM_SCHEMA(discovery_payload)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 267, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 589, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 427, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: extra keys not allowed @ data['retain']

And 

Log Details (ERROR)
Fri Aug 09 2019 23:12:30 GMT+0200 (sentraleuropeisk sommertid)
Exception in async_discover_sensor when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'name': 'voltronic', 'state_topic': 'homeassistant/sensor/voltronic', 'retain': True, 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 81, in async_discover_sensor
    config = PLATFORM_SCHEMA(discovery_payload)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 267, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 589, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 427, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: extra keys not allowed @ data['retain']

And so on, and this is what it puts to the mqtt topic

`ubuntu@ubuntu:~$ mosquitto_sub -h 192.168.10.156 -v -t "homeassistant/#"
homeassistant/sensor/voltronic_Inverter_mode/config {
            "name": "voltronic_Inverter_mode",
            "unit_of_measurement": "",
            "state_topic": "homeassistant/sensor/voltronic_Inverter_mode",
            "icon": "mdi:solar-power",
            "retain": true
        }
homeassistant/sensor/voltronic_AC_grid_voltage/config {
            "name": "voltronic_AC_grid_voltage",
            "unit_of_measurement": "V",
            "state_topic": "homeassistant/sensor/voltronic_AC_grid_voltage",
            "icon": "mdi:power-plug",
            "retain": true
        }
homeassistant/sensor/voltronic_AC_grid_frequency/config {
            "name": "voltronic_AC_grid_frequency",
            "unit_of_measurement": "Hz",
            "state_topic": "homeassistant/sensor/voltronic_AC_grid_frequency",
            "icon": "mdi:current-ac",
            "retain": true
        }
homeassistant/sensor/voltronic_AC_out_voltage/config {
            "name": "voltronic_AC_out_voltage",
            "unit_of_measurement": "V",
            "state_topic": "homeassistant/sensor/voltronic_AC_out_voltage",
            "icon": "mdi:power-plug",
            "retain": true
        }
homeassistant/sensor/voltronic_AC_out_frequency/config {
            "name": "voltronic_AC_out_frequency",
            "unit_of_measurement": "Hz",
            "state_topic": "homeassistant/sensor/voltronic_AC_out_frequency",
            "icon": "mdi:current-ac",
            "retain": true
        }
[... 169 lines cut ...]
            "name": "voltronic_Out_source_priority",
            "unit_of_measurement": "",
            "state_topic": "homeassistant/sensor/voltronic_Out_source_priority",
            "icon": "mdi:grid",
            "retain": true
        }
homeassistant/sensor/voltronic_Charger_source_priority/config {
            "name": "voltronic_Charger_source_priority",
            "unit_of_measurement": "",
            "state_topic": "homeassistant/sensor/voltronic_Charger_source_priority",
            "icon": "mdi:solar-power",
            "retain": true
        }
homeassistant/sensor/voltronic_Battery_redischarge_voltage/config {
            "name": "voltronic_Battery_redischarge_voltage",
            "unit_of_measurement": "V",
            "state_topic": "homeassistant/sensor/voltronic_Battery_redischarge_voltage",
            "icon": "mdi:battery-negative",
            "retain": true
        }
homeassistant/sensor/voltronic/config {
            "name": "voltronic",
            "state_topic": "homeassistant/sensor/voltronic",
            "retain": true
`

rddkrn, 2019-08-13:
Hello,
I have the same error. My sensor show nothing but "-". 
Thank you for your hard work.

ned-kelly, 2019-08-27:
Hi guys,

Can you try and run `inverter_poller -d -1` directly, and see what it returns?

ned-kelly, 2019-09-14:
Hi @manju850,

The project as it currently stands is not setup for Hass.io, the idea here is that you manually clone/run the project on a SBC of your choice (Raspberry/Orange Pi etc) and then integrate the MQTT data into your HA deployment using Sensors etc... 

If you want to make the relevant changes and submit a PR for native Hass.io integration please do! - I'm just running 'bare-bones' HomeAssistant in Docker without Hass.io so am unable to test/support this for now, sorry.

ned-kelly, 2019-09-15:
Hi @md7automation,

Forgot to mention that the `data['retain']` key that was sent with the MQTT data has now been removed (with the most recent update of the code) which will fix this issue - It was a recent feature added into HA.

Cheers,

closed 2019-09-15 by ned-kelly