ned-kelly/docker-voltronic-homeassistant #4: no response from inverter_poller cli command

nabeel-ahmad, 2019-08-01:
I was able to successfully run the docker project on raspberry pi 3 A+ after latest fixes.
I am even able to execute the raw commands and getting ACK from the inverter and the mode also changing successfully..

```
pi@raspberrypi:~ $ sudo docker container exec voltronic-mqtt timeout 10 /opt/inverter-cli/bin/inverter_poller -d -r POP01
Thu Aug  1 17:37:15 2019 INVERTER: Debug set
Thu Aug  1 17:37:16 2019 INVERTER: Current CRC: D2 69
Thu Aug  1 17:37:16 2019 INVERTER: POP01 reply size (7 bytes)
Thu Aug  1 17:37:16 2019 INVERTER: POP01: 7 bytes read: (ACK
Thu Aug  1 17:37:16 2019 INVERTER: POP01 query finished
Reply:  ACK
```

But the below command is not giving any output..

```
sudo docker container exec voltronic-mqtt timeout 10 /opt/inverter-cli/bin/inverter_poller -1
```

Also on MQTT lens I can see messages being published every 5 minutes with /config topic but I don't see the inverter state messages which are supposed to come every 30 sec

Am I missing something?

rddkrn, 2019-08-17:
I have the same problem. My home assistant got no data from my pi zero. I am stuck here and don't know what to do. Any suggestion will very appreciate. Thanks.

ned-kelly, 2019-08-27:
Can you try running in "single run, debug mode" and see what's returned:

`inverter_poller -d -1`

plozachm, 2019-08-31:
Hi @ned-kelly,
I have the same issue as @nabeel-ahmad , when I run the poller with debug option:
`i@raspberrypi:~ $ docker exec -it voltronic-mqtt /bin/bash
root@raspberrypi:/opt# /opt/inverter-cli/bin/inverter_poller -d -1
Sat Aug 31 14:27:23 2019 INVERTER: Debug set
Sat Aug 31 14:27:23 2019 INVERTER: Current CRC: 49 C1
Sat Aug 31 14:27:24 2019 INVERTER: QMOD reply size (5 bytes)
Sat Aug 31 14:27:24 2019 INVERTER: QMOD: 5 bytes read: (B
Sat Aug 31 14:27:24 2019 INVERTER: QMOD query finished
Sat Aug 31 14:27:24 2019 INVERTER: Current CRC: B7 A9
Sat Aug 31 14:27:25 2019 INVERTER: QPIGS reply size (110 bytes)
Sat Aug 31 14:27:25 2019 INVERTER: QPIGS: 110 bytes read: (000.0 00.0 230.1 50.0 0000 0000 000 369 26.88 004 100 0032 0004 079.5 26.97 00000 00110110 00 00 00123 110
Sat Aug 31 14:27:25 2019 INVERTER: QPIGS query finished
Sat Aug 31 14:27:25 2019 INVERTER: Current CRC: F8 54
Sat Aug 31 14:27:25 2019 INVERTER: QPIRI reply size (97 bytes)
Sat Aug 31 14:27:25 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 13.0 230.0 50.0 13.0 3000 3000 24.0 23.0 21.0 30.4 27.0 2 60 120 0 1 2 1 01 0 0 00.0 0 1� 00123 110
Sat Aug 31 14:27:25 2019 INVERTER: Current CRC: B4 DA
Sat Aug 31 14:27:25 2019 INVERTER: QPIWS reply size (36 bytes)
Sat Aug 31 14:27:25 2019 INVERTER: QPIWS: 36 bytes read: (00000100000000000000000000000000
Sat Aug 31 14:27:25 2019 INVERTER: QPIWS query finished
Sat Aug 31 14:27:30 2019 INVERTER: Current CRC: F8 54
Sat Aug 31 14:27:31 2019 INVERTER: QPIRI reply size (97 bytes)
Sat Aug 31 14:27:31 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 13.0 230.0 50.0 13.0 3000 3000 24.0 23.0 21.0 30.4 27.0 2 60 120 0 1 2 1 01 0 0 00.0 0 1� 00123 110
Sat Aug 31 14:27:36 2019 INVERTER: Current CRC: F8 54
Sat Aug 31 14:27:36 2019 INVERTER: QPIRI reply size (97 bytes)
Sat Aug 31 14:27:36 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 13.0 230.0 50.0 13.0 3000 3000 24.0 23.0 21.0 30.4 27.0 2 60 120 0 1 2 1 01 0 0 00.0 0 1� 00123 110
Sat Aug 31 14:27:42 2019 INVERTER: Current CRC: F8 54
Sat Aug 31 14:27:42 2019 INVERTER: QPIRI reply size (97 bytes)
Sat Aug 31 14:27:42 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 13.0 230.0 50.0 13.0 3000 3000 24.0 23.0 21.0 30.4 27.0 2 60 120 0 1 2 1 01 0 0 00.0 0 1� 00123 110
Sat Aug 31 14:27:47 2019 INVERTER: Current CRC: F8 54
Sat Aug 31 14:27:48 2019 INVERTER: QPIRI reply size (97 bytes)
Sat Aug 31 14:27:48 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 13.0 230.0 50.0 13.0 3000 3000 24.0 23.0 21.0 30.4 27.0 2 60 120 0 1 2 1 01 0 0 00.0 0 1� 00123 110
Sat Aug 31 14:27:53 2019 INVERTER: Current CRC: F8 54
Sat Aug 31 14:27:56 2019 INVERTER: QPIRI read timeout
Sat Aug 31 14:27:56 2019 INVERTER: QPIRI reply too short (32 bytes)
Sat Aug 31 14:28:01 2019 INVERTER: Current CRC: F8 54
Sat Aug 31 14:28:04 2019 INVERTER: QPIRI read timeout
Sat Aug 31 14:28:04 2019 INVERTER: QPIRI reply too short (8 bytes)
Sat Aug 31 14:28:09 2019 INVERTER: Current CRC: F8 54
Sat Aug 31 14:28:09 2019 INVERTER: QPIRI reply size (97 bytes)
Sat Aug 31 14:28:09 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 13.0 230.0 50.0 13.0 3000 3000 24.0 23.0 21.0 30.4 27.0 2 60 120 0 1 2 1 01 0 0 00.0 0 1� 00123 110
Sat Aug 31 14:28:14 2019 INVERTER: Current CRC: F8 54
Sat Aug 31 14:28:15 2019 INVERTER: QPIRI reply size (97 bytes)
Sat Aug 31 14:28:15 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 13.0 230.0 50.0 13.0 3000 3000 24.0 23.0 21.0 30.4 27.0 2 60 120 0 1 2 1 01 0 0 00.0 0 1� 00123 110
Sat Aug 31 14:28:20 2019 INVERTER: Current CRC: F8 54
Sat Aug 31 14:28:20 2019 INVERTER: QPIRI reply size (97 bytes)
Sat Aug 31 14:28:20 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 13.0 230.0 50.0 13.0 3000 3000 24.0 23.0 21.0 30.4 27.0 2 60 120 0 1 2 1 01 0 0 00.0 0 1� 00123 110
^C
`

ned-kelly, 2019-09-01:
Looking at this, it's not a connectivity issue as you're getting a correct response from the inverter, so my thoughts ....

 - The string of numbers you see in your responses (such as `QPIGS ` etc) is actually the response from the inverter, so communication between your pi and the inverter is working for most of the commands
- I'm wondering if your firmware version on the inverter is different, as your reply from the `QPIRI ` command is actually different than mine (I'm in AU so we're running 220-250v here)...
- You could try and adapt the [`inverter.cpp`](https://github.com/ned-kelly/docker-voltronic-homeassistant/blob/master/sources/inverter-cli/inverter.cpp#L178-L185) QPIRI query to handle your different response
- The other option (If you're a bit shabby with C is to just comment out the `qpiri*` queries in the app and re-compile it --  You will lose the _"Device Rating Information inquiry"_
- If you check [page 3](http://forums.aeva.asn.au/uploads/293/HS_MS_MSX_RS232_Protocol_20140822_after_current_upgrade.pdf) of the Protocol Manual, you'll see that the response for `W` should only be `0` or `1` so I'm wondering if upgrading the firmware will help remove the additional character you're seeing in the response.

What version of the `Main CPU` and `SCC` firmware are you running?

ned-kelly, 2019-09-01:
Hi @nabeel-ahmad  & @rddkrn - I have a feeling you may also have the same issue, could you please check my above post and confirm if you also have the same problem?

Thanks

rddkrn, 2019-09-01:
> Hi @nabeel-ahmad & @rddkrn - I have a feeling you may also have the same issue, could you please check my above post and confirm if you also have the same problem?
> 
> Thanks

Helo,
This is my result
root@647f3ae930eb:/opt# /opt/inverter-cli/bin/inverter_poller -d -1
INVERTER: Debug set
Sun Sep  1 05:18:33 2019 INVERTER: Current CRC: 49 C1
Sun Sep  1 05:18:33 2019 INVERTER: QMOD reply size (5 bytes)
Sun Sep  1 05:18:33 2019 INVERTER: QMOD: 5 bytes read: (L
Sun Sep  1 05:18:33 2019 INVERTER: QMOD query finished
Sun Sep  1 05:18:33 2019 INVERTER: Current CRC: B7 A9
Sun Sep  1 05:18:34 2019 INVERTER: QPIGS reply size (110 bytes)
Sun Sep  1 05:18:34 2019 INVERTER: QPIGS: 110 bytes read: (230.9 50.0 230.9 50.0 1824 1773 036 383 00.20 000 000 0049 0000 324.7 00.00 00000 00010110 00 00 01791 010
Sun Sep  1 05:18:34 2019 INVERTER: QPIGS query finished
Sun Sep  1 05:18:34 2019 INVERTER: Current CRC: F8 54
Sun Sep  1 05:18:34 2019 INVERTER: QPIRI reply size (98 bytes)
Sun Sep  1 05:18:34 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
Sun Sep  1 05:18:34 2019 INVERTER: Current CRC: B4 DA
Sun Sep  1 05:18:34 2019 INVERTER: QPIWS reply size (36 bytes)
Sun Sep  1 05:18:34 2019 INVERTER: QPIWS: 36 bytes read: (00000000000000000000001000000000
Sun Sep  1 05:18:34 2019 INVERTER: QPIWS query finished
Sun Sep  1 05:18:39 2019 INVERTER: Current CRC: F8 54
Sun Sep  1 05:18:40 2019 INVERTER: QPIRI reply size (98 bytes)
Sun Sep  1 05:18:40 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
Sun Sep  1 05:18:45 2019 INVERTER: Current CRC: F8 54
Sun Sep  1 05:18:48 2019 INVERTER: QPIRI read timeout
Sun Sep  1 05:18:48 2019 INVERTER: QPIRI reply too short (40 bytes)
Sun Sep  1 05:18:53 2019 INVERTER: Current CRC: F8 54
Sun Sep  1 05:18:56 2019 INVERTER: QPIRI read timeout
Sun Sep  1 05:18:56 2019 INVERTER: QPIRI reply too short (8 bytes)
Sun Sep  1 05:19:01 2019 INVERTER: Current CRC: F8 54
Sun Sep  1 05:19:01 2019 INVERTER: QPIRI reply size (98 bytes)
Sun Sep  1 05:19:01 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
Sun Sep  1 05:19:06 2019 INVERTER: Current CRC: F8 54
Sun Sep  1 05:19:07 2019 INVERTER: QPIRI reply size (98 bytes)
Sun Sep  1 05:19:07 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
Sun Sep  1 05:19:12 2019 INVERTER: Current CRC: F8 54
Sun Sep  1 05:19:12 2019 INVERTER: QPIRI reply size (98 bytes)
Sun Sep  1 05:19:12 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
Sun Sep  1 05:19:17 2019 INVERTER: Current CRC: F8 54
Sun Sep  1 05:19:18 2019 INVERTER: QPIRI reply size (98 bytes)
Sun Sep  1 05:19:18 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
Sun Sep  1 05:19:23 2019 INVERTER: Current CRC: F8 54
Sun Sep  1 05:19:23 2019 INVERTER: QPIRI reply size (98 bytes)
Sun Sep  1 05:19:23 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
Sun Sep  1 05:19:28 2019 INVERTER: Current CRC: F8 54
Sun Sep  1 05:19:29 2019 INVERTER: QPIRI reply size (98 bytes)
Sun Sep  1 05:19:29 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
Sun Sep  1 05:19:34 2019 INVERTER: Current CRC: F8 54
Sun Sep  1 05:19:34 2019 INVERTER: QPIRI reply size (98 bytes)
Sun Sep  1 05:19:34 2019 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
^C
I try to adapt the the main.cpp. I got the sensor value in my hassio by change the code in main.cpp as follow:
"if (ups_qmod_changed && ups_qpigs_changed) { //&& ups_qpiri_changed"

This will ignore "ups_qpiri_change" value. I dont no the other effect of this change. 

Hope it will be a better solution.

ned-kelly, 2019-09-01:
> > Hi @nabeel-ahmad & @rddkrn - I have a feeling you may also have the same issue, could you please check my above post and confirm if you also have the same problem?
> > Thanks
> 
> Helo,
> This is my result
> root@647f3ae930eb:/opt# /opt/inverter-cli/bin/inverter_poller -d -1
> INVERTER: Debug set
> Sun Sep 1 05:18:33 2019 INVERTER: Current CRC: 49 C1
> Sun Sep 1 05:18:33 2019 INVERTER: QMOD reply size (5 bytes)
> Sun Sep 1 05:18:33 2019 INVERTER: QMOD: 5 bytes read: (L
> Sun Sep 1 05:18:33 2019 INVERTER: QMOD query finished
> Sun Sep 1 05:18:33 2019 INVERTER: Current CRC: B7 A9
> Sun Sep 1 05:18:34 2019 INVERTER: QPIGS reply size (110 bytes)
> Sun Sep 1 05:18:34 2019 INVERTER: QPIGS: 110 bytes read: (230.9 50.0 230.9 50.0 1824 1773 036 383 00.20 000 000 0049 0000 324.7 00.00 00000 00010110 00 00 01791 010
> Sun Sep 1 05:18:34 2019 INVERTER: QPIGS query finished
> Sun Sep 1 05:18:34 2019 INVERTER: Current CRC: F8 54
> Sun Sep 1 05:18:34 2019 INVERTER: QPIRI reply size (98 bytes)
> Sun Sep 1 05:18:34 2019 INVERTER: QPIRI: incorrect start/stop bytes. Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
> Sun Sep 1 05:18:34 2019 INVERTER: Current CRC: B4 DA
> Sun Sep 1 05:18:34 2019 INVERTER: QPIWS reply size (36 bytes)
> Sun Sep 1 05:18:34 2019 INVERTER: QPIWS: 36 bytes read: (00000000000000000000001000000000
> Sun Sep 1 05:18:34 2019 INVERTER: QPIWS query finished
> Sun Sep 1 05:18:39 2019 INVERTER: Current CRC: F8 54
> Sun Sep 1 05:18:40 2019 INVERTER: QPIRI reply size (98 bytes)
> Sun Sep 1 05:18:40 2019 INVERTER: QPIRI: incorrect start/stop bytes. Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
> Sun Sep 1 05:18:45 2019 INVERTER: Current CRC: F8 54
> Sun Sep 1 05:18:48 2019 INVERTER: QPIRI read timeout
> Sun Sep 1 05:18:48 2019 INVERTER: QPIRI reply too short (40 bytes)
> Sun Sep 1 05:18:53 2019 INVERTER: Current CRC: F8 54
> Sun Sep 1 05:18:56 2019 INVERTER: QPIRI read timeout
> Sun Sep 1 05:18:56 2019 INVERTER: QPIRI reply too short (8 bytes)
> Sun Sep 1 05:19:01 2019 INVERTER: Current CRC: F8 54
> Sun Sep 1 05:19:01 2019 INVERTER: QPIRI reply size (98 bytes)
> Sun Sep 1 05:19:01 2019 INVERTER: QPIRI: incorrect start/stop bytes. Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
> Sun Sep 1 05:19:06 2019 INVERTER: Current CRC: F8 54
> Sun Sep 1 05:19:07 2019 INVERTER: QPIRI reply size (98 bytes)
> Sun Sep 1 05:19:07 2019 INVERTER: QPIRI: incorrect start/stop bytes. Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
> Sun Sep 1 05:19:12 2019 INVERTER: Current CRC: F8 54
> Sun Sep 1 05:19:12 2019 INVERTER: QPIRI reply size (98 bytes)
> Sun Sep 1 05:19:12 2019 INVERTER: QPIRI: incorrect start/stop bytes. Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
> Sun Sep 1 05:19:17 2019 INVERTER: Current CRC: F8 54
> Sun Sep 1 05:19:18 2019 INVERTER: QPIRI reply size (98 bytes)
> Sun Sep 1 05:19:18 2019 INVERTER: QPIRI: incorrect start/stop bytes. Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
> Sun Sep 1 05:19:23 2019 INVERTER: Current CRC: F8 54
> Sun Sep 1 05:19:23 2019 INVERTER: QPIRI reply size (98 bytes)
> Sun Sep 1 05:19:23 2019 INVERTER: QPIRI: incorrect start/stop bytes. Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
> Sun Sep 1 05:19:28 2019 INVERTER: Current CRC: F8 54
> Sun Sep 1 05:19:29 2019 INVERTER: QPIRI reply size (98 bytes)
> Sun Sep 1 05:19:29 2019 INVERTER: QPIRI: incorrect start/stop bytes. Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
> Sun Sep 1 05:19:34 2019 INVERTER: Current CRC: F8 54
> Sun Sep 1 05:19:34 2019 INVERTER: QPIRI reply size (98 bytes)
> Sun Sep 1 05:19:34 2019 INVERTER: QPIRI: incorrect start/stop bytes. Buffer: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 0 2 9 01 0 0 54.0 0 1 0001791 010
> ^C
> I try to adapt the the main.cpp. I got the sensor value in my hassio by change the code in main.cpp as follow:
> "if (ups_qmod_changed && ups_qpigs_changed) { //&& ups_qpiri_changed"
> 
> This will ignore "ups_qpiri_change" value. I dont no the other effect of this change.
> 
> Hope it will be a better solution.
> 
> Many thanks.

Hi @rddkrn - your `QPIRI` command is returning 98 bytes where as my inverter's returning 97 - so I'm guessing also slightly different firmware - Care to post your firmware version so we can compare?

You should be fine to just change the `if (query("QPIRI", 97)) {` to `if (query("QPIRI", 98)) {` in `inverter.cpp` as you don't have any malformed responses like plozachm's experiencing.

Let me know how you go.

Cheers,

rddkrn, 2019-09-03:
Hi @ned-kelly, 

Thank you for your suggestion. My inverter is MPP Solar 5048 MG with firmware version of 70.71.

Waiting for update from your work.

Thank you,

ned-kelly, 2019-09-03:
> Waiting for update from your work.

@rddkrn, could you please make the suggested changes and let us know how you go?

Thanks,

rddkrn, 2019-09-03:
Hi @ned-kelly, I changed the if (query("QPIRI", 97)) { to if (query("QPIRI", 102)) { in inverter.cpp, and I got correct responses from my inverter. I though the problem I have before was the byte size in QPIRI. In my case, 102 is the correct byte size for QPIRI function.

This is the responses I got

root@pizeromppsolar:~# docker exec -it voltronic-mqtt /bin/bash
root@0146102243c4:/opt#  /opt/inverter-cli/bin/inverter_poller -d -1
INVERTER: Debug setTue Sep  3 15:01:25 2019 INVERTER: Current CRC: 49 C1
Tue Sep  3 15:01:26 2019 INVERTER: QMOD reply size (5 bytes)
Tue Sep  3 15:01:26 2019 INVERTER: QMOD: 5 bytes read: (L
Tue Sep  3 15:01:26 2019 INVERTER: QMOD query finished
Tue Sep  3 15:01:26 2019 INVERTER: Current CRC: B7 A9
Tue Sep  3 15:01:26 2019 INVERTER: QPIGS reply size (110 bytes)
Tue Sep  3 15:01:26 2019 INVERTER: QPIGS: 110 bytes read: (219.9 49.9 219.9 49.9 1693 1633 033 306 00.20 000 000 0035 0000 000.0 00.00 00000 00010000 00 00 00000 010
Tue Sep  3 15:01:26 2019 INVERTER: QPIGS query finished
Tue Sep  3 15:01:26 2019 INVERTER: Current CRC: F8 54
Tue Sep  3 15:01:27 2019 INVERTER: QPIRI reply size (102 bytes)
Tue Sep  3 15:01:27 2019 INVERTER: QPIRI: 102 bytes read: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 1 2 9 01 0 0 54.0 0 1 000
Tue Sep  3 15:01:27 2019 INVERTER: QPIRI query finished
Tue Sep  3 15:01:27 2019 INVERTER: Current CRC: B4 DA
Tue Sep  3 15:01:27 2019 INVERTER: QPIWS reply size (36 bytes)
Tue Sep  3 15:01:27 2019 INVERTER: QPIWS: 36 bytes read: (00000000000000000000001000000000
Tue Sep  3 15:01:27 2019 INVERTER: QPIWS query finished
INVERTER: ampfactor from config is 1.00
INVERTER: wattfactor from config is 1.01
{
  "Inverter_mode":3,
  "AC_grid_voltage":219.9,
  "AC_grid_frequency":49.9,
  "AC_out_voltage":219.9,
  "AC_out_frequency":49.9,
  "PV_in_voltage":0.0,
  "PV_in_current":0.0,
  "PV_in_watts":0.0,
  "PV_in_watthour":0.0000,
  "SCC_voltage":0.0000,
  "Load_pct":33,
  "Load_watt":1633,
  "Load_watthour":54.4333,
  "Load_va":1693,
  "Bus_voltage":306,
  "Heatsink_temperature":35,
  "Battery_capacity":0,
  "Battery_voltage":0.20,
  "Battery_charge_current":0,
  "Battery_discharge_current":0,
  "Load_status_on":1,
  "SCC_charge_on":0,
  "AC_charge_on":0,
  "Battery_recharge_voltage":46.0,
  "Battery_under_voltage":42.0,
  "Battery_bulk_voltage":56.4,
  "Battery_float_voltage":54.0,
  "Max_grid_charge_current":30,
  "Max_charge_current":60,
  "Out_source_priority":1,
  "Charger_source_priority":2,
  "Battery_redischarge_voltage":-nan,
  "Warnings":"00000000000000000000001000000000"
}
Tue Sep  3 15:01:27 2019 INVERTER: All queries complete, exiting loop.
root@0146102243c4:/opt#

Thanks,

ned-kelly, 2019-09-03:
> This is the responses I got
> 
> Tue Sep 3 15:01:27 2019 INVERTER: QPIRI: 102 bytes read: (220.0 22.7 220.0 50.0 22.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 1 2 9 01 0 0 54.0 0 1 000
> Tue Sep 3 15:01:27 2019 INVERTER: QPIRI query finished

It would  be good if you can work out what the additional `000` is so we can add it into the program for the 5kw version, and change the byte size - I'm only running a 3kw version so I don't seem to have the that field by the looks - The `000` _last three characters in the response_ isn't documented in the my version of the Protocol Manual (uploaded to this project).

Feel free to submit a PR once you've worked it out - I think we should possibly let users select their inverter size/type (based on the `inverter.conf`) and then we can have different logic for the additional fields and byte sizes...

plozachm, 2019-09-06:
Hi @ned-kelly,
Sorry for the late answer, it's an off-grid setup and I had no access until now. Also, thanks a lot for your feedback, it was very helpful.
I have updated 98 bytes [inverter.cpp#L179](https://github.com/ned-kelly/docker-voltronic-homeassistant/blob/master/sources/inverter-cli/inverter.cpp#L179), rebuilt the image and got this output:
``````

ned-kelly, 2019-09-11:
So i think the answer here is to add an option to set the "byte size" for inverters running different firmware...

This does mean however, that if there are additional fields/values in the string response it will not be parsed by the tool - as we don't know what those field mean unless someone goes and gets an updated protocol manual.

closed 2019-09-11 by ned-kelly

nabeel-ahmad, 2019-09-16:
Thank you guys.. I am also using 5KW inverter and the issue got resolved after changing QPIRI byte size from 97 to 98.

v0d0r, 2020-02-19:
Hi  I seem to have a similar issue. However after changing my QPIRI byte size to 104 and recompiling I still get an error. 
Any suggestions?

./inverter_poller -d -1
Wed Feb 19 18:28:54 2020 INVERTER: Debug set
Wed Feb 19 18:28:54 2020 INVERTER: Current CRC: 49 C1
Wed Feb 19 18:28:54 2020 INVERTER: QMOD reply size (5 bytes)
Wed Feb 19 18:28:54 2020 INVERTER: QMOD: 5 bytes read: (L
Wed Feb 19 18:28:54 2020 INVERTER: QMOD query finished
Wed Feb 19 18:28:54 2020 INVERTER: Current CRC: B7 A9
Wed Feb 19 18:28:55 2020 INVERTER: QPIGS reply size (110 bytes)
Wed Feb 19 18:28:55 2020 INVERTER: QPIGS: 110 bytes read: (225.8 50.0 225.8 50.0 0714 0714 023 408 27.10 000 100 0040 00.0 000.0 00.00 00000 00010101 00 00 00000 110
Wed Feb 19 18:28:55 2020 INVERTER: QPIGS query finished
Wed Feb 19 18:28:55 2020 INVERTER: Current CRC: F8 54
Wed Feb 19 18:28:55 2020 INVERTER: QPIRI reply size (104 bytes)
Wed Feb 19 18:28:55 2020 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 13.0 230.0 50.0 13.0 3000 3000 24.0 23.0 21.0 28.2 27.0 0 30 060 0 0 2 1 01 0 0 27.0 0 1Q�
Wed Feb 19 18:28:55 2020 INVERTER: Current CRC: B4 DA
Wed Feb 19 18:28:55 2020 INVERTER: QPIWS reply size (36 bytes)
Wed Feb 19 18:28:55 2020 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: (100000000000000000000000000000000000 24.0 23.0 21.0 28.2 27.0 0 30 060 0 0 2 1 01 0 0 27.0 0 1Q�
Wed Feb 19 18:29:00 2020 INVERTER: Current CRC: F8 54

Dovespark, 2020-04-25:
Anyone got this running on a MPP solar PIP-5048MK inverter yet?
Ive changed Qpiri and Qpigs to different values mentioned above, with no success yet.

[image]

ned-kelly, 2020-04-26:
@Dovespark - Looks like you're not getting any response back still on some commands hence the NAK responses...

Make sure you’ve:

* Set the Baud Rate to 2400 and RAW if you’re using Serial (try do it directly on the host and not within the container) - I ended up adding this in as a cron-job on boot…

* Looking at the log - Change your QPIRI, and QPIGS based on the 5KW config (see below for working config for 5kw version)

```
qpiri=98
qpiws=36
qmod=5
qpigs=110
```

* Potentially try and reboot the inverter (power it off/on) and see if that clears things up after fixing your config (if the issues happen again – I found that I had to reboot the inverter every 6 months or so before I upgraded my firmware because the serial was a bit buggy)

* Try and upgrade your firmware to the latest versions (if your game) - http://forums.aeva.asn.au/viewtopic.php?t=4332

* Lastly, try and connect directly to the serial port of the inverter manually `screen /dev/ttyUSB0 2400,cs8` (after you've stopped the docker containers and are not using watchpower or any other services that will send traffic down the inverter) and then try and manually run the `QMOD` command, then check how many bytes the response is.

Vandiliz3r, 2020-05-18:
Hi Guys,

I cannot get the values to show in my Home Assistant. Could you point me in the correct direction.

Mon May 18 19:17:41 2020 INVERTER: Current CRC: B7 A9
Mon May 18 19:17:42 2020 INVERTER: QPIGS reply size (110 bytes)
Mon May 18 19:17:42 2020 INVERTER: QPIGS: 110 bytes read: (242.8 49.7 230.1 49.8 0736 0548 014 381 51.50 000 100 0032 0000 000.0 00.00 00000 00010101 00 00 00000 110
Mon May 18 19:17:42 2020 INVERTER: QPIGS query finished
Mon May 18 19:17:42 2020 INVERTER: Current CRC: F8 54
Mon May 18 19:17:42 2020 INVERTER: QPIRI reply size (102 bytes)
(100000000000000(230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 46.0 110s.  Buffer: (100000000000000000000000000000000000�
Mon May 18 19:17:42 2020 INVERTER: Current CRC: B4 DA
Mon May 18 19:17:42 2020 INVERTER: QPIWS reply size (36 bytes)
(100000000000000(230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 46.0 110s.  Buffer: (242.8 49.7 230.1 49.8 0736 0548 0140�
Mon May 18 19:17:47 2020 INVERTER: Current CRC: F8 54
Mon May 18 19:17:48 2020 INVERTER: QPIRI reply size (102 bytes)
Mon May 18 19:17:48 2020 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 46.0 44.0 51.6 51.5 2 30 040 0 1 3 9 01 0 1 49.0 0 1 000 0�0 110
Mon May 18 19:17:48 2020 INVERTER: Current CRC: B4 DA
Mon May 18 19:17:48 2020 INVERTER: QPIWS reply size (36 bytes)
Mon May 18 19:17:48 2020 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: (100000000000000000000000000000000000 48.0 46.0 44.0 51.6 51.5 2 30 040 0 1 3 9 01 0 1 49.0 0 1 000 0�0 110
Mon May 18 19:17:53 2020 INVERTER: Current CRC: F8 54
Mon May 18 19:17:53 2020 INVERTER: QPIRI reply size (102 bytes)
Mon May 18 19:17:53 2020 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: 756 0565 015 378 51.50 000 100 0032 0000 000.0 00.00 000(230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 46.0 110
Mon May 18 19:17:53 2020 INVERTER: Current CRC: B4 DA
Mon May 18 19:17:53 2020 INVERTER: QPIWS reply size (36 bytes)
Mon May 18 19:17:53 2020 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: (100000000000000000000000000000000000000 000.0 00.00 000(230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 46.0 110
Mon May 18 19:17:58 2020 INVERTER: Current CRC: F8 54
Mon May 18 19:18:00 2020 INVERTER: QPIRI reply size (102 bytes)
Mon May 18 19:18:00 2020 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (L
Mon May 18 19:18:00 2020 INVERTER: Current CRC: B4 DA
Mon May 18 19:18:00 2020 INVERTER: QPIWS reply size (36 bytes)
Mon May 18 19:18:00 2020 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: 032 0000 000.0 00.00 00001 00010101 5000 5000 48(243.8 49.9 229.0 49.8 0824 0624 016 377 51.50 000 1000 110
Mon May 18 19:18:05 2020 INVERTER: Current CRC: F8 54
Mon May 18 19:18:06 2020 INVERTER: QPIRI reply size (102 bytes)
Mon May 18 19:18:06 2020 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: 713 0527 014 384 51.60 000 100 0032 0000 000.0 00.00 000(230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 46.0 110
Mon May 18 19:18:06 2020 INVERTER: Current CRC: B4 DA
Mon May 18 19:18:06 2020 INVERTER: QPIWS reply size (36 bytes)
Mon May 18 19:18:06 2020 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: (100000000000000000000000000000000000000 000.0 00.00 000(230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 46.0 110
Mon May 18 19:18:11 2020 INVERTER: Current CRC: F8 54
Mon May 18 19:18:11 2020 INVERTER: QPIRI reply size (102 bytes)
(230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 46.0 44.0 51.6 51.5 2(243.5 49.8 230 11000000000000000000�
Mon May 18 19:18:11 2020 INVERTER: Current CRC: B4 DA
Mon May 18 19:18:12 2020 INVERTER: QPIWS reply size (36 bytes)
Mon May 18 19:18:12 2020 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: 2 49.9 0736 0534 014 382 51.50 000 1230.0 50.0 21.7 5000 5000 48.0 46.0 44.0 51.6 51.5 2(243.5 49.8 230 110
Mon May 18 19:18:17 2020 INVERTER: Current CRC: F8 54
Mon May 18 19:18:17 2020 INVERTER: QPIRI reply size (102 bytes)
Mon May 18 19:18:17 2020 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: 9.9 230.3 49.8 0713 0526 014 383 51.50 000 100 0032 0000(230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 46.0 110
Mon May 18 19:18:17 2020 INVERTER: Current CRC: B4 DA
Mon May 18 19:18:17 2020 INVERTER: QPIWS reply size (36 bytes)
Mon May 18 19:18:17 2020 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: (1000000000000000000000000000000000050 000 100 0032 0000(230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 46.0 110
Mon May 18 19:18:22 2020 INVERTER: Current CRC: F8 54
Mon May 18 19:18:24 2020 INVERTER: QPIRI reply size (102 bytes)
Mon May 18 19:18:24 2020 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 46.0 44.0 51.6 51.5 2 30 040 0 1 3 9 01 0 1 4(10000000000000 110
Mon May 18 19:18:24 2020 INVERTER: Current CRC: B4 DA
Mon May 18 19:18:24 2020 INVERTER: QPIWS reply size (36 bytes)