ned-kelly/docker-voltronic-homeassistant #5: Add in options to configure the 'byte size' response from various commands

ned-kelly, 2019-09-11:
Several inverters seem to return a different response depending on the firmware version they are running.

- It appears that they all start with the same data as per the protocol manual (battery/pv voltage/amps, etc) however some inverters seem to return additional fields (so far have only seen one additional field) with some extra data that causes the response object to not match the expected bytes.
- An option to add the expected response size in for each command should be added to the configuration and/or we should calculate the response size object automatically based on the fields returned. 
- Refer to [This Issue](https://github.com/ned-kelly/docker-voltronic-homeassistant/issues/4) for further details.

nrm21, 2019-10-16:
@ned-kelly @manio

Hi,
I just looked back at my modifications to manio's original code for this inverter project and I made a modification (see below for exact spot) to calculate the length of the return message inside of the query function.  It's based on where the '0d' character is located.  Perhaps doing that yourself will clear up this bug (since no one will need to choose 97 or 98 byte length based on version)?  Also I'm thinking if in the future the developers of the inverter make more changes to the firmware that change the length of the message in the modbus protocol might also aggravate any bugs that rely on fixed byte message size.

Just some food for though.

https://github.com/nrm21/skymax-demo/commit/0e0a7f857ab1b7002a3874a8cd103774ad6d3b5e#diff-3b16e6ac57023f3a0d12fa51aa05dc86L49

ned-kelly, 2020-01-14:
Thanks mate, hopefully can test once i've got another inverter on my hands!

raph2i, 2020-01-26:
Hey guys, my brother has installed an 5000 inverter. Here i had to change the QPIRI value to 102.

v0d0r, 2020-02-19:
I originally posted [https://github.com/ned-kelly/docker-voltronic-homeassistant/issues/4](url) but I see that issue is closed.
I seem to have a similar issue as issue4 however after changing my QPIRI byte size to 104 and recompiling I still get an error.
Any suggestions would be appreciated?

./inverter_poller -d -1
Wed Feb 19 18:28:54 2020 INVERTER: Debug set
Wed Feb 19 18:28:54 2020 INVERTER: Current CRC: 49 C1
Wed Feb 19 18:28:54 2020 INVERTER: QMOD reply size (5 bytes)
Wed Feb 19 18:28:54 2020 INVERTER: QMOD: 5 bytes read: (L
Wed Feb 19 18:28:54 2020 INVERTER: QMOD query finished
Wed Feb 19 18:28:54 2020 INVERTER: Current CRC: B7 A9
Wed Feb 19 18:28:55 2020 INVERTER: QPIGS reply size (110 bytes)
Wed Feb 19 18:28:55 2020 INVERTER: QPIGS: 110 bytes read: (225.8 50.0 225.8 50.0 0714 0714 023 408 27.10 000 100 0040 00.0 000.0 00.00 00000 00010101 00 00 00000 110
Wed Feb 19 18:28:55 2020 INVERTER: QPIGS query finished
Wed Feb 19 18:28:55 2020 INVERTER: Current CRC: F8 54
Wed Feb 19 18:28:55 2020 INVERTER: QPIRI reply size (104 bytes)
Wed Feb 19 18:28:55 2020 INVERTER: QPIRI: incorrect start/stop bytes. Buffer: (230.0 13.0 230.0 50.0 13.0 3000 3000 24.0 23.0 21.0 28.2 27.0 0 30 060 0 0 2 1 01 0 0 27.0 0 1Q�
Wed Feb 19 18:28:55 2020 INVERTER: Current CRC: B4 DA
Wed Feb 19 18:28:55 2020 INVERTER: QPIWS reply size (36 bytes)
Wed Feb 19 18:28:55 2020 INVERTER: QPIWS: incorrect start/stop bytes. Buffer: (100000000000000000000000000000000000 24.0 23.0 21.0 28.2 27.0 0 30 060 0 0 2 1 01 0 0 27.0 0 1Q�
Wed Feb 19 18:29:00 2020 INVERTER: Current CRC: F8 54

v0d0r, 2020-02-20:
Update on my issue:  Like [https://github.com/rddkrn](url) fix in issue 4 by commenting out I get a nice json output but also not the ideal and not sure what the impact is of just commenting it out but for now I can at least get some stats out of my inverter.

"
"if (ups_qmod_changed && ups_qpigs_changed) { //&& ups_qpiri_changed"

This will ignore "ups_qpiri_change" value. I dont no the other effect of this change."

raph2i, 2020-02-22:
Hey guys, i created a pull request
https://github.com/ned-kelly/docker-voltronic-homeassistant/pull/13

ned-kelly, 2020-02-27:
We still need to create some additional config for other parameters that are fetched, not just the QPIRI -  It's also worth checking out @manio's auto-calculations and seeing if we can get this working, so will leave this issue open for now.

wax911, 2020-03-10:
Hi @ned-kelly I've just created a PR with some additional config params

source-creator, 2020-03-10:
Changing QPIRI receive buffer size to 102 also worked for me

I was getting error: 
```
INVERTER: QPIRI reply size (97 bytes)
INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: ...
```

Inverter Model Name:
Unit label says: `FCS KS5 5000W`
Manual says: `FCS MKSII 5-48`

```
$ lsusb
Bus 001 Device 006: ID 0665:5161 Cypress Semiconductor USB to Serial
```

ned-kelly, 2020-03-19:
This has now been implemented in: https://github.com/ned-kelly/docker-voltronic-homeassistant/pull/14

closed 2020-03-19 by ned-kelly

dpunzeng, 2020-12-26:
Hi guys, I am new on the project. Just put everything together and also ran into this issue. It has obviously also to do with multiple phases (I am running 3 Apex 5Kw-3Phase) and I am getting longer replys. Tried to use the skymax demo but there is no config for buffer size. Any news on this? Greetings vrom Vienna diet

manio, 2020-12-27:
Hi

> Hi guys, I am new on the project. Just put everything together and also ran into this issue. 

What issue exactly?

> It has obviously also to do with multiple phases (I am running 3 Apex 5Kw-3Phase) and I am getting longer replys. Tried to use the skymax demo but there is no config for buffer size. Any news on this? Greetings from Vienna diet

I don't think that buffer size affect reply times at all. Depending on your inverter model (rather the protocol/firmware) it just works or not at all...
You can just adjust buffer size in the code and see if it is a problem in your case or not...

dpunzeng, 2020-12-28:
Ok, thanks. I changed my plan. This free sw does it all with 2 klicks running- if anybody is interested. Best Diet

http://www.multisibcontrol.net/

ans-gw, 2021-02-17:
Hi all..
according to my experience with inverter Pip 5048ME, and Pip 5048GE,Qpiri can change according to different inverter and even different firmware:
 Pip 5048GE:Fw 2043 Qpiri=98 
 Pip 5048ME Fw 71.82 Qpiri=102 ,Fw 71.85 Qpiri=103.

Now QPIGS show additional value data...last second one(2901) is PV Input in Watt...very needed one...also because 
in the current inverter poller me and others users in the forum can not get data from  "PV_in_watts":0.0,  "PV_in_watthour":0.0000 and "SCC_voltage":0.0000.

Shoul add new entry :"PV_panels_power_watts"
I personally try to add this 3 new entry to main.cpp file..but not work...Inverter poller dont show new entryes!
I even dont know if is the right file to modify...my knowledge is what it is!
                printf("  \"Battery_fan_voltage\":%.1f,\n", batt_fan_voltage);
                printf("  \"Eeprom_version\":%.1f,\n", eeprom_version);
                printf("  \"PV_panels_power_watts\":%.1f,\n", pv_panels_power_watts);
                printf("}\n");

Please implement this data...or tell me where must add id,please.
Tue Feb 16 05:07:18 2021 INVERTER: QPIGS reply size (110 bytes)
Tue Feb 16 05:07:18 2021 INVERTER: QPIGS: 110 bytes read: (232.6 50.0 229.9 49.9 0391 0312 007 402 54.40 042 072 0066 0042 284.6 00.00 00000 00010010 00 00 02901 010
Tue Feb 16 05:07:18 2021 INVERTER: QPIGS query finished
Tue Feb 16 05:07:18 2021 INVERTER: Current CRC: F8 54
Tue Feb 16 05:07:19 2021 INVERTER: QPIRI reply size (103 bytes)
Tue Feb 16 05:07:19 2021 INVERTER: QPIRI: 103 bytes read: (230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 51.0 47.0 55.2 53.8 2 010 080 0 2 3 9 01 0 0 53.0 0 0 090
Tue Feb 16 05:07:19 2021 INVERTER: QPIRI query finished

This are the resoult data:
  "Inverter_mode":4,
  "AC_grid_voltage":232.6,
  "AC_grid_frequency":50.0,
  "AC_out_voltage":229.9,
  "AC_out_frequency":49.9,
  "PV_in_voltage":284.6,
  "PV_in_current":18.9,
  "PV_in_watts":0.0,
  "PV_in_watthour":0.0000,
  "SCC_voltage":0.0000,
  "Load_pct":7,
  "Load_watt":312,
  "Load_watthour":5.2000,
  "Load_va":391,
  "Bus_voltage":402,
  "Heatsink_temperature":66,
  "Battery_capacity":72,
  "Battery_voltage":54.40,
  "Battery_charge_current":42,
  "Battery_discharge_current":0,
  "Load_status_on":1,
  "SCC_charge_on":1,
  "AC_charge_on":0,
  "Battery_recharge_voltage":51.0,
  "Battery_under_voltage":47.0,
  "Battery_bulk_voltage":55.2,
  "Battery_float_voltage":53.8,
  "Max_grid_charge_current":10,
  "Max_charge_current":80,
  "Out_source_priority":2,
  "Charger_source_priority":3,
  "Battery_redischarge_voltage":-0.0,
  "Warnings":"00000000000000000000000000000000"

manio, 2021-02-22:
@ned-kelly 
Hi! Last few days I was trying to figure out with @ans-gw (regarding above post) what is he trying to tell us...
From what I understand him he is claiming, that in his firmware for PIP 5048Mg he has additional fields in 110-long QPIGS response.

For instance:
```QPIGS: 110 bytes read: (230.4 49.9 229.8 50.0 0367 0268 007 403 52.70 021 055 0056 0021 260.4 00.00 00000 0001001000 00 01543 010```
The one before last value is PV input power (in watts) - in this case it is 1543 W. I've created a simple patch for him to test obtaining additional values with inverter-poller (CLI):
```
diff --git a/sources/inverter-cli/main.cpp b/sources/inverter-cli/main.cpp
index 7650fab..f8fd31f 100644
--- a/sources/inverter-cli/main.cpp
+++ b/sources/inverter-cli/main.cpp
@@ -132,6 +132,7 @@ int main(int argc, char* argv[]) {
     float scc_voltage;
     int batt_discharge_current;
     char device_status[9];
+    int dummy1,dummy2,dummy3,dummy4;

     // Reply2
     float grid_voltage_rating;
@@ -216,7 +217,8 @@ int main(int argc, char* argv[]) {
             if (reply1 && reply2 && warnings) {

                 // Parse and display values
-                sscanf(reply1->c_str(), "%f %f %f %f %d %d %d %d %f %d %d %d %f %f %f %d %s", &voltage_grid, &freq_grid, &voltage_out, &freq_out, &load_va, &load_watt, &load_percent, &voltage_bus, &voltage_batt, &batt_charge_current, &batt_capacity, &temp_heatsink, &pv_input_current, &pv_input_voltage, &scc_voltage, &batt_discharge_current, &device_status);
+                sscanf(reply1->c_str(), "%f %f %f %f %d %d %d %d %f %d %d %d %f %f %f %d %s %d %d %d %d", &voltage_grid, &freq_grid, &voltage_out, &freq_out, &load_va, &load_watt, &load_percent, &voltage_bus, &voltage_batt, &batt_charge_current, &batt_capacity, &temp_heatsink, &pv_input_current, &pv_input_voltage, &scc_voltage, &batt_discharge_current, &device_status,
+&dummy1, &dummy2, &dummy3, &dummy4);
                 sscanf(reply2->c_str(), "%f %f %f %f %f %d %d %f %f %f %f %f %d %d %d %d %d %d - %d %d %d %f", &grid_voltage_rating, &grid_current_rating, &out_voltage_rating, &out_freq_rating, &out_current_rating, &out_va_rating, &out_watt_rating, &batt_rating, &batt_recharge_voltage, &batt_under_voltage, &batt_bulk_voltage, &batt_float_voltage, &batt_type, &max_grid_charge_current, &max_charge_current, &in_voltage_range, &out_source_priority, &charger_source_priority, &machine_type, &topology, &out_mode, &batt_redischarge_voltage);

                 // There appears to be a discrepancy in actual DMM measured current vs what the meter is
@@ -275,6 +277,10 @@ int main(int argc, char* argv[]) {
                 printf("  \"Charger_source_priority\":%d,\n", charger_source_priority);
                 printf("  \"Battery_redischarge_voltage\":%.1f,\n", batt_redischarge_voltage);
                 printf("  \"Warnings\":\"%s\"\n", warnings->c_str());
+                printf("  \"Test1\":\"%d\"\n", dummy1);
+                printf("  \"Test2\":\"%d\"\n", dummy2);
+                printf("  \"Test3\":\"%d\"\n", dummy3);
+                printf("  \"Test4\":\"%d\"\n", dummy4);
                 printf("}\n");

                 // Delete reply string so we can update with new data when polled again...
```

And he confirmed that this value is in Test3. He'd like to have this called eg. "PV_input_watt" to distinguish this from already existing "PV_in_watts" which we have currently but in his firmware it shows zero.

What is your opinion on such a possible change?

garethmo, 2021-02-22:
@manio I'm hoping this gets done as I have the same problem, and it will stop the client side calculation I'm having to do to work our input watts :)

manio, 2021-02-22:
@garethmo the changes you was asking is just the diff in my above post :)

garethmo, 2021-02-22:
Hi @manio 

I've added the above amendments to my main.cpp file, composed it but I'm clearing doing something stupid.... I don't see the print out Test 1-4

manio, 2021-02-22:
@garethmo 
https://github.com/ned-kelly/docker-voltronic-homeassistant#using-inverter_poller-binary-directly

ans-gw, 2021-02-23:
@ned-kelly  @manio 
This morning(with the sun) i test new inverter-cli and is working...
Tue Feb 23 09:27:07 2021 INVERTER: QPIGS: 110 bytes read: (232.9 49.9 230.1 50.0 0437 0361 008 401 52.80 015 056 0053 0015 296.5 00.00 00000 00010010 00 00 01372 010
 "Battery_redischarge_voltage":-0.4 =this data is not true...battery is in charging now.
  "Warnings":"00000000000000000000000000000000"
  "Test1":"0"
  "Test2":"0"
  "Test3":"1372" this is  "PV_input_watt". This value is True..I compare with WatchPower data.
  "Test4":"10"

Please integarte this in the project.

Thank you very much

manio, 2021-02-23:
@garethmo @ans-gw 
try this: https://skyboo.net/temp/ha.diff

garethmo, 2021-02-23:
> @garethmo @ans-gw
> try this: https://skyboo.net/temp/ha.diff

Thanks @manio ill look into that, i haven't used cmake on my PI before so ill have a bit of reading to do before I get it right, ill have a look at the changes and feedback

ans-gw, 2021-02-24:
I try out this patch,
patching was ok,but Make give some error:
- Configuring done
-- Generating done
-- Build files have been written to: /opt/ha-inverter-mqtt-agent/sources/inverter-cli
pi@InverterPi4:/opt/ha-inverter-mqtt-agent/sources/inverter-cli $ sudo make
Scanning dependencies of target inverter_poller
[ 20%] Building CXX object CMakeFiles/inverter_poller.dir/inputparser.cpp.o
[ 40%] Building CXX object CMakeFiles/inverter_poller.dir/inverter.cpp.o
/opt/ha-inverter-mqtt-agent/sources/inverter-cli/inverter.cpp: In member function ‘bool cInverter::query(const char*, int)’:
/opt/ha-inverter-mqtt-agent/sources/inverter-cli/inverter.cpp:119:32: warning: pointer of type ‘void *’ used in arithmetic [-Wpointer-arith]
         n = read(fd, (void*)buf+i, replysize-i);
                      ~~~~~~~~~~^~
[ 60%] Building CXX object CMakeFiles/inverter_poller.dir/main.cpp.o
/opt/ha-inverter-mqtt-agent/sources/inverter-cli/main.cpp: In function ‘int main(int, char**)’:
/opt/ha-inverter-mqtt-agent/sources/inverter-cli/main.cpp:135:25: error: conflicting declaration ‘int pv_input_watts’
     int dummy1, dummy2, pv_input_watts;
                         ^~~~~~~~~~~~~~
/opt/ha-inverter-mqtt-agent/sources/inverter-cli/main.cpp:129:11: note: previous declaration as ‘float pv_input_watts’
     float pv_input_watts;
           ^~~~~~~~~~~~~~
/opt/ha-inverter-mqtt-agent/sources/inverter-cli/main.cpp:220:391: error: ‘pv_input_watt’ was not declared in this scope
 oltage, &batt_discharge_current, &device_status, &dummy1, &dummy2, &pv_input_watt);
                                                                     ^~~~~~~~~~~~~

/opt/ha-inverter-mqtt-agent/sources/inverter-cli/main.cpp:220:391: note: suggested alternative: ‘pv_input_watts’
 oltage, &batt_discharge_current, &device_status, &dummy1, &dummy2, &pv_input_watt);
                                                                     ^~~~~~~~~~~~~
                                                                     pv_input_watts
make[2]: *** [CMakeFiles/inverter_poller.dir/build.make:89: CMakeFiles/inverter_poller.dir/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/inverter_poller.dir/all] Error 2
make: ***

manio, 2021-02-24:
Fixed

ans-gw, 2021-02-24:
All ok now.
Tomorrow morning,can test with the inverter data with sun.

Thank you very much.

garethmo, 2021-02-25:
Hi @manio I've tried your fix but I also seem to be getting this error as the above

warning: pointer of type ‘void *’ used in arithmetic [-Wpointer-arith]

full print out below

-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/ha-inverter-mqtt-agent/sources/inverter-cli
Scanning dependencies of target inverter_poller
[ 20%] Building CXX object CMakeFiles/inverter_poller.dir/inputparser.cpp.o
[ 40%] Building CXX object CMakeFiles/inverter_poller.dir/inverter.cpp.o
/opt/ha-inverter-mqtt-agent/sources/inverter-cli/inverter.cpp: In member function ‘bool cInverter::query(const char*, int)’:
/opt/ha-inverter-mqtt-agent/sources/inverter-cli/inverter.cpp:119:32: warning: pointer of type ‘void *’ used in arithmetic [-Wpointer-arith]
         n = read(fd, (void*)buf+i, replysize-i);
                      ~~~~~~~~~~^~
[ 60%] Building CXX object CMakeFiles/inverter_poller.dir/main.cpp.o
[ 80%] Building CXX object CMakeFiles/inverter_poller.dir/tools.cpp.o
[100%] Linking CXX executable inverter_poller
[100%] Built target inverter_poller

If I call inverter_poller I seem to get the below CRC failed with no output of the stats

Thu Feb 25 08:20:30 2021 INVERTER: QPIRI reply size (98 bytes)
m...Feb 25 08:20:30 2021 INVERTER: QPIRI: CRC Failed!  Reply size: 98  Buffer: (228.4 49.8 230.0 50.0 21.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 2 1 1 01 0 0 54.0 0 1�

keen to get your thoughts 

thanks Gareth

manio, 2021-02-25:
Guys, I think we should continue this discussion in the new issue:
https://github.com/ned-kelly/docker-voltronic-homeassistant/issues/40

Don't mess the things here...
This issue is closed and not directly related.