ned-kelly/docker-voltronic-homeassistant #9: ImportError: No module named shutil_get_terminal_size

cmsimmmons, 2020-01-12:
Hi there,

Firstly thanks for developing this app for the Pi, it looks awesome!

Secondly I'm sorry for being such a noob, but I'm struggling to get it to work :-(. I have minimal experience with this type stuff :-(. 

I've followed all the instructions (I think), but having issues getting the docker-compose command to work... here is what is happening when I run it, I'd greatly appreciate some help if you have any suggestions?

> pi@raspberrypi:~ $ docker-compose up -d
> Traceback (most recent call last):
>   File "/usr/local/bin/docker-compose", line 6, in <module>
>     from compose.cli.main import main
>   File "/usr/local/lib/python2.7/dist-packages/compose/cli/main.py", line 52, in <module>
>     from .formatter import ConsoleWarningFormatter
>   File "/usr/local/lib/python2.7/dist-packages/compose/cli/formatter.py", line 15, in <module>
>     from backports.shutil_get_terminal_size import get_terminal_size
> ImportError: No module named shutil_get_terminal_size

Thank you!

Casey

cmsimmmons, 2020-01-13:
So after lots of googling, I found a post somewhere and someone recommended to uninstall configparser, which i did, and the issue is now gone BUT a new issue has been created :-(... see latest below:

> Traceback (most recent call last):
>   File "/usr/local/bin/docker-compose", line 6, in <module>
>     from compose.cli.main import main
>   File "/usr/local/lib/python2.7/dist-packages/compose/cli/main.py", line 18, in <module>
>     import docker
>   File "/usr/local/lib/python2.7/dist-packages/docker/__init__.py", line 2, in <module>
>     from .api import APIClient
>   File "/usr/local/lib/python2.7/dist-packages/docker/api/__init__.py", line 2, in <module>
>     from .client import APIClient
>   File "/usr/local/lib/python2.7/dist-packages/docker/api/client.py", line 10, in <module>
>     from .build import BuildApiMixin
>   File "/usr/local/lib/python2.7/dist-packages/docker/api/build.py", line 6, in <module>
>     from .. import auth
>   File "/usr/local/lib/python2.7/dist-packages/docker/auth.py", line 9, in <module>
>     from .utils import config
>   File "/usr/local/lib/python2.7/dist-packages/docker/utils/__init__.py", line 3, in <module>
>     from .decorators import check_resource, minimum_version, update_headers
>   File "/usr/local/lib/python2.7/dist-packages/docker/utils/decorators.py", line 4, in <module>
>     from . import utils
>   File "/usr/local/lib/python2.7/dist-packages/docker/utils/utils.py", line 13, in <module>
>     from .. import tls
>   File "/usr/local/lib/python2.7/dist-packages/docker/tls.py", line 5, in <module>
>     from .transport import SSLHTTPAdapter
>   File "/usr/local/lib/python2.7/dist-packages/docker/transport/__init__.py", line 3, in <module>
>     from .ssladapter import SSLHTTPAdapter
>   File "/usr/local/lib/python2.7/dist-packages/docker/transport/ssladapter.py", line 23, in <module>
>     from backports.ssl_match_hostname import match_hostname
> ImportError: No module named ssl_match_hostname

Any suggestions?

Thanks!

Casey

cmsimmmons, 2020-01-13:
hmm.... I seem to have got it to work ;-)

closed 2020-01-13 by cmsimmmons

diegonc, 2020-07-20:
@cmsimmmons can you explain what you did to fix it? I'm facing the same problem right now. Thanks