ned-kelly/docker-voltronic-homeassistant #10: No Inverter Data Relayed

davidatkadpro, 2020-01-14:
I have successfully mounted the container on my laptop that is plugged into the inverter via USB,

When I start the container the MQTT messages with all the topics and their configs, ie.
home/garage/inverter/sensor/mppsolar_Inverter_mode/config
This has a message body with a json of the name, unit, topic, and icon.

This re-sends every 5 minutes, almost like some form of watchdog.

But there is no message with the inverter data being transmitted.

I have the inverter.config set to device=/dev/hidraw0 
But I did notice when you inspect the container, "Devices" doesn't have the hidraw0 device, I tried loading a Debian version of hidraw0 but that doesn't seem to work seem to add hidraw0 to "Devices".

Is there something I am missing (apart from hair loss lol)?

ned-kelly, 2020-01-14:
Are you using serial or USB?

If you've got the correct mapping set - can you try and follow the debugging that's mentioned in this thread: https://github.com/ned-kelly/docker-voltronic-homeassistant/issues/4

Unfortunately my inverter's currently out of action (lightning strike) so I'm unable to troubleshoot / test any fixes currently.

davidatkadpro, 2020-01-14:
> Are you using serial or USB?
> 
> If you've got the correct mapping set - can you try and follow the debugging that's mentioned in this thread: #4
> 
> Unfortunately my inverter's currently out of action (lightning strike) so I'm unable to troubleshoot / test any fixes currently.

Oh man that sucks, was it on the same earth as the dwelling, I was worried about mine when a storm rolled through the other day? 

I am connecting my laptops USB to the Inverters USB.

Interesting, when setting inverter_poller I get:
```
Tue Jan 14 07:20:30 2020 INVERTER: Debug set
Tue Jan 14 07:20:30 2020 INVERTER: Unable to open device file (errno=2 No such file or directory)
```
Then it just polls this continuously:
`Tue Jan 14 07:20:35 2020 INVERTER: Unable to open device file (errno=2 No such file or directory)`
Not sure if this is referring to the USB driver or what.

davidatkadpro, 2020-01-14:
When I do a inspect of the container under devices I only have:
```
            "Devices": [
                {
                    "PathOnHost": "/dev/bus/usb",
                    "PathInContainer": "/dev/bus/usb",
                    "CgroupPermissions": "rwm"
                },
                {
                    "PathOnHost": "/dev/ttyUSB0",
                    "PathInContainer": "/dev/ttyUSB0",
                    "CgroupPermissions": "rwm"
                },
                {
                    "PathOnHost": "/dev/ttyUSB1",
                    "PathInContainer": "/dev/ttyUSB1",
                    "CgroupPermissions": "rwm"
                },
                {
                    "PathOnHost": "/dev/ttyS0",
                    "PathInContainer": "/dev/ttyS0",
                    "CgroupPermissions": "rwm"
                },
                {
                    "PathOnHost": "/dev/ttyS1",
                    "PathInContainer": "/dev/ttyS1",
                    "CgroupPermissions": "rwm"
                },
                {
                    "PathOnHost": "/dev/ttyS2",
                    "PathInContainer": "/dev/ttyS2",
                    "CgroupPermissions": "rwm"
                }
            ],
```

ned-kelly, 2020-01-14:
Yeah same structure...

Can you try and change your device to `/dev/ttyUSB0` or `/dev/ttyUSB1` and see how you go?

No such file or directory errors just mean there's something funky with the mapping, or it's a different device ... 

Also can you run `lsusb` on the host and see if it's detecting the inverter?

Cheers,

davidatkadpro, 2020-01-14:
well, after all that I discovered on the windows desktop version of docker:

> Can I pass through a USB device to a container?
> Unfortunately, it is not possible to pass through a USB device (or a serial port) to a container as it requires support at the hypervisor level.

It looks like I need to get myself a Raspberry Pi after all. 

Or maybe installing virtual box and run docker with that for now.

ned-kelly, 2020-01-15:
> It looks like I need to get myself a Raspberry Pi after all.

I've been running the [Pi Zero](https://www.aliexpress.com/item/4000049806939.html?spm=a2g0o.productlist.0.0.cc06152fRoTVg3&algo_pvid=e32f988d-bc2d-4629-b2bb-d7c3b10e4356&algo_expid=e32f988d-bc2d-4629-b2bb-d7c3b10e4356-0&btsid=db5f6929-1d1c-4aa1-9745-f12b41eb432b&ws_ab_test=searchweb0_0,searchweb201602_3,searchweb201603_53) without too many issues - Great little device for the price... Just make sure you get a decent quality (genuine) SD card from somewhere reputable like Amazon as crappy SD's seem to hang the device after running for a few days.

closed 2020-01-15 by ned-kelly