ned-kelly/docker-voltronic-homeassistant #15: Add support to install as a Home Assistant Add-on

NeilDuToit92, 2020-03-10:
Since this seems to support adding itself to Home Assistant automatically via MQTT, it would be nice if it can be installed as an add-on in Home Assistant instead of a separate docker container. Makes it even easier for non technical people to use.

Link to add-on docs: https://developers.home-assistant.io/docs/hassio_addon_tutorial/

raph2i, 2020-03-11:
I'll be working on this, when #14 is merged.

ned-kelly, 2020-03-19:
Thanks @raph2i !!!

@NeilDuToit92 - At this stage, the main point of running it in docker, is that generally the device connected to your inverter (such as a raspberry pi etc, that actually runs the logic) would not be in the same physical location as where your home assistant server is...

My inverter for example is in a shed/bunker type room outside with a big battery array (And the raspberry pi has other docker containers running also with other sensors connected such as battery voltage monitors etc), where as my HA setup, Nas, Routers etc are all in a "server cupboard" ... I realise this does not fit everyone's model, but for most people using a highly integrated Home Assistant setup it's going to be somewhat similar.

If we're going to add in support for this to be installed to HaaS for more of a "plug and play" type setup it may also be worth adding some setup instructions (and perhaps someone can test) using something like [esp-link](https://github.com/jeelabs/esp-link) so that people can use an esp32 or something similar to transmit the data from the inverter to the device that's running Hass.io (assuming your Hass device is not physically sitting right next to your inverter).

@raph2i do you want to make a new feature / branch for this one perhaps?

closed 2020-03-19 by ned-kelly