ned-kelly/docker-voltronic-homeassistant #16: Unknown state of sensors

DoctorO123, 2020-03-27:
Hi
My inverter is an Axpert 5kva , I am using the USB connection- followed instructions - running docker on Pi3. 

All sensors now detected on HA, but no values returned. I am not familiar with programming so apologise in advance if this has already been addressed in other correspondence.

Thanks

ned-kelly, 2020-03-31:
@DoctorO123 - Could you please have a read through the following issue here: https://github.com/ned-kelly/docker-voltronic-homeassistant/issues/4 

Then, change your `qpiri, qpiws, qmod, qpigs` (in the `config/inverter.conf` file) to match whatever you see when you run inverter_cli in debug 

Cheers,

ned-kelly, 2020-04-07:
Closing due to inactivity - Please re-open if this is still an issue.

closed 2020-04-07 by ned-kelly