ned-kelly/docker-voltronic-homeassistant #19: create Simple node red flow

dansass, 2020-04-15:
could u possibly make a node red flow to send and receive the status of the inverter. i am used to node red and very new to docker and how it all works. i am having a very hard time installing it on my rpi 3 and spent 6 hours with no luck. error after error :(

dansass, 2020-04-17:
What would actually be very nice is a video tutorial on how to set it up. I finally got docker running as well as home assistant running on docker. But i am stuck on getting my hass dashboard to look like your dashboard in the screenshot. Also no data coming from inverter. Hope to hear from you soon! Thanks

ned-kelly, 2020-04-17:
Hi @dansass - Happy to accept a PR for some setup guides, however installing docker & docker compose is pretty straightforward (We already have 10k + Pulls on docker hub, so obviously the rest of the community hasn't had too many issues also) So it may just be worth submitting a PR with wiki entries with some links to various docker & docker compose setup guides on other sites (Always welcome more documentation :) )

----

As for Node Red, Id be happy to look at exploring, however as the project uses the HA MQTT Topic to actually create the various 'sensor' items, it's not a straightforward change... You should be able to use Node Red to auto create your own triggers and automations (such as auto switching to 'PCP02' during rainy days, and back to 'PCP03' on sunny days etc...)

----

Lastly, Could you please have a read through the following issue here: https://github.com/ned-kelly/docker-voltronic-homeassistant/issues/4

Then, change your qpiri, qpiws, qmod, qpigs (in the config/inverter.conf file) to match whatever you see when you run inverter_cli in debug

** *Hint* ** if you're using a 24v inverter and not the 48v version you'll need to probably change it to the following:

```
qpiri=97

# This allows you to modify the buffersize for the qpiws command
qpiws=36

# This allows you to modify the buffersize for the qmod command
qmod=5

# This allows you to modify the buffersize for the qpigs command
qpigs=110
``` 

Cheers.

dansass, 2020-04-17:
Hey ned-kelly I got a "win win" plan. How about u call me +1 2266020319 and u help me get started from scratch. Then when everything is done and works i could send u some money with paypal for the time u took to help me. Than i can make a youtube video going step by step through the process. It would help everyone including newbies and myself! Hope to hear from you soon! Thanks

ned-kelly, 2020-04-18:
> i tried to download the 2 cards and use them but i always get this

It looks like the 'vertical-stack-in-card' is not actually loaded in your HA setup still - your best bet is going to be to follow the install instructions on their github project (https://github.com/ofekashery/vertical-stack-in-card) and post an issue there if you're still having config issues - Unfortunately i'm not using hass (Just run plain HA in Docker) so I'm not sure on the correct directory structure, but I have a feeling `/opt/ha-inverter-mqtt-agent/` is not going to be the path to your `www/` hass.io directory.

dansass, 2020-04-18:
Sorry I thought hass and plain HA are the same thing. I am running HA on docker too

ned-kelly, 2020-04-19:
So your www directory is going to be a sub directory of your config/ directory for home assistant - not this container ... you’ll also need to make sure you’ve mapped your config directory in the main HA container to save the files on disk - that way if you restart or update the HA container you won’t lose any configuration, your HA database files etc.

There’s an example docker compose file here which will help you get started properly (if you’re not using docker compose or some sort of container orchestration I’d advise to at least use docker compose) - https://www.home-assistant.io/docs/installation/docker/

closed 2020-04-19 by ned-kelly