ned-kelly/docker-voltronic-homeassistant #20: QPIGS: incorrect start/stop bytes

andrea1388, 2020-05-14:
I have a solarpower24 inverter Model Primo VM 3000-24 plus (probably it's the voltronic Axpert MKS 3K-24 Plus).
inverter_poller returns errors on the inverter reply.
This is part of the output with the command inverter_poller -1 -d
Any idea to solve the problem. 
Thanks, Andrea.

Thu May 14 18:35:28 2020 INVERTER: Current CRC: F8 54
Thu May 14 18:35:30 2020 INVERTER: QPIRI reply size (98 bytes)
Thu May 14 18:35:30 2020 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 13.0 230.0 50.0 13.0 3000 3000 24.0 23.5 21.5 29.2 27.0 1 02 060 0 1 1 1 01 0 0 26.0 0 1e?
00053 010cs

Thu May 14 18:35:26 2020 INVERTER: Current CRC: B7 A9
Thu May 14 18:35:28 2020 INVERTER: QPIGS reply size (110 bytes)
Thu May 14 18:35:28 2020 INVERTER: QPIGS: incorrect start/stop bytes.  Buffer: (237.8 50.0 229.9 49.9 0252 0175 008 342 24.84 000 077 0037 0002 042.9 24.86 00005 00110110 00 00 00053 010cs

Thu May 14 18:35:25 2020 INVERTER: Current CRC: 49 C1
Thu May 14 18:35:26 2020 INVERTER: QMOD reply size (5 bytes)
Thu May 14 18:35:26 2020 INVERTER: QMOD: incorrect start/stop bytes.  Buffer: (B??
00000000000000000000000000??
0 24.0 23.5 21.5 29.2 27.0 1 02 060 0 1 1 1 01 0 0 26.0 0 1e?
00055 010V-

tcapanna, 2020-05-14:
Can you please post your inverter.conf ?

andrea1388, 2020-05-16:
Problem seems to be solved.
My inverter ends line with a LF instead of a CR.
Changed it in the inverter.cpp and now it works.
I also rewrite the query method.
Now it empties rx line before sending a command and wait for a 0x0a to process the answer. So predefined answer size is not needed.

.....
    //send a command
    char c;
    while(read(fd,(void *)&c,1)>=1); // empty buffer
    write(fd, &buf, n);
    time(&started);

    replysize=0;
    do {
        n = read(fd, (void *)&c, 1);
	if (n < 0) {
            if (time(NULL) - started > 2) {
                lprintf("INVERTER: %s read timeout", cmd);
                break;
            } else {
                usleep(10);
                continue;
            }
        }
	buf[replysize]=c;
        replysize++;
	if(replysize>=1023) {replysize=0;break;}	
	//printf("rs=%d,c=%x\n",replysize,c);
	if(c==0x0a) break;
    } while (true);
    close(fd);
    buf[replysize]=0;
    if (replysize>3) {

        lprintf("INVERTER: %s reply size (%d bytes)", cmd, replysize);

        if (buf[0]!='(' || buf[replysize-1]!=0x0a) {
            lprintf("INVERTER: %s: incorrect start/stop bytes.  Buffer: %s", cmd, buf);
            return false;
        }

closed 2020-05-19 by andrea1388

gadget78, 2020-07-14:
i also seem to have this issue..
after many hours trying to get the qpiri qpigs at the correct value !!
installed various programs to try an capture raw live data from the inverter.... 
and using hexdecimal confirmed fault... 

are you able to give any info on what you modified in inverter.cpp to get it working ??

andrea1388, 2020-07-17:
Try the version on my repo https://github.com/andrea1388/docker-voltronic-homeassistant <https://github.com/andrea1388/docker-voltronic-homeassistant>
Let me know

gadget78, 2020-08-01:
thought i would update my findings here, 
i did use your version, and it did work, 
BUT i realised it was due to the my qpiri,qpigs etc being all 1 out...

reason i found this out was i captured what was being sent out in Hex Decimal to properly check mine, 
as you said that your inverter ends a line with LF and not CR
but found out my inverter ends the line with 0D 0A
which converting over is CR LF 

so for example my Warnings data buffer in hex decimal is... 
28 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 EB E4 0D 0A
which is a 28 or a ( which marks the start, then thirty two "30", which are 0's thus no warnings, then EB E4 which is the check sum, then 0D 0A which marks the end... 
now is it just mine that pushes BOTH CR and LF ? the manually only states <CR> 
i suppose thats why the qpiri/qpigs etc have been made configurable ! 

but still thankyou :)

porkytin, 2021-12-11:
hello, I'm using the repo post by andrea and still can't get the readings. I've confirmed that my inverter also responds with CR LF at the end. What inverter.conf are you using @gadget78 ?

andrea1388, 2021-12-30:
Run the software in a debugger, put a breakpoint on line 124, after

`n = read(fd, (void *)&c, 1);
`
and see what the read returns.
If it can help I wrote a very simplified version of that software.
Andrea.

porkytin, 2021-12-31:
Hi, thank you for replying. I've already solved my problem. Apparently, the USB-USB connection is not very reliable, changing to RS232-USB using [this adapter](https://www.unitek-products.com/products/usb-2-0-to-serial-converter) allowed me to get the readings.

emaklav, 2024-02-23:
@andrea1388 Andrea, hi! Can you help me with te same error? I was try to start from your repo, but see this one:

root@074eff6467bd:/opt# ./inverter-cli/bin/inverter_poller -d -1
Fri Feb 23 07:18:37 2024 INVERTER: Debug set
Fri Feb 23 07:18:37 2024 INVERTER: Current CRC: 49 C1
Fri Feb 23 07:18:40 2024 INVERTER: QMOD read timeout
Fri Feb 23 07:18:40 2024 INVERTER: QMOD reply too short (0 bytes)
Fri Feb 23 07:18:40 2024 INVERTER: Current CRC: B7 A9
Fri Feb 23 07:18:43 2024 INVERTER: QPIGS read timeout
Fri Feb 23 07:18:43 2024 INVERTER: QPIGS reply size (14 bytes)

Can you say how to modify inverter.cpp?

andrea1388, 2024-02-23:
Hi emaklav, first make sure the serial link is working properly and data arrives to the pc. If everything is ok then place a breakpoint after a read (line 123) to see what happens.
Andrea

lukolszewski, 2024-08-10:
I have exact same issue, after changing in line 138 in inverter.cpp value 0x0d to 0x0a (CR to LF) my inverter also works fine. I'm using an EASUN SMH-II inverter BTW.

lukolszewski, 2025-11-21:
Guys, coming here a year later, as my inverter had this problem again and this time I did something else. I think my qpiri was too low. I had it set to 97 and it had to be 98. The best way to find out what it should be is to set it to some big value like 120, then run inverter_polled -d -1 and see where it errors saying "qpiri too small, received 98 bytes" then you know to se it to 98. Then you do the rest if needed.