ned-kelly/docker-voltronic-homeassistant #22: Buffer Sizes not working for Axpert King 5kw

Vandiliz3r, 2020-05-18:
I have tried getting the correct buffer sizes for the inverter.conf however I cannot get the correct values.

Could you please advise as all my sensors are showing as unknown in home assistant..

Thanks

Vandiliz3r, 2020-05-22:
Hi All,

Please help I am really struggling with the Axpert King Invertor setup with the above.
Any help would be appreciated, i am not really a programmer or anything otherwise i would have climbed into the code to try resolve.

Regards

ned-kelly, 2020-05-26:
Hi @Vandiliz3r can you please try reading through the discussions in https://github.com/ned-kelly/docker-voltronic-homeassistant/issues/5 and https://github.com/ned-kelly/docker-voltronic-homeassistant/issues/4 and change the variables in your inverter.conf file accordingly.

Vandiliz3r, 2020-05-26:
> Hi @Vandiliz3r can you please try reading through the discussions in #5 and #4 and change the variables in your inverter.conf file accordingly.

Thats exactly what i have done.. None of the bytes i insert work. The debuger also does not tell me how many bites are short or whatever..

This is why i asked if anyone has tried this one the axpert king. 
:([image]

shaundekok, 2020-06-25:
Hi, I am getting the following response:

Thu Jun 25 13:20:26 2020 INVERTER: Current CRC: B7 A9
Thu Jun 25 13:20:29 2020 INVERTER: QPIGS read timeout
Thu Jun 25 13:20:29 2020 INVERTER: QPIGS reply too short (16 bytes)
Thu Jun 25 13:20:29 2020 INVERTER: Current CRC: F8 54
Thu Jun 25 13:20:32 2020 INVERTER: QPIRI read timeout
Thu Jun 25 13:20:32 2020 INVERTER: QPIRI reply too short (8 bytes)
Thu Jun 25 13:20:32 2020 INVERTER: Current CRC: B4 DA
Thu Jun 25 13:20:35 2020 INVERTER: QPIWS read timeout
Thu Jun 25 13:20:35 2020 INVERTER: QPIWS reply too short (8 bytes)
Thu Jun 25 13:20:40 2020 INVERTER: Current CRC: 49 C1
Thu Jun 25 13:20:40 2020 INVERTER: QMOD reply size (5 bytes)
�hu Jun 25 13:20:40 2020 INVERTER: QMOD: incorrect start/stop bytes.  Buffer: ^0
Thu Jun 25 13:20:40 2020 INVERTER: Current CRC: B7 A9
Thu Jun 25 13:20:43 2020 INVERTER: QPIGS read timeout
Thu Jun 25 13:20:43 2020 INVERTER: QPIGS reply too short (8 bytes)
Thu Jun 25 13:20:43 2020 INVERTER: Current CRC: F8 54
Thu Jun 25 13:20:46 2020 INVERTER: QPIRI read timeout
Thu Jun 25 13:20:46 2020 INVERTER: QPIRI reply too short (8 bytes)
Thu Jun 25 13:20:46 2020 INVERTER: Current CRC: B4 DA
Thu Jun 25 13:20:49 2020 INVERTER: QPIWS read timeout
Thu Jun 25 13:20:49 2020 INVERTER: QPIWS reply too short (8 bytes)

Any insite?

wax911, 2020-07-02:
@Vandiliz3r I have the exact same model, and I've never had this issue (with the exception of initial installation). What parameters are you using?

Vandiliz3r, 2020-07-02:
Could you post yours then I can try again I have uninstalled this. I will definitely give it a go again!

wax911, 2020-07-02:
Here you go:
```sh
# Basic configuration options for the actual inverter polling process...

device=/dev/hidraw0

# How many times per hour is the program going to run...
# This is used to calculate the PV & Load Watt Hours between runs...
# If unsure, leave as default - it will run every minute...

# (120 = every 30 seconds)...
run_interval=120

# This allows you to modify the amperage in case the inverter is giving an incorrect
# reading compared to measurement tools.  Normally this will remain '1'
amperage_factor=1.0

# This allows you to modify the wattage in case the inverter is giving an incorrect
# reading compared to measurement tools.  Normally this will remain '1'
watt_factor=1.01

# The following settings allow you to modify runtime buffers 
# mentioned in https://github.com/ned-kelly/docker-voltronic-homeassistant/issues/5

# This allows you to modify the buffersize for the qpiri command
qpiri=104

# This allows you to modify the buffersize for the qpiws command
qpiws=67

# This allows you to modify the buffersize for the qmod command
qmod=5

# This allows you to modify the buffersize for the qpigs command
qpigs=110
```

Vandiliz3r, 2020-07-02:
Thanks! I will definately give this a go next week, Lockdown has got me busy here.

NielMorgan92, 2020-07-05:
@wax911 
Thanks for this, I also have an Axpert 5KVA 4KW version (Not a KING) but this helped for me.
I first tried only changing qpiri to 102 or 104 but i still got the "QPIRI: incorrect start/stop bytes." error
What solved it for me was changing qpiws to 67 mine was set to default (i think it was 46)

closed 2021-01-20 by Vandiliz3r