ned-kelly/docker-voltronic-homeassistant #24: No data coming Through

shaundekok, 2020-06-25:
Hi,

I am using this successfully on on Axpert 5KVA, but not getting any data from another on the same hardware. I ran the debug and am getting the following:

Thu Jun 25 13:24:54 2020 INVERTER: Debug set
Thu Jun 25 13:24:54 2020 INVERTER: Current CRC: 49 C1
Thu Jun 25 13:24:54 2020 INVERTER: QMOD reply size (5 bytes)
c configuration options for the actual inverter polling process...s.  Buffer: ^0

# The device to read from...
# Use: /dev/ttyS0 if you have a serial device,
#      /dev/ttyUSB0 if a USB<>Serial,
#      /dev/hidraw0 if you're connecting via the USB port on the inverter.

device=/dev/hidraw0

# How many times per hour is the program going to run...
# This is used to calculate the PV & Load Watt Hours between runs...
# If unsure, leave as default - it will run every minute...

# (120 = every 30 seconds)...
run_interval=120

# This allows you to modify the amperage in case the inverter is giving an incorrect
# reading compared to measurement tools.  Normally this will remain '1'
amperage_factor=1.0

# This allows you to modify the wattage in case the inverter is giving an incorrect
# reading compared to measurement tools.  Normally this will remain '1'
watt_factor=1.01

# The following settings allow you to modify runtime buffers.
# N.B. These values may not be applicable to all inverter types, as such you will 
# nee
Thu Jun 25 13:24:54 2020 INVERTER: Current CRC: B7 A9
Thu Jun 25 13:24:57 2020 INVERTER: QPIGS read timeout
Thu Jun 25 13:24:57 2020 INVERTER: QPIGS reply too short (8 bytes)
Thu Jun 25 13:24:57 2020 INVERTER: Current CRC: F8 54
Thu Jun 25 13:25:00 2020 INVERTER: QPIRI read timeout
Thu Jun 25 13:25:00 2020 INVERTER: QPIRI reply too short (8 bytes)
Thu Jun 25 13:25:00 2020 INVERTER: Current CRC: B4 DA
Thu Jun 25 13:25:03 2020 INVERTER: QPIWS read timeout
Thu Jun 25 13:25:03 2020 INVERTER: QPIWS reply too short (8 bytes)
Thu Jun 25 13:25:08 2020 INVERTER: Current CRC: 49 C1
Thu Jun 25 13:25:08 2020 INVERTER: QMOD reply size (5 bytes)
�hu Jun 25 13:25:08 2020 INVERTER: QMOD: incorrect start/stop bytes.  Buffer: ^0
Thu Jun 25 13:25:08 2020 INVERTER: Current CRC: B7 A9
Thu Jun 25 13:25:11 2020 INVERTER: QPIGS read timeout
Thu Jun 25 13:25:11 2020 INVERTER: QPIGS reply too short (0 bytes)
Thu Jun 25 13:25:11 2020 INVERTER: Current CRC: F8 54
Thu Jun 25 13:25:14 2020 INVERTER: QPIRI read timeout
Thu Jun 25 13:25:14 2020 INVERTER: QPIRI reply too short (8 bytes)
Thu Jun 25 13:25:14 2020 INVERTER: Current CRC: B4 DA
Thu Jun 25 13:25:17 2020 INVERTER: QPIWS read timeout
Thu Jun 25 13:25:17 2020 INVERTER: QPIWS reply too short (8 bytes)
Thu Jun 25 13:25:22 2020 INVERTER: Current CRC: 49 C1
Thu Jun 25 13:25:22 2020 INVERTER: QMOD reply size (5 bytes)
�hu Jun 25 13:25:22 2020 INVERTER: QMOD: incorrect start/stop bytes.  Buffer: ^0
Thu Jun 25 13:25:22 2020 INVERTER: Current CRC: B7 A9
Thu Jun 25 13:25:25 2020 INVERTER: QPIGS read timeout
Thu Jun 25 13:25:25 2020 INVERTER: QPIGS reply too short (8 bytes)
Thu Jun 25 13:25:25 2020 INVERTER: Current CRC: F8 54
Thu Jun 25 13:25:28 2020 INVERTER: QPIRI read timeout
Thu Jun 25 13:25:28 2020 INVERTER: QPIRI reply too short (8 bytes)
Thu Jun 25 13:25:28 2020 INVERTER: Current CRC: B4 DA

Any throughts?

Vandiliz3r, 2020-06-25:
Hey Dude

Could not get this working, however i found a paid solution from a company called ICC software.
They support everything you would want in MQTT and have a nice programmed dashboard that you can install on a PI.

https://iccsoftware.co.za/

That was my solution at the end of the day was worth every $1 ($60) and constant updates and features added regularly.

Regards

alfagero, 2020-06-26:
I don't think your suggestion is very fair to the software developer(ned-kelly). I think this software makes a lot of work.
and if you ever want to have other software then try https://github.com/njfaria/SolPipLog
I use that a few month. works 24/7 perfect. and it is free. you can donate(i made this)
I'll try ned-kelly docker too, probably in winter, because I work with Home Assistant too.

Vandiliz3r, 2020-06-26:
Don't mistake my suggestion as a slap in the face, I could not get it to work after several day and they could not point me in the correct direction so i went for a PAID software that I did not initially want to do.

However if i was able to get this to work i would definitely install it for a few of my friends who all have Axpert Kings.

alfagero, 2020-06-26:
I don't think your suggestion is very fair to the software developer(ned-kelly) to post a commercial link . I think this software makes a lot of work.
and if you ever want to have other software then try https://github.com/njfaria/SolPipLog
I use that a few month. works 24/7 perfect. and it is free. you can donate(i made this)
I'll try ned-kelly docker too, probably in winter, because I work with Home Assistant too.

MuhammadMurtaza, 2020-07-14:
For some reason the compiled program inverter_poller included doesn't work. I compiled it from manio's repo and then copied it into ned-kelly's docker using docker cp command and it now works. No more "Reply too short errors"

gadget78, 2020-07-14:
not looked at manios repo...
but i to was having problems getting it to capture the data, with error of "reply too short errors" etc...

i found #20 were he also had problems, and they tracked it down to the inverter ending lines with a LF instead of a CR....
i found his modifications, and used his inverter.cpp
[andrea1388 inverter.cpp](https://github.com/andrea1388/docker-voltronic-homeassistant/blob/master/sources/inverter-cli/inverter.cpp)
maybe manios also has a workaround ??... 

goes to show that this implementation is going to have issues going forward ?

MuhammadMurtaza, 2020-07-15:
You are right. Perhaps it is also because there are many variants of the inverter itself. e.g. There are some commands and settings that doesnt apply to my inverter. The basic manufacturer is voltronic power but the implementation is slightly different for each country and region. 
I also had issues with the code when I was using a slightly longer cable from inverter to my PC where I was testing it out. I was getting too many reply too shorts and was almost giving up when accidentally switched to shorter cable. 
Anyone having similar issue should also try changing the USB cable.

nz51ww, 2020-08-15:
@MuhammadMurtaza can you detail in a step by step how to copy the manio's inverter_poller into the container with the Docker cp command?  I'm able to read inverter data with Manio's inverter_poller but isn't working with Ned_Kelly's container.
Thanks,

MuhammadMurtaza, 2020-08-17:
Yeah basically just compile manio's inverter_poller and docker cp the file into I believe /opt/inverter-cli/bin (please check if its correct as I dont have the machine where I deployed the code) 
navigate to where manio's inverter_poller is compiled on your host and just use:
`sudo docker cp inverter_poller mycontainerID:/opt/inverter-cli/bin`
Hope this helps

petervanderwalt, 2020-09-09:
> Yeah basically just compile manio's inverter_poller and docker cp the file into I believe /opt/inverter-cli/bin (please check if its correct as I dont have the machine where I deployed the code) [...]

Your help also helped me (EASUN 3200w) -  manio's works, neds, sadly does not