ned-kelly/docker-voltronic-homeassistant #25: 2 converters in parallel mode

gilspringer, 2020-07-21:
Do you have solution for 2 converters in parallel mode ?

Thx

gadget78, 2020-07-21:
have you tried this atall ? i would imagine you would need to have them both connected separately, with two instances running ??
have you tried just connected to one ?? and have issues ?

bobwho69, 2021-01-02:
I too am interested in getting parallel data from my 4x Inverters as they are all a slightly different setup.  I have seen people use 4x dockers and 4x USB cables but that seems pretty taxing.

The manual describes us needing to call for another set of data QPGSn<cr> for parallel information.  I am interested in getting the total output power and total charging current.  But to me it looks like the command would poll a certain inverter at a time, meaning instead of using QPPIGS, we use PGGSn for each inverter.

I am not code savvy having learnt C++ over 20 years ago and not used it  or any other programming language since.  It looks like QPGSn was not coded in sources/inverter-cli/main.cpp but it might not be hard for someone to copy/paste a starting point and include the additional arguments.  It would take me a long time to re-learn how to code...

Sarelseekat, 2021-11-06:
Has anyone found a working solution to this challenge? I too have x3 inverter running in parallel and would very much like to run polling from one raspberry pi. With the chip shortage and my main controller running Hass.io it seems a bit overkill to have a Pi for each inverter and an extra on for Hass.io

Would love to hear from anyone who has found a solution too this.

bobwho69, 2021-11-06:
I have had success using [jblance/mpp-solar](https://github.com/jblance/mpp-solar), running the qpgs0-3 commands on one usb connection.

Sarelseekat, 2021-11-16:
Thanks so much - Will check it out.

Sarelseekat, 2021-11-26:
> I have had success using [jblance/mpp-solar](https://github.com/jblance/mpp-solar), running the qpgs0-3 commands on one usb connection.

Is there a way to qpgs on my current Nedkelly deployment running on raspberry pi?