ned-kelly/docker-voltronic-homeassistant #26: Newer 5KW Hybrid V II Inverters use P18 protocol - hoping to modify/update this work

HaroldMuvuti, 2020-07-24:
Hi Ned-kelly

Greetings from Africa.

Thank you and Great work, we have this working with an earlier 4KW inverter.

We have been trying get the newer 5KW  Hybrid Inverter to work but as you know the V II inverters use P18 Protocol.
looks like the protocol has the same CRC code.
So I have been using  inveterter_poller -r Tnnn -d -1  to communicate with the inverter

root@98f9b19ab295:/#  /opt/inverter-cli/inverter_poller -r Tnnn -d -1
Fri Jul 24 02:56:50 2020 INVERTER: Debug set
Fri Jul 24 02:56:50 2020 INVERTER: Current CRC H722: 9A ED
vri Jul 24 02:56:50 2020 BUF: Tnnn▒▒
Fri Jul 24 02:56:50 2020 N is: -1
Fri Jul 24 02:56:50 2020 N is: 7
vri Jul 24 02:56:50 2020 BUF: (NAKss
Fri Jul 24 02:56:50 2020 INVERTER: Tnnn reply H0722 size (7 bytes)
vri Jul 24 02:56:50 2020 INVERTER: Tnnn: incorrect start bytes.  Buffer: (NAKss
Fri Jul 24 02:56:50 2020 INVERTER: Tnnn: 7 bytes read: (NAK
Fri Jul 24 02:56:50 2020 INVERTER: Tnnn query finished
Reply:  NAK

But when we try using the ^P00X commands which have replaced the QXXX commands used by the earlier inverter we have failed to get a reply successfully e.g.

root@98f9b19ab295:/#  /opt/inverter-cli/inverter_poller -r ^P007PGS0 -d -1
Fri Jul 24 03:05:01 2020 INVERTER: Debug set
Fri Jul 24 03:05:01 2020 INVERTER: Current CRC H722: 29 B6
Fri Jul 24 03:05:06 2020 BUF: ^P007PGS0)▒
Fri Jul 24 03:05:06 2020 N is: 7
Fri Jul 24 03:05:06 2020 BUF: ^P007PGS0)▒
Fri Jul 24 03:05:06 2020 INVERTER: ^P007PGS0 reply H0722 size (7 bytes)
Fri Jul 24 03:05:06 2020 INVERTER: ^P007PGS0: incorrect stop bytes.  Buffer: ^P007PGS0)▒
Reply:
root@98f9b19ab295:/#

Any ideas/suggestions on how we can get talking to the inverter successfully.

The idea is that after we can successfully send commands and recieve back data then we can update/modify the inverter.cpp to handle the diffferent parameters being
Mode
^P00X commands instead of QXXX commands
and any other new/different fields

I think we are close. Willing to work.

Many Thanks
Harold

gadget78, 2020-07-24:
Do you have the Watchpower program working with this inverter ??
If so you could use that to sniff the protocol it's using a etc....

HaroldMuvuti, 2020-07-24:
We have SolarPower working with the inverter and have sniffed the comms using wireshark.

The inverter uses the protocol in the attached protocal manual. The commands start with ^P instead of Q.

gadget78, 2020-08-01:
not sure why you tried with the raw command of Tnnn, and ^P00X  
the inverter_poller program will create the checksum, and add the leading character etc... so the follow should work .. 
inverter_poller -r PID -d -1

which should give back the serial number of the Inverter..
where this would normally be QID in the conventional scheme/manual

if this works, then a simple modification to the program from Q to P can be done.. but would need to be edited in a number of places !

HaroldMuvuti, 2020-08-02:
Thank you

Can not wait to try.

Will update soon.

Best Regards
BG!
Harold

HaroldMuvuti, 2020-08-04:
Breakthrough

Now reading/communicating with inverter.

1. New inverter replies with a ^ instead of a (
So I changed line 138 in inverter.cpp file 
From        if (buf[0]!='(' || buf[replysize-1]!=0x0d) {
To             if (buf[0]!='^' || buf[replysize-1]!=0x0d) {

inverter_poller -r ^P00XXX -d -1 is now working

The buffer setting in the inverter.conf  file for these new V II Hybrid 5KW inverters are:

QMOD       10      now    ^P006MOD
QPIRI         93       now   ^P007PIRI
QPIGS        111    now    ^P005GS
QPIWS       42      now    ^P006FWS

BG!
H

gch1p, 2020-10-24:
I just finished a full-featured program for interacting with this exact inverter model https://github.com/gch1p/isv

BradleyGeldenhuys, 2021-08-02:
> QMOD 10 now ^P006MOD

I have replicated this and not got the same success as you.  Can you perhaps share your files that where modified?

HaroldMuvuti, 2021-08-03:
Might take a day or two to get files to you.

Shalom
H

inventari, 2021-08-30:
@HaroldMuvuti have you shared the files somewhere?

preussal, 2023-02-02:
I also have this inverter and would be interested in a solution