ned-kelly/docker-voltronic-homeassistant #28: Get data from the inverter faster

dansass, 2020-09-29:
Hello i have been using your docker container and i love it! i use node red with mqtt instead of docker and it works great.
I have a generator running all day and there is a external transfer i can control from the pi to switch when there is a overload.

My problem it only reads data from the inverter only every 30 seconds and sometimes it goes overload between the 30s and the inverter shuts down.  

I have increased the run  interval in the config/inverter.conf file but it doesn't seem to make a difference.

Do i have to change something else as well or what am i doing wrong?

dansass, 2020-10-03:
Just to see would happen, i changed the run interval and ran the "docker-compose up -d" and now nothing works. seems to be something with the usb cause i also have a usb to rs485 converter hooked up to an epever charge controller and that isnt responding ether anymore.

dansass, 2020-10-03:
Got it! i copied and pasted the default docker-compose.yml file than ran the "docker-compose up -d" again than it worked. now i remember that i changed something in there and that screwed it up. 

But now i am still trying to read data faster

Can someone help me?

wax911, 2020-10-09:
Hey @dansass if you want to increase the run interval you'll have to change than in `sources/entrypoint.sh` specifically:
[image]
> **Line 15** is usually defaulted to 30 which executes every 30 seconds

There are two ways to do this, you can either change the file in your workspace directly then use `docker cp` to overwrite to the location inside the container which should be `/opt/inverter-mqtt/entrypoint.sh` according to the Dockerfile or you could get inside the container and change the file from there. 

How fast you can get updates depends a lot on your inverter interface comm speed, e.g. in my case with a RPi 4 I ran several commands with the time command to get a basic idea of the minimum time between each execution and it was around 3 - 4 seconds on average. I'm sure you could also just remove the `-n #` from the watch command and it should just run continuously without breaks (I personally have not tested this yet)

> Also don't forget to update your `inverter.conf` file **run_interval** just to assure that you get the correct kWh reading

codefreak558, 2022-09-17:
Hey @wax911 I've tried editing the `sources/inverter-mqtt/entrypoint.sh` outside the container and when recreating the container it doesn't change anything. Is there something specfic that I am supposed to do? As when I overwrite the files in the container and recreate the container it still doesn't update any faster

wax911, 2022-09-17:
Hey @codefreak558 try accessing your containers shell and verify that your changes exist, if you haven't already done so.
> ```sh
> docker exec -it container-name bash
> ```

In my case the fastest I could get was 4 seconds (rpi 4), which accounted for writing, reading, and mqtt publishing. I would suggest that you monitor both your mqtt topics going to the inverter and using something like htop or running the script locally to benchmark just to be sure that nothing else is slowing things down. In the past I had alpine which has terrible `jq` performance

codefreak558, 2022-09-17:
Thanks,

I did end up getting it to work