ned-kelly/docker-voltronic-homeassistant #30: I think description in inverter.conf is wrong

djmoonshine, 2020-10-16:
Hello,

The description in inverter.conf says that run_interval should be how many times per hour the code sends updates to mqtt.

But in the code for watt_hours there is this:
load_watthour = (float)load_watt / (3600 / runinterval);

The "3600 / runinterval"actually calculates the number of times it's running if the runinterval is set to the number of seconds there are between runs. So I think this might be wrong or am I misunderstanding how it should work?

nrm21, 2020-11-04:
Hi, I believe I am the one who originally wrote that code (before ned-kelly forked it with permission).

No I think you have it right.  IIRC that interval was the number of seconds delay between calls of the executable.  When I wrote it my intention was always to have telegraf call the executable and take the results it spat out and shove them into an influxdb or other time series DB like prometheus.  But of course the program won't have a way of knowing what interval telegraf is calling the executable to "run once and exit" at.  So that variable was intended to let the program know so it could calculate the right wattage.

wdross, 2022-02-23:
I came here looking to see if I had this right (or wrong) as well.
What I was coming up with was the runinterval (config value run_interval) needs to be the **_seconds between calls_**.

The definition of WattHours is Watts * Hours, so Watts * (run_interval_seconds / 3600) would be another way to express it.  
Which is the same as Watts / (3600 / run_interval)!

So the comments in the config file should be modified, and I think the cpp code should be changed to promote the pv_input_watts before division to prevent losing precision (just like the load_watt is).