ned-kelly/docker-voltronic-homeassistant #34: No output to InfluxDB

therokh, 2020-12-03:
Hi,

I installed this tonight and got it working to the point of polling data into home assistant via MQTT. I have enabled influx in the config and pointed it to my InfluxDB but I am getting no input into Influx, and no error output to check.

I have tested that the container can write to Influx manually, using the same curl command and filling in the variables.

`root@6e3dd1700b8f:/opt# curl -i -XPOST "http://10.3.5.15:8086/write?db=solar&precision=s" -u ":" --data-binary "solar,device=garage_inverter test=1"
HTTP/1.1 204 No Content
Content-Type: application/json
Request-Id: edc3aad4-3538-11eb-a32b-00505681a71b
X-Influxdb-Build: OSS
X-Influxdb-Version: 1.8.2
X-Request-Id: edc3aad4-3538-11eb-a32b-00505681a71b
Date: Thu, 03 Dec 2020 07:26:59 GMT
`

therokh, 2020-12-03:
This is caused by a missing space in the INFLUX_ENABLED check added in commit 515d8196f60a0f13df532a538189caa45e590921.
I've fixed and tested that it resolves my issue. See pull request #35

marval, 2020-12-04:
Sorry about that ... bash is not really a language I am confident at 😞

therokh, 2020-12-04:
> 
> 
> Sorry about that ... bash is not really a language I am confident at 😞

It's no problem and an easy fix :)

ninlee, 2020-12-20:
i have this problem - no data to influxDB 

i have little to no idea what i am doing but i got as far as getting the mqtt data to homeassistant.

when i setup grafana data sources i have to use http://a0d7b954-influxdb:8086 in the URL do i have to use that same thing in the host feild of the mqrr.json file?

i see that some changes were maded to mqtt-push.sh file what exactly should i change or was the fix already made to the github files?

i have this docker container running on a pi and homeassistant running on a VM on a different machine with influxDb and grafana as add ons in homeassistant.

i am very new to this and any help would be appreciated thank you

rdaguerre, 2021-01-10:
i have same problem. anyone have a solution? thanks

therokh, 2021-01-10:
> 
> 
> i have same problem. anyone have a solution? thanks

Have a look at the attached pull request https://github.com/ned-kelly/docker-voltronic-homeassistant/pull/35, you can make the change manually for now and it will resolve the issue if you have the same cause.

PR #35 (2021-01-17): Fix IF check space issue for INFLUX_ENABLED
Fixes #34
closed 2021-01-17 by ned-kelly