ned-kelly/docker-voltronic-homeassistant #36: Hass.IO - Addon

raph2i, 2020-12-21:
Hey guys,

i've tried to create a hass.io-addon from this container.
But we have to change a lot to achieve this.

Goal: open hass.io userbase to use this addon without need for a second host / supervised installation.

To-do:
 - Use hassio alpine base image (https://github.com/home-assistant/docker-base)
 - We would need to read in all configuration via bashio (https://developers.home-assistant.io/docs/add-ons/configuration/)
 - Find a possibility to mount /dev/hidrawX via hassio into the docker-container

bruceborrett, 2023-01-18:
I was able to get this working by mainly just adding the following config file to the repo (although it did take me a couple weeks to figure out what needs to go in this file):

config.yaml
```
# Home Assistant addon configuration

name: "Voltronic"
description: "Monitor and control Voltronic based inverters from Home Assistant."
version: "1.0.0"
slug: "voltronic"
init: false
udev: true
devices:
  - /dev/hidraw0
  - /dev/hidraw1
  - /dev/hidraw2
arch:
  - aarch64
  - amd64
  - armhf
  - armv7
  - i386
```

I also had to rename Dockerfile.dev to Dockerfile since unfortunately it does not seem possible to specify a custom Dockerfile path for an addon.

So basically to summarize, the steps I took were as follows:

1. Cloned the repo into the addons folder
2. Added the above config file
3. Renamed Dockerfile.dev to Dockerfile
4.  Configured inverter.conf and mqtt.json as per normal

The next step would probably be to make it possible configure mqtt and inverter options from the addon config page, but I have already spent too much time on this for now, maybe I will come back to it later.

GitGab19, 2025-01-09:
In case it's needed, I just created a custom add-on for homeassistant based on the [fork](https://github.com/catalinbordan/docker-voltronic-homeassistant) from @catalinbordan.

https://github.com/GitGab19/addon-voltronic-inverters