ned-kelly/docker-voltronic-homeassistant #39: Missing data in response ?

garethmo, 2021-02-15:
Hi Guys,

I recently got a Kodac 5kw which I believe is the VMiii.., I hooked up my Pi this weekend and got the inverter talking to this awesome software. 

I seem to have a issue some missing data as you can see from the below payload, i have 3 fields that are not returning any data data

  "PV_in_watts":0.0,
  "PV_in_watthour":0.0000,
  "SCC_voltage":0.0000,

I have looked at the current issues list and cant see any thing on this, I have also noticed that a few of the examples that have been given from other people these fields are blank.

I've had a look in watchpower and the values are pulling though to the Bluetooth app so they def there, which leads me to think it may be a code issue. any hints to point me in the right direction would be great. in the interim ill dig some more 

main CPU version is 41.16

Payload returned 

Mon Feb 15 06:04:29 2021 INVERTER: Debug set
Mon Feb 15 06:04:29 2021 INVERTER: Current CRC: 49 C1
Mon Feb 15 06:04:29 2021 INVERTER: QMOD reply size (5 bytes)
Mon Feb 15 06:04:29 2021 INVERTER: QMOD: 5 bytes read: (L
Mon Feb 15 06:04:29 2021 INVERTER: QMOD query finished
Mon Feb 15 06:04:29 2021 INVERTER: Current CRC: B7 A9
Mon Feb 15 06:04:29 2021 INVERTER: QPIGS reply size (110 bytes)
Mon Feb 15 06:04:29 2021 INVERTER: QPIGS: 110 bytes read: (233.5 49.8 233.5 49.8 0326 0244 006 347 49.80 009 047 0040 02.9 157.9 00.00 00000 10010110 00 00 00461 010
Mon Feb 15 06:04:29 2021 INVERTER: QPIGS query finished
Mon Feb 15 06:04:29 2021 INVERTER: Current CRC: F8 54
Mon Feb 15 06:04:30 2021 INVERTER: QPIRI reply size (98 bytes)
Mon Feb 15 06:04:30 2021 INVERTER: QPIRI: 98 bytes read: (230.0 22.6 230.0 50.0 22.6 5200 5200 48.0 46.0 42.0 56.4 54.0 0 30 060 0 1 1 1 01 0 0 54.0 0 1
Mon Feb 15 06:04:30 2021 INVERTER: QPIRI query finished
Mon Feb 15 06:04:30 2021 INVERTER: Current CRC: B4 DA
INVERTER: ampfactor from config is 1.00
INVERTER: wattfactor from config is 1.01
{
  "Inverter_mode":3,
  "AC_grid_voltage":233.5,
  "AC_grid_frequency":49.8,
  "AC_out_voltage":233.5,
  "AC_out_frequency":49.8,
  "PV_in_voltage":157.9,
  "PV_in_current":2.9,
  "PV_in_watts":0.0,
  "PV_in_watthour":0.0000,
  "SCC_voltage":0.0000,
  "Load_pct":6,
  "Load_watt":244,
  "Load_watthour":8.1333,
  "Load_va":326,
  "Bus_voltage":347,
  "Heatsink_temperature":40,
  "Battery_capacity":47,
  "Battery_voltage":49.80,
  "Battery_charge_current":9,
  "Battery_discharge_current":0,
  "Load_status_on":1,
  "SCC_charge_on":1,
  "AC_charge_on":0,
  "Battery_recharge_voltage":46.0,
  "Battery_under_voltage":42.0,
  "Battery_bulk_voltage":56.4,
  "Battery_float_voltage":54.0,
  "Max_grid_charge_current":30,
  "Max_charge_current":60,
  "Out_source_priority":1,
  "Charger_source_priority":1,
  "Battery_redischarge_voltage":2519003719839266771674359440343040.0,
  "Warnings":""
}
Mon Feb 15 06:04:31 2021 INVERTER: All queries complete, exiting loop.

Thanks again
Gareth

garethmo, 2021-02-18:
I found this page below where he experiences the same problem and puts it down to the connection, thoughts ??

https://powerforum.co.za/topic/4969-my-diy-home-automation/page/2/

"The only problem is payloads on low speed USB need to be broken up into 8 bytes chunks so currently this is limited to 5 character commands ( 5chars +2 CRC +1 CR = 8bytes ) 

So commands like POP00 work 100% but ones like PBFTxx.x ( float voltage ) are hit and miss.

This problem doesnt exist if you're using a normal RS232 serial port but not much actually have these anymore."

hanserasmus, 2021-04-28:
I have exactly this same issue. Would also love to have any inputs as to how we can get this fixed to work with the raspberry pi.

AlexPoliteRudenko, 2023-06-02:
Same problem for me. Axpert VM IV 5600. @garethmo I used forked repo from this repo https://github.com/catalinbordan/docker-voltronic-homeassistant that solves problem with 8 bytes and still do not receive values for these fields. I assume, the problem could be with missing value for SCC_voltage as toher values are calculated basing on it. Does anyone get it work for Axpert 5,6 kwt inverter?

garethmo, 2023-06-02:
> Same problem for me. Axpert VM IV 5600. @garethmo I used forked repo from this repo https://github.com/catalinbordan/docker-voltronic-homeassistant that solves problem with 8 bytes and still do not [...]

Yip looks like the same, I managed to just do my calculations in the front end, so I'm not too worried about missing that watt value

AlexPoliteRudenko, 2023-06-02:
@garethmo please, can you share how you managed to do this. I'm totally new to ha. What values do you use for calculation?

garethmo, 2023-06-03:
> @garethmo please, can you share how you managed to do this. I'm totally new to ha. What values do you use for calculation?

If you go to devices and settings and create a helper. Just multipy the Volts and amps to give you watts ;)

AlexPoliteRudenko, 2023-06-04:
@garethmo thanks, I get it. Seems pretty easy. One problem - my inverter stops responding ((( Even restart did not help/ Have you faced such issue?

garethmo, 2023-06-04:
I havent, but just check your mqtt is running