ned-kelly/docker-voltronic-homeassistant #40: QPIGS new value shown.

ans-gw, 2021-02-18:
Inverter Pip-5048Mg Fw71.85 QPIRI=103...FW71.82=QPIRI=102
This my Qpigs answer:
Tue Feb 16 05:07:18 2021 INVERTER: QPIGS reply size (110 bytes)
Tue Feb 16 05:07:18 2021 INVERTER: QPIGS: 110 bytes read: (232.6 50.0 229.9 49.9 0391 0312 007 402 54.40 042 072 0066 0042 284.6 00.00 00000 00010010 00 00 02901 010

Have this new value shows:00 00 02901 010
00=Battery voltage  offset for Fan on
00=Eeprom version
02901=Pv In Power   (THIS very important)
010= device status

I can not see this value on the inverter poller resoult ...i try to modify Main.cpp...but i no have the knowledge to do.
Shoul add new entry :"PV_panels_power_watts"
I personally try to add this 3 new entry to main.cpp file..but not work...Inverter poller dont show new entryes!
I even dont know if is the right file to modify...my knowledge is what it is!
printf(" "Battery_fan_voltage":%.1f,\n", batt_fan_voltage);
printf(" "Eeprom_version":%.1f,\n", eeprom_version);
printf(" "PV_panels_power_watts":%.1f,\n", pv_panels_power_watts);
printf("}\n");

This are the resoult data:
"Inverter_mode":4,
"AC_grid_voltage":232.6,
"AC_grid_frequency":50.0,
"AC_out_voltage":229.9,
"AC_out_frequency":49.9,
"PV_in_voltage":284.6,
"PV_in_current":18.9,
"PV_in_watts":0.0,
"PV_in_watthour":0.0000,
"SCC_voltage":0.0000,
"Load_pct":7,
"Load_watt":312,
"Load_watthour":5.2000,
"Load_va":391,
"Bus_voltage":402,
"Heatsink_temperature":66,
"Battery_capacity":72,
"Battery_voltage":54.40,
"Battery_charge_current":42,
"Battery_discharge_current":0,
"Load_status_on":1,
"SCC_charge_on":1,
"AC_charge_on":0,
"Battery_recharge_voltage":51.0,
"Battery_under_voltage":47.0,
"Battery_bulk_voltage":55.2,
"Battery_float_voltage":53.8,
"Max_grid_charge_current":10,
"Max_charge_current":80,
"Out_source_priority":2,
"Charger_source_priority":3,
"Battery_redischarge_voltage":-0.0,
"Warnings":"00000000000000000000000000000000"

Can please fix this?
On the current inverter poller me and others users in the forum can not get data from "PV_in_watts":0.0, "PV_in_watthour":0.0000 and "SCC_voltage":0.0000,
so this will be a perfect fix for many users.

Thank you very much for help.

garethmo, 2021-02-22:
This is the same issue i posted issue 39, i haven't had any success in the code but its easy enough to calc if you look at the code.

just volts x amps.. i do this client side while im looking for a solution on the polling

ans-gw, 2021-02-22:
Till 1 week ago i was using that formula as solution with my Pip5048Ge...
but now with Pip 5048 Mg(FW 71.82 and 71.85) is not possible anymore because inverter send out incorrect data:
Inverter send out same value for Battery Charge Current and Pv In Current!!..so with that formula i was producing 9800 watt!!! Very good...but not real.(can see below)

In the past 2 days Manio kindly and patiently made corrections to my wrongly modified Main.cpp file and send me back;See here:https://github.com/ned-kelly/docker-voltronic-homeassistant/issues/5#issuecomment-783180111
I  had to  debug and recompile in my raspberryPi4 and tested the new file Inverter-cli.Now it shows new value:
Mon Feb 22 09:34:11 2021 INVERTER: QPIGS: 110 bytes read: (230.4 49.9 229.8 50.0 0367 0268 007 403 52.70 021 055 0056 0021 260.4 00.00 00000 00010010 00 00 01543 010

Mon Feb 22 09:34:12 2021 INVERTER: QPIRI: 102 bytes read: (230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 51.0 47.0 55.2 53.8 2 10 060 0 2 3 9 01 0 0 53.0 0 1 090

"Inverter_mode":4,
  "AC_grid_voltage":230.4,
  "AC_grid_frequency":49.9,
  "AC_out_voltage":229.8,
  "AC_out_frequency":50.0,
  "PV_in_voltage":260.4,
  "PV_in_current":21.0,
  "PV_in_watts":0.0,
  "PV_in_watthour":0.0000,
  "SCC_voltage":0.0000,
  "Load_pct":7,
  "Load_watt":268,
  "Load_watthour":8.9333,
  "Load_va":367,
  "Bus_voltage":403,
  "Heatsink_temperature":56,
  "Battery_capacity":55,
  "Battery_voltage":52.70,
  "Battery_charge_current":21,
  "Battery_discharge_current":0,
  "Load_status_on":1,
  "SCC_charge_on":1,
  "AC_charge_on":0,
  "Battery_recharge_voltage":51.0,
  "Battery_under_voltage":47.0,
  "Battery_bulk_voltage":55.2,
  "Battery_float_voltage":53.8,
  "Max_grid_charge_current":10,
  "Max_charge_current":60,
  "Out_source_priority":2,
  "Charger_source_priority":3,
  "Battery_redischarge_voltage":-0.4,
  "Warnings":"00000000000000000000000000000000"
  "Test1":"0"
  "Test2":"0"
  "Test3":"1543"
  "Test4":"10"

garethmo, 2021-02-22:
Thanks @ans-gw , ill look into this, perhaps I can can modify my .cpp file and get the data too as i have this exact problem, can you add a link to your modified file so i can have a complete view of the change ?

ans-gw, 2021-02-22:
> Thanks @ans-gw , ill look into this, perhaps I can can modify my .cpp file and get the data too as i have this exact problem, can you add a link to your modified file so i can have a complete view of the change ?

The new file main.cpp must be compiled in your device...The file generated is  Inverter-poller,
is not sending any data to Home Assistant..You can only see Inverter poller data as in in previus post.

Need ned-kelly to integrate this inside the project to unable data been shown in home assistant.

garethmo, 2021-02-22:
Hi @ans-gw  I've modified my file but cant seem to get it up, perhaps a error on my side . Can you paste your main.cpp text here

manio, 2021-02-25:
@garethmo
The warning you mentioned has not influence on the whole compilation process - this is only a warning which is not related with my patch...
You probably has it wrong configured (the reply size in the config).
Please ask @ans-gw for the help as he managed to get it working...

garethmo, 2021-02-25:
@manio thanks for the reply, you were 100% correct apologies  I was referencing a incorrect conf file... once corrected i can see the payload now in the response. i did notice this at the end **Bus error ** which i had not seen before. before the modification. 

 Thu Feb 25 09:26:22 2021 INVERTER: Debug set
Thu Feb 25 09:26:22 2021 INVERTER: Current CRC: 49 C1
Thu Feb 25 09:26:22 2021 INVERTER: QMOD reply size (5 bytes)
Thu Feb 25 09:26:22 2021 INVERTER: QMOD: 5 bytes read: (L
Thu Feb 25 09:26:22 2021 INVERTER: QMOD query finished
Thu Feb 25 09:26:22 2021 INVERTER: Current CRC: B7 A9
Thu Feb 25 09:26:22 2021 INVERTER: QPIGS reply size (110 bytes)
Thu Feb 25 09:26:22 2021 INVERTER: QPIGS: 110 bytes read: (231.7 49.8 231.7 49.8 2201 2188 044 376 53.50 021 093 0035 08.8 148.8 00.00 00000 00010110 00 00 01312 010
Thu Feb 25 09:26:22 2021 INVERTER: QPIGS query finished
Thu Feb 25 09:26:22 2021 INVERTER: Current CRC: F8 54
Thu Feb 25 09:26:23 2021 INVERTER: QPIRI reply size (98 bytes)
Thu Feb 25 09:26:23 2021 INVERTER: QPIRI: 98 bytes read: (230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 46.0 42.0 56.4 54.0 0 30 060 0 2 1 1 01 0 0 54.0 0 1
Thu Feb 25 09:26:23 2021 INVERTER: QPIRI query finished
Thu Feb 25 09:26:23 2021 INVERTER: Current CRC: B4 DA
INVERTER: ampfactor from config is 1.00
INVERTER: wattfactor from config is 1.01
{
  "Inverter_mode":3,
  "AC_grid_voltage":231.7,
  "AC_grid_frequency":49.8,
  "AC_out_voltage":231.7,
  "AC_out_frequency":49.8,
  "PV_in_voltage":148.8,
  "PV_in_current":8.8,
  "PV_in_watts":0.0,
  "PV_in_watthour":0.0000,
  "SCC_voltage":0.0000,
  "Load_pct":44,
  "Load_watt":2188,
  "Load_watthour":72.9333,
  "Load_va":2201,
  "Bus_voltage":376,
  "Heatsink_temperature":35,
  "Battery_capacity":93,
  "Battery_voltage":53.50,
  "Battery_charge_current":21,
  "Battery_discharge_current":0,
  "Load_status_on":1,
  "SCC_charge_on":1,
  "AC_charge_on":0,
  "Battery_recharge_voltage":46.0,
  "Battery_under_voltage":42.0,
  "Battery_bulk_voltage":56.4,
  "Battery_float_voltage":54.0,
  "Max_grid_charge_current":30,
  "Max_charge_current":60,
  "Out_source_priority":2,
  "Charger_source_priority":1,
  "Battery_redischarge_voltage":147846839953773730208366180891300462592.0,
  "Warnings":""
  "PV_input_watts":"1312"
}
Thu Feb 25 09:26:24 2021 INVERTER: All queries complete, exiting loop.
Bus error

ans-gw, 2021-02-25:
Hi,
also for me all ok...

Thu Feb 25 06:34:09 2021 INVERTER: QPIGS: 110 bytes read: (231.6 50.0 229.9 49.9 0436 0367 008 418 54.10 018 069 0062 0018 329.6 00.00 00000 00010010 00 00 01539 010

 "Out_source_priority":2,
  "Charger_source_priority":3,
  "Battery_redischarge_voltage":-0.0,
  "Warnings":"00000000000000000000000000000000"
  "PV_input_watts":"1539"

garethmo, 2021-02-25:
@ans-gw do you also get the Bus error after the payload ?, and have you managed to see the data in HA ?

ans-gw, 2021-02-25:
no have error payload....
can not see data in HA....i think Ned Kelly must integrate the patch first...
...me i dint know how to do...i m very noob...i have a pi4 from 2 months only....
i think you knows more than me...   ...
I started...but I m not able to go on...so please follow this...i think we are half of the way

manio, 2021-02-25:
Guys, the required changes for ha are in my last patch, it just needs to be compiled correctly or properly build in docker, no idea because i am not using ha...

ans-gw, 2021-02-25:
mmmm.... this is out of my possibility

> Guys, the required changes for ha are in my last patch, it just needs to be compiled correctly or properly build in docker, no idea because i am not using ha...

mmm..for me is out of my knowledge...

ans-gw, 2021-02-25:
@garethmo 
...can you try to integrate the new patch and send to home assistant ?
i really need Pv Input Watt ...is the most wanted data !

I post my process how to do the update...for everybody want to try and cooperate,
please post your too.

1  cd /opt/ha-inverter-mqtt-agent:
  sudo docker-compose down
2  cd /opt 
    sudo rm -r ha-inverter-mqtt-agent
3 from root directory :
  sudo git clone https://github.com/ned-kelly/docker-voltronic-homeassistant.git /opt/ha-inverter-mqtt-agent
4 cd /opt/ha-inverter-mqtt-agent
  sudo wget https://skyboo.net/temp/ha.diff
  sudo patch -p1 < ha.diff
5 # Configure the 'device=' directive (in inverter.conf) to suit for RS232 or USB.. 
  sudo vi config/inverter.conf
# Configure your MQTT server's IP/Host Name, Port, Credentials, HA topic, and name of the Inverter that you want displayed in Home Assistant...
# If your MQTT server does not need a username/password just leave these values empty.
    sudo vi config/mqtt.json
copy inverter.conf from opt/ha-inverter-mqtt-agent/config  to opt/ha-inverter-mqtt-agent/sources/inverter-cli 
 and overwrite the inverter.conf already exist in
  inverter-cli. 
6 cd /opt/ha-inverter-mqtt-agent/sources/inverter-cli
  sudo  cmake .  
  sudo  make
   ./inverter-poller ….  ./inverter_poller -d -1

garethmo, 2021-02-26:
Hi @ans-gw i will try integrate the new patch and send the data to home assistant as i need to use this too. at the moment i have no idea how to do it but ill read up next week and see if i can get started :)

ans-gw, 2021-02-26:
> Hi @ans-gw i will try integrate the new patch and send the data to home assistant as i need to use this too. at the moment i have no idea how to do it but ill read up next week and see if i can get started :)

Ok...very good...

Everybody can try to contubute...

ans-gw, 2021-03-04:
Hello...hello...
did somebody try out ?

ans-gw, 2021-03-09:
> Hi @ans-gw i will try integrate the new patch and send the data to home assistant as i need to use this too. at the moment i have no idea how to do it but ill read up next week and see if i can get started :)

Hi ,did you try ...have some news?

TheRoarman, 2021-03-21:
Hi all, Thank you for the work that has already been completed thus far. I'm a bit short on time but will try if I get a chance as I have the same problem with my Kodak OG5.48 (VMIII).
I quickly patched the inverter-cli as @ans-gw did and it worked 100%, but as i said, if i get a chance later I'll look at the docker code and HA later.

ans-gw, 2021-03-21:
> Kodak OG5.48 (

Hi...welcome....It would be very appreciated...
me i dont know how to do.

Which are your Kodak OG5.48 (VMIII) Qpiri;qpiws ecc setting?
Please post also your Qpiri and Qpiws result....
would like to see if have more data from  this inverter Model...

Thank you for cooperation

TheRoarman, 2021-03-21:
Hi @ans-gw , Sure, no problem. I still have lots of work to get my system running the way i want it. At the moment I am doubting the firmware on the inverter. I'm not happy how the MPPT controler handles the PV charging. There are definately some bugs in the firmware. Thus I need the data collection to work correctly so I can figure out the bugs.

Sun Mar 21 14:02:40 2021 INVERTER: Debug set
Sun Mar 21 14:02:40 2021 INVERTER: Current CRC: 49 C1
Sun Mar 21 14:02:40 2021 INVERTER: QMOD reply size (5 bytes)
Sun Mar 21 14:02:40 2021 INVERTER: QMOD: 5 bytes read: (B
Sun Mar 21 14:02:40 2021 INVERTER: QMOD query finished
Sun Mar 21 14:02:40 2021 INVERTER: Current CRC: B7 A9
Sun Mar 21 14:02:40 2021 INVERTER: QPIGS reply size (110 bytes)
Sun Mar 21 14:02:40 2021 INVERTER: QPIGS: 110 bytes read: (229.9 50.2 229.9 50.1 0712 0635 014 369 52.90 003 097 0037 06.2 145.1 00.00 00000 10010110 00 00 00907 110
Sun Mar 21 14:02:40 2021 INVERTER: QPIGS query finished
Sun Mar 21 14:02:40 2021 INVERTER: Current CRC: F8 54
Sun Mar 21 14:02:41 2021 INVERTER: QPIRI reply size (98 bytes)
Sun Mar 21 14:02:41 2021 INVERTER: QPIRI: 98 bytes read: (230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 47.0 43.0 58.4 54.0 1 20 060 1 2 1 1 01 0 0 54.0 0 1
Sun Mar 21 14:02:41 2021 INVERTER: QPIRI query finished
Sun Mar 21 14:02:41 2021 INVERTER: Current CRC: B4 DA
Sun Mar 21 14:02:41 2021 INVERTER: QPIWS reply size (40 bytes)
Sun Mar 21 14:02:41 2021 INVERTER: QPIWS: 40 bytes read: (000000000000000000000000000000000000
Sun Mar 21 14:02:41 2021 INVERTER: QPIWS query finished
INVERTER: ampfactor from config is 1.00
INVERTER: wattfactor from config is 1.01
{
  "Inverter_mode":4,
  "AC_grid_voltage":229.9,
  "AC_grid_frequency":50.2,
  "AC_out_voltage":229.9,
  "AC_out_frequency":50.1,
  "PV_in_voltage":145.1,
  "PV_in_current":6.2,
  "PV_in_watts":0.0,
  "PV_in_watthour":0.0000,
  "SCC_voltage":0.0000,
  "Load_pct":14,
  "Load_watt":635,
  "Load_watthour":21.1667,
  "Load_va":712,
  "Bus_voltage":369,
  "Heatsink_temperature":37,
  "Battery_capacity":97,
  "Battery_voltage":52.90,
  "Battery_charge_current":3,
  "Battery_discharge_current":0,
  "Load_status_on":1,
  "SCC_charge_on":1,
  "AC_charge_on":0,
  "Battery_recharge_voltage":47.0,
  "Battery_under_voltage":43.0,
  "Battery_bulk_voltage":58.4,
  "Battery_float_voltage":54.0,
  "Max_grid_charge_current":20,
  "Max_charge_current":60,
  "Out_source_priority":2,
  "Charger_source_priority":1,
  "Battery_redischarge_voltage":-0.3,
  "Warnings":"000000000000000000000000000000000000"
  "PV_input_watts":"907"

ans-gw, 2021-03-21:
Hi @TheRoarman ...
thanks ...i think must add "PV_input_watts":("907") in mqtt.json....
I see data are same as my Inverter 5048 Mg ...
Which kind of battery you use?   58.4 54.0V ?!
mmm charge problem...try update Firmware first.

Thanks for cooperation

TheRoarman, 2021-03-22:
> Hi @TheRoarman ...
> thanks ...i think must add "PV_input_watts":("907") in mqtt.json....
> I see data are same as my Inverter 5048 Mg ...
> Which kind of battery you use? 58.4 54.0V ?!
> mmm charge problem...try update Firmware first.
> 
> Thanks for cooperation

Hi @ans-gw and others,
I rebuilt the docker image using local source files and input_watts is now registered but unfortunately now the cli is "core dumping". so no new values are coming through. I'll have to spend some time to find out why the CLI works outside the docker but cores in the docker. I'm new to docker.

PLEASE PLEASE PLEASE. this is just a hack to get it working on my system. The code needs to be updated properly and ned-kelly needs to update his docker multiarch build. I'll let you know if I make any progress. If you want to do the same, read the 'docker-compose.yml' and Dockerfile.dev to re-build docker image.

Vandiliz3r, 2021-03-23:
Yeah i came right with the above method. Thanks guys!

TheRoarman, 2021-03-23:
@manio and @ans-gw , found two small bugs in the ha.diff.
1. on line 220. There is an extra %d that should be removed at the end. This caused the cli command to SegFault in docker.
2. on line 278. A comma "," is needed before the \n to show JSON that there is another entry comming.
I don' t have a place to put a modified ha.diff for you. so I've attached a diff to this message. The diff includes the changes in docker files for a local build. 
1. docker-compose down
2. apply the diff
3. docker-compose build
4. docker-compose up -d

Well, this is what I did. Hopefully in the morning when the sun comes up, so do my PV values. :-)
I'm thinking of doing a pull request but there are a few other things I want to look at first. My MQTT subs from HA is not working and my SCC values are ZERO.

Good luck.

manio, 2021-03-24:
@TheRoarman
thanks, i don't see any diff :)
keep in mind that the ha.diff was only rather a proof-of-concept and the changes has to be done in @ned-kelly official repository.
Unfortunatelly @ned-kelly seems to be inactive on github :(

TheRoarman, 2021-03-24:
> @TheRoarman
> thanks, i don't see any diff :)
> keep in mind that the ha.diff was only rather a proof-of-concept and the changes has to be done in @ned-kelly official repository.
> Unfortunatelly @ned-kelly seems to be inactive on github :(

Hi @manio , you are correct and thats' why I need to do a pull request sometime. Sorry, I forgot to attach the file. Here is it.
[ha_local_docker_build_v2.zip](https://github.com/ned-kelly/docker-voltronic-homeassistant/files/6197305/ha_local_docker_build_v2.zip)

ned-kelly, 2021-03-24:
Hi Guys, happy to accept [new PR's](https://github.com/ned-kelly/docker-voltronic-homeassistant/pulls) (there's none atm) however I'm super busy the next 6 months with a large work project so won't be adding any more features till Q3 this year...

That said, I'm also in the process of building a new house (which will be fully IoT connected) so I'll probably implement a bunch of requested features in Q3-Q4.

Happy to accept any (bug free) PR's in the mean-time :)

manio, 2021-03-24:
@TheRoarman 
Ok, I just put it temporarily here for everyone interested:
https://skyboo.net/temp/ha_local_docker_build_v2.diff

ans-gw, 2021-03-24:
Hi..Im very glad to see today here the creators of this project...
Thanks to Manio for his help and patience to help  to implement new inverter data and ned-kelly for this fantastic project.
Hope to see soon  diff officially apply to the project.

ans-gw, 2021-03-24:
> @manio and @ans-gw , found two small bugs in the ha.diff.
> 
> 1. on line 220. There is an extra %d that should be removed at the end. This caused the cli command to SegFault in docker.
> 2. on line 278. A comma "," is needed before the \n to show JSON that there is another entry comming.
>    I don' t have a place to put a modified ha.diff for you. so I've attached a diff to this message. The diff includes the changes in docker files for a local build.
> 3. docker-compose down
> 4. apply the diff
> 5. docker-compose build
> 6. docker-compose up -d
> 
> Well, this is what I did. Hopefully in the morning when the sun comes up, so do my PV values. :-)
> I'm thinking of doing a pull request but there are a few other things I want to look at first. My MQTT subs from HA is not working and my SCC values are ZERO.
> 
> Good luck.

Hi,did you manage to get the new value into Ha by Mqtt?...has new patch V2 solved?
If yes,please post the full process as I posted before...will be usefull for all noobs as me and others too.

Thank you

InsanityFlea, 2021-04-06:
It works, thank you! 

Just change the config/mqtt.json line to show

"PV_input_watts":  "pv_input_watts",

Then change your dashboard card sensor to
entity: sensor.voltronic_pv_input_watts

ans-gw, 2021-04-14:
Is working..
test with new Pip5048MGX inverter....Qpiri=103.
Readings from inverter are working...but can not send data to inverter to changeinverter mode/charge priority:POPxx and PCPxx.

ans-gw, 2021-04-14:
Is working..
test with new Pip5048MGX inverter....Qpiri=103.
   "Charger_source_priority":3,
  "Battery_redischarge_voltage":-0.0,
  "Warnings":"",
  "PV_input_watts":"825"

Readings from inverter are working...but can not send data to inverter to change inverter mode/charge priority:POPxx and PCPxx.???!!!   
Something wrong in the patch?
Please check with your inverters if have same problem.

InsanityFlea, 2021-04-30:
For the people who cant build it or confused, I have a multi-arch image with the patch, this image polls every 5 seconds. You can change your inverter.conf to run_interval=720 to get really fast data

1. Change your docker-compose.yml image
cd /opt/ha-inverter-mqtt-agent
sudo nano docker-compose.yml
change the line **image: bushrangers/ha-voltronic-mqtt** to **image: insanityflea/voltronic-mqtt**

2. Change the mqtt.json
cd /opt/ha-inverter-mqtt-agent/config
sudo nano mqtt.json
Change the line **"PV_in_watts":  "pv_input_watts",** to **"PV_input_watts":  "pv_input_watts",**

3. Change your dashcard from **sensor.voltronic_pv_in_watts** to **sensor.voltronic_pv_input_watts**

ans-gw, 2021-05-12:
> For the people who cant build it or confused, I have a multi-arch image with the patch, this image polls every 5 seconds. You can change your inverter.conf to run_interval=720 to get really fast [...]

I tryed...can not modify "run_interval"= any number i try, interval is same: 38 second...

Please give some suggestions
Thank you

InsanityFlea, 2021-05-12:
> I tryed...can not modify "run_interval"= any number i try, interval is same: 38 second...
>
> Please give some suggestions
> Thank you

Then you are hardware/mqtt limited if you cant get the data any faster, I even trimmed down the commands my data comes in every 10 seconds.

ans-gw, 2021-05-12:
> Then you are hardware/mqtt limited if you cant get the data any faster, I even trimmed down the commands my data comes in every 10 seconds.

...10 second would be nice for me too...
...."hardware/mqtt  limited "?!...what you mean?...where could be?..
I have Raspberry Pi4 8Gb with Hassio and other Pi4 4Gb with Inverter and Bms connected to...and Node Red running.

38 seconds are very annoing...
Thanks for helps

garethmo, 2021-05-13:
> I tryed...can not modify "run_interval"= any number i try, interval is same: 38 second...
>
> Please give some suggestions
> Thank you

> For the people who cant build it or confused, I have a multi-arch image with the patch, this image polls every 5 seconds. You can change your inverter.conf to run_interval=720 to get really fast [...]

Is this the link to your docker image https://hub.docker.com/r/insanityflea/voltronic-mqtt  ?

InsanityFlea, 2021-05-14:
Yes

ans-gw, 2021-05-14:
> ...10 second would be nice for me too...
> ...."hardware/mqtt limited "?!...what you mean?...where could be?..
> I have Raspberry Pi4 8Gb with Hassio and other Pi4 4Gb with Inverter and Bms connected [...]

...all working now.
I set "run_interval"=360; polls every 14 second

Thank you.

SunnyPerth, 2021-05-17:
Hi all, @ans-gw. 

I've just build myself an off grid installation with a 5048MGX and a decent size battery.  Looking to automate a few things, including the ability to automatically charge from grid during sunny days. (this as I have an existing solar setup that does feed into the grid). Looks like this solution will work, and over the last few days I've come quite a bit, but hope you guys can give me a few hints and tips.  I'm running the software on a PI 3b, with HA, Mosquitto and InfluxDB on my Synology NAS. 

(1) @ans-gw  - I understand you have the SAME inverter (MPP 5048MGX). **What are your final qpiri, qpiws, qmod, qpigs values?** I've currently set them to : 102, 40, 5 and 110.  This, comes back saying no errors (but looks like there redischarge voltage is incorrect:
>   "Battery_redischarge_voltage":2564639141447483030128240755736576.0,
>   "Warnings":"100000000000000000000000000000000000",
>   "PV_input_watts":"0"
I've tried various values for qpiri and qpiws, but without much luck (changing these two give an error). 

(2)  I've installed influxdb2, which works. I can log in via the user interface, have HA talking to influxdb2 and storing sensor data in the correct bucket. However, I understand that mqtt.json configuration file we should be able to write data directly into influxdb as well. I can not get that part to work.  (I did setup a influx1 db, which I see directly accepting data from the raspberry.    
**Are you using Influxdb2 or a 1.x version?  If influxdb2, can you share how you did configure influxdb2 to accept the influxdb1 API call?** 

(3) (sorry last question) - Do I have to recompile the Voltronic code once I update the configuration file? Or is a restart of the container enough? Just wondering. 

Thanks in advance! 
y.

ans-gw, 2021-05-19:
Hi @SunnyPerth,
im not an expert..i m quite noob..but i try to answer.
My setting are: 103 36 5 110 
Which Fw have your Mgx?   Your Bms brand?

infux db i no use...(hard to configure for me).

About installation:
as you read i got some problem in beginning:
You must do sudo docker-compose down  and  sudo docker-compose up -d 
Change your docker-compose.yml image
cd /opt/ha-inverter-mqtt-agent
sudo nano docker-compose.yml
change the line image: bushrangers/ha-voltronic-mqtt to image: insanityflea/voltronic-mqtt
**and uncomment for create local build....=must cancel "#" from the line in docker-compose.yml.**

I also dowloaded manually the image  insanityflea/voltronic-mqtt:="docker pull insanityflea/voltronic-mqtt" because was not working in the other way.

Not sure if my steps are properly correct,but now is working

SunnyPerth, 2021-05-20:
Hi @ans-gw, 

Thanks for getting back to me, really appreciated!  I'm having a MPP Solar MPP-5048MGX. Not sure how to check firmware, but the software tells me main CPU version is at 00071.04. 
I tried your settings, however both qpiri ( at 103)  and qpiws (at 36) give an error.  I have to set them to 102 (qpiri) and 40 (qpiws) to get a reading. Still get that faulty Battery_redischarge_voltage reading though. 

Building the docker container I have already insanityFlea's build file, and compile locally. 

Hmmm. Must be something else that is wrong with reading those last bits from the inverter!

cool around Influxdb - yeah, it's a bit of a riddle to setup. I don't think this fine little piece of software is written to work with influxdb2 yet; and influxdb2 is a lot easier to work with (imho)(but I'm 100% new to it). 

Thanks buddy - will keep troubleshooting! 

Cheers, y.

ans-gw, 2021-05-21:
> Hi @ans-gw,
>
> Thanks for getting back to me, really appreciated! I'm having a MPP Solar MPP-5048MGX. Not sure how to check firmware, but the software tells me main CPU version is at 00071.04.
> I tried [...]

closed 2021-05-21 by ans-gw