ned-kelly/docker-voltronic-homeassistant #43: Not sending mqtt

TaariqSurtee, 2021-05-24:
Hi. my installation was working perfectly, then all of a sudden it is no longer sending MQTT messages. 
MQTT is working since my other Sonoff devices work.
The rasp-pi is getting info from the inverter (see below).
It is simply not sending MQTT.
I rebuilt from scratch, but nothing works.

command:
sudo docker exec -it voltronic-mqtt bash -c '/opt/inverter-cli/bin/inverter_poller -d -1'
gives:
{
  "Inverter_mode":3,
  "AC_grid_voltage":236.4,
  "AC_grid_frequency":49.8,
  "AC_out_voltage":229.4,
  "AC_out_frequency":49.8,
  "PV_in_voltage":83.8,
  "PV_in_current":14.0,
  "PV_in_watts":806.3,
  "PV_in_watthour":26.8754,
  "SCC_voltage":57.0200,
  "Load_pct":16,
  "Load_watt":767,
  "Load_watthour":25.5667,
  "Load_va":803,
  "Bus_voltage":393,
  "Heatsink_temperature":42,
  "Battery_capacity":100,
  "Battery_voltage":56.80,
  "Battery_charge_current":0,
  "Battery_discharge_current":0,
  "Load_status_on":1,
  "SCC_charge_on":1,
  "AC_charge_on":1,
  "Battery_recharge_voltage":44.0,
  "Battery_under_voltage":42.0,
  "Battery_bulk_voltage":58.4,
  "Battery_float_voltage":56.4,
  "Max_grid_charge_current":30,
  "Max_charge_current":60,
  "Out_source_priority":1,
  "Charger_source_priority":0,
  "Battery_redischarge_voltage":2582386249850678241749194600611840.0,
  "Warnings":"000000000000000000000000000000000000"
}
Mon May 24 13:22:06 2021 INVERTER: All queries complete, exiting loop.

Asher-02, 2021-05-28:
mine doing the same, started about the 13th May, was all working fine and just stopped, also able to do inverter poll with correct values, have tested mqtt and it is revieving data from another app

mskovdk, 2021-05-28:
Same here. Can't figure out what has gone wrong :(

I can't test with inverter_poller -d -1, as the files are not in my pi. Tried to CMake them, but get a "could not open file" error. Sigh. Feeling challenged...

mskovdk, 2021-05-29:
I found that an update of Mosquitto on Home Assistant, from version 5.1 to 6.0.1 had broken Mosquitto. I reverted to 5.1 from the Supervisor / Snapshots page in Home Assistant, and was up and running in no time. 
The reason may be illegal characters in the published data. Maybe a "/"...

RSd108, 2021-06-03:
If a valid mqtt clientid is sent through with the mosquitto_pub command then all works as expected

casio1964, 2021-06-03:
> If a valid mqtt clientid is sent through with the mosquitto_pub command then all works as expected

How to get valid clientid???)))

RSd108, 2021-06-03:
> 
> 
> > If a valid mqtt clientid is sent through with the mosquitto_pub command then all works as expected
> 
> How to get valid clientid???)))

I have created a pull request to fix this

Asher-02, 2021-06-04:
thanx very much, modified the files and rebuilt it and all is working, thx

Andries21, 2021-06-18:
I still can't get Home Assistant to receive any info. All I get in the mqtt logs are:
  New client connected from 192.168.1.190 as mosqpub/20673-7b6b98707 (p1, c1, k60, u'rct').
  Client mosqpub/20673-7b6b98707 disconnected.
  New connection from 192.168.1.190 on port 1883.
This repeats over and over. The system log gives me:
  21-06-18 07:56:50 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'rct'
  21-06-18 07:56:51 INFO (MainThread) [supervisor.auth] Successful login for 'rct'
And the "sudo docker exec -it voltronic-mqtt bash -c '/opt/inverter-cli/bin/inverter_poller -d -1'"
gives me the correct info. What am I going wrong?

casio1964, 2021-06-18:
Same here unfortunately (((

casio1964, 2021-06-20:
Damn, looks like no one not able to explain how to make this addon works with 6.0.1. Im real noob in mqtt, made all changes from pull request, and nothing changed.

casio1964, 2021-06-28:
https://drive.google.com/file/d/1DPPN4rnxWiBEpzLvPzgZDOR-W5qmp3GW/view?usp=sharing
Mosquitto 5.1 hassio image

polyar6688, 2021-07-01:
> https://drive.google.com/file/d/1DPPN4rnxWiBEpzLvPzgZDOR-W5qmp3GW/view?usp=sharing
> Mosquitto 5.1 hassio image

I had the same problem as you. Install the 5.1 again and at least, I receive the mqtt topics ... even so, it connects and disconnects me immediately to the home mqtt. Also, I can't place the client id
[image]

gadget78, 2021-07-16:
i forked this a while back, and had set it up a little different WITH a unique ID, so its easier to set up TWO inverter on same MQTT / location etc ... 
ive just updated my fork so it works too, didnt need to add much as already had it configured so it set itself up  better on MQTT server !... 
now going to try get it on docker, so its easier to deploy WITH-IN HomeAssistant or remotely using say portainer etc :)

ahmedalkhairy, 2022-04-08:
> Hi - any updates? Same problem like @Andries21 has. Getting logs but no values in dashboard:
> 
> 1633594750: Client voltronic_bd8041d0cdf131a6ba4e5b3360b8bc5a disconnected. 1633594750: New connection from 192.168.178.71 on port 1883. 1633594750: New client connected from 192.168.178.71 as voltronic_bd8041d0cdf131a6ba4e5b3360b8bc5a (p1, c1, k60, u'mqtt').
> 
> **Solution Update!** Had to set qpiri to 102 in inverter.conf !!

how do you reach the logs ?

amamoh, 2022-10-15:
[image]
Hello, I can read all data in Home Assistant dashboard but cannot send any command.
For example change source priority, I’ve tried manually send mqtt command and also not working.
Maybe it has something to do with MPP disconnecting instantly after making connection to Mosquitto.

garethmo, 2023-09-06:
> [image] Hello, I can read all data in Home Assistant dashboard but cannot send any command. For example change source priority, I’ve tried manually send mqtt command and also not working. Maybe it has something to do with MPP disconnecting instantly after making connection to Mosquitto.

did you figure out how to send the mqtt and get a response

TaariqSurtee, 2023-09-07:
Greetings everyone. Kudo's to Ned for everything he did here. However, I had endless issues with MQTT so I moved to ESPHOME. See here: https://esphome.io/components/pipsolar.html