ned-kelly/docker-voltronic-homeassistant #46: Remote serial port

proasnet, 2021-06-25:
Hi, your work see as very good idea and work. Now I have a Axpert inverter monitored by Watchopower in a virtual PC. As port, I am using Moxa nPort, that is serial tcp server.
Is possible use this network port with your docker?
In other project (Paradox Alarm Interface) is implemented communications via network serial port, so I think, that theoreticaly is possible add it a similar feature to your project.

gadget78, 2021-07-23:
this project is designed around MQTT protocol...
which HomeAssistant then picks up and uses...

so depends what you wanted to use as a client to display the information ...

proasnet, 2021-07-24:
Yes. I am understand it. But MQTT is comunication between your docker and home assistant. But how way you use for comunication with inverter? Axperts has usb and rs232 ports. Usb is not a practice for remote monitoring. The better way is a virtual serial device over ip, in my case moxa nPort or other as Advantech etc...
In my first post I member an other addon for home assistant for comunication with security panel with serial port ( Paradox Alarm Interface ) that has a very good idea for remote serial port, because HA as docker not support a remote serial devices. This addon can use a esp8266 as a telnet client and ip parameters are in addon config. All messages between security panel and addon is over telnet messages. This PAI addon sending and receiving MQTT messages as localhost to HA, because is hosted in HA. This is as example for good way to implement by HA not supported remote serial ports.
Please, can you see on it and if you not have support for remote serial port, then add this feature? Its nice.
Paradox Alarm Interface is on github too.
Thank you very much.

gadget78, 2021-07-24:
MQTT is not just a communication between docker and HA, 
it can be used remotely too... 
so with DOCKER on say a PI zero at the Inverter, 
the HomeAssistant server can be anywhere else, 
as the MQTT connection works over a network, and is already used my MANY IoT like Tasmota etc... 

the only benefit for your proposed serial over telnet, would be the advantage of the smaller esp8266 device....
there is a chance this could be rebuilt to work on a esp8266 device, as i know they support programs like MQTT as the Tasmota uses this network feature.

proasnet, 2021-07-25:
Yes I know Tasmota. I am using it in my home automation with HA. But wifi devices with Tasmota in ESP I have only for lights. Other technology devices is better connected by wired ethernet. Is any way use for example arduino uno with wiznet card? For comunication between your docker and RS232 on inverter? This arduino converter in this case supplies moxa serial device. I dont know what is easyer for you, if you will design it. Telnet or tcp serial...
Thank you