ned-kelly/docker-voltronic-homeassistant #47: MQTT Disconnected + Not Client ID

polyar6688, 2021-07-01:
Three problems:
1 - I cannot identify myself in the mqtt server, with the ClientID
[image]
[image]
[image]
[image]
![image](https://user-images.githubusercontent.com/49263906/12In theory, I do everything, modifying in mqtt.json, also in init, in push and in subscribe.
But connecting to the mqtt server from home, it gives me a random id4052154-418ded00-d9f4-11eb-80a8-0ccfa68e202c.png)

2 - Although it connects to the mqtt Server of the homeassistant, it continually disconnects it
[image]

3 - I can't get the mqtt send to be done every 5 seconds
[image]
[image]
[image]

Who can give me, from now on, very grateful

InsanityFlea, 2021-07-16:
I have the same issue, wont push topics, digging into it right now It's since version 6.0 of Mosquitto, version 5.1.1 worked perfectly.

InsanityFlea, 2021-07-16:
**Setting up the ACL fixed the issue.**

Access Control Lists (ACLs)

It is possible to restrict access to topics based upon the user logged in to Mosquitto. In this scenario, it is recommended to create individual users for each of your clients and create an appropriate ACL.

See the following links for more information:

    Mosquitto topic restrictions
    Mosquitto.conf man page

Add the following configuration to enable unrestricted access to all topics.

    Enable the customize flag

      customize:
        active: true
        folder: mosquitto

    Create /share/mosquitto/acl.conf with the contents:

    acl_file /share/mosquitto/accesscontrollist

    Create /share/mosquitto/accesscontrollist with the contents:

    user [YOUR_MQTT_USER]
    topic readwrite #

The /share folder can be accessed via SMB, or on the host filesystem under /usr/share/hassio/share.

GuyFace, 2021-08-08:
> **Setting up the ACL fixed the issue.**
>
> Access Control Lists (ACLs)
>
> It is possible to restrict access to topics based upon the user logged in to Mosquitto. In this scenario, it is recommended to [...]

I tried the acl route and still no luck...just keeps connecting and immediately after disconnects again..

amamoh, 2022-10-15:
Thx, this made my integration fully working.
I'll share your solution on HA forum.