ned-kelly/docker-voltronic-homeassistant #48: Not working with new 7.2KW Voltronic

InsanityFlea, 2021-07-15:
Sadly tried everything, seems something is off

InsanityFlea, 2021-07-15:
Thu Jul 15 22:15:46 2021 INVERTER: Current CRC: 49 C1
Thu Jul 15 22:15:49 2021 INVERTER: QMOD read timeout
Thu Jul 15 22:15:49 2021 INVERTER: QMOD reply too short (0 bytes)
Thu Jul 15 22:15:49 2021 INVERTER: Current CRC: B7 A9
Thu Jul 15 22:15:49 2021 INVERTER: QPIGS reply size (110 bytes)
Thu Jul 15 22:15:49 2021 INVERTER: QPIGS: 110 bytes read: (225.2 50.0 225.2 50.0 0675 0632 009 400 50.70 000 099 0023 00.0 000.0 00.00 00000 00010000 00 00 00000 010
Thu Jul 15 22:15:49 2021 INVERTER: QPIGS query finished
Thu Jul 15 22:15:49 2021 INVERTER: Current CRC: F8 54
Thu Jul 15 22:15:49 2021 INVERTER: QPIRI reply size (98 bytes)
Thu Jul 15 22:15:49 2021 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 31.3 230.0 50.0 31.3 7200 7200 48.0 46.0 46.0 53.2 53.2 3 30 040 0 1 3 9 01 0 0 54.0 0 1 4800000 010
Thu Jul 15 22:15:49 2021 INVERTER: Current CRC: B4 DA
Thu Jul 15 22:15:49 2021 INVERTER: QPIWS reply size (36 bytes)
Thu Jul 15 22:15:49 2021 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: (100000000000000000000000000000010000 48.0 46.0 46.0 53.2 53.2 3 30 040 0 1 3 9 01 0 0 54.0 0 1 4800000 010
Thu Jul 15 22:15:54 2021 INVERTER: Current CRC: 49 C1
Thu Jul 15 22:15:54 2021 INVERTER: QMOD reply size (5 bytes)
Thu Jul 15 22:15:54 2021 INVERTER: QMOD: 5 bytes read: (L
Thu Jul 15 22:15:54 2021 INVERTER: QMOD query finished

InsanityFlea, 2021-07-15:
Do I need to change the buffer sizes ?

dominicknoetze, 2021-07-16:
I had the same problem and changed over to usb (/dev/hidraw0)with the following settings in order to get it working. 
Qpiri=108
Qpiws=45
Qmod=5
Qpigs=110

closed 2021-07-16 by InsanityFlea