ned-kelly/docker-voltronic-homeassistant #49: Make a wiki so that we can share qpiri config

RHeynsZa, 2021-08-01:
As title

Couldnt find a place to share my Kodak 5kW King settings

```
# The following settings allow you to modify runtime buffers.
# N.B. These values may not be applicable to all inverter types, as such you will 
# need to run docker exec -it voltronic-mqtt bash -c '/opt/inverter-cli/bin/inverter_poller -d -1'
# or simply inverter_poller -d -1 to check for warnings or errors
# mentioned in https://github.com/ned-kelly/docker-voltronic-homeassistant/issues/5

# This allows you to modify the buffersize for the qpiri command
qpiri=104

# This allows you to modify the buffersize for the qpiws command
qpiws=40

# This allows you to modify the buffersize for the qmod command
qmod=5

# This allows you to modify the buffersize for the qpigs command
qpigs=110
```

kronna, 2022-11-28:
Mecer SOL-I-AX-3VP 3000VA/3000W

Working on below settings with qpiws commented out:

qpiri=97
#qpiws=67
qmod=5
qpigs=110

hanserasmus, 2022-12-15:
Axpert MKSII (https://me3energy.ng/axpert-mks-ii-5kw-48v)

qpiri=102
qpiws=36
qmod=5
qpigs=110

Running through device=/dev/hidraw0

netmilk, 2024-01-23:
Axpert King II 5KW

```
# This allows you to modify the buffersize for the qpiri command
qpiri=105

# This allows you to modify the buffersize for the qpiws command
qpiws=40

# This allows you to modify the buffersize for the qmod command
qmod=5

# This allows you to modify the buffersize for the qpigs command
qpigs=110

```