ned-kelly/docker-voltronic-homeassistant #50: Support for home-energy-management in newest HA update

gadget78, 2021-08-08:
In the newest HA update from 2021.8.4 they now have a dedicated ENERGY section..
more info here ... 
https://www.home-assistant.io/blog/2021/08/04/home-energy-management/

im sure this system already supports most of these things, but just needs configuring properly !..
has anyone else looked into this as yet ? before delve in ? 

ive only updated my fork couple weeks ago to support the newest MQTT update.. 
am now looking into getting it updated to support this,
and also have a easier way to installing this within HA (via the portainer Docker add-on ) so will document that too ... 

is there some need for these, or is there a better integration that people are using instead now ?

gadget78, 2021-08-11:
just to let people know.. i now have this working .. 
have updated my fork of it if people wanted to try it ? .. 
can issue PULL REQUEST  ? 

am going to update the readme/install notes on mine,  as i have found a MUCH easier way of installing/updating it, all WITHIN homeassistant
(all done it via the portainer add-on)

Waz-Cpt, 2021-08-24:
I would love to know the easiest way to move from nekelly's fork to your fork as I see that nedkelly has not been active since January.

gadget78, 2021-09-09:
depends on you current set up, you could trail it?.. 

if you HomeAssistant is running on the same setup where you have your inverter plugged into,
you can use the HomeAssistant portainer plugin, and create new docker image from mine
then disable current one, and enable mine, and see how you like it, can always go back if its not suited.... 
i will in time, redo the readme, to include a step by step tutorial if its needed ??

Waz-Cpt, 2021-09-09:
I got it working with some tweaks. I had some issues with your build.

I am struggling to understand what you did and how it integrates with the Power Meter integration.

I will need to learn a bit more and a friend said he will help teach me how to create a fork and have the ability to accept changes from other forks.

I was also trying to understand some of the logic behind your tweaks.

Need to do a proper deep dive into the code. I believe that I am getting a better understanding.

I am still struggling to figure out how to send proper raw commands to the invert to change some settings.

tanqt5041, 2021-10-09:
[image]
@gadget78 May I know how did u configure for this? My one always showing 0 for everything

vimepri, 2022-04-30:
> solo para que la gente sepa ... ahora tengo esto funcionando ... ¿he actualizado mi bifurcación si la gente quería probarlo? .. puede emitir PULL SOLICITUD?
> 
> voy a actualizar el archivo Léame/notas de instalación en el mío, ya que encontré una manera MUCHO más fácil de instalarlo/actualizarlo, todo DENTRO de homeassistant (todo hecho a través del complemento portainer)

hello, would you have a tutorial or a description of how to implement this in homeassistantOS?
I'm new to this and I'm interested in being able to do it from one place.

I have consulted your readme file, but I can't find the steps to follow to install it from portainer in HA