ned-kelly/docker-voltronic-homeassistant #51: Possible values for QPIRI and QPIWS: VOLTRONIC AXPERT INVERTER 3KW (2021)

EduardTheThird, 2021-08-12:
Struggling with finding the QPIRI and QPIWS values for my VOLTRONIC AXPERT INVERTER 3KW (2021).
I've tried multiple configs found in the other issues logged on the project.
Is there some list of possible options for what QPIRI and QPIWS could be? As blindly guessing might take some time to do ;-)

My inverter.conf

**qpiri=104
qpiws=67
qmod=5
qpigs=110**

```
Thu Aug 12 07:20:19 2021 INVERTER: Debug set

Thu Aug 12 07:20:19 2021 INVERTER: Current CRC: 49 C1
Thu Aug 12 07:20:19 2021 INVERTER: QMOD reply size (5 bytes)
Thu Aug 12 07:20:19 2021 INVERTER: QMOD: 5 bytes read: (L
Thu Aug 12 07:20:19 2021 INVERTER: QMOD query finished

Thu Aug 12 07:20:19 2021 INVERTER: Current CRC: B7 A9
Thu Aug 12 07:20:20 2021 INVERTER: QPIGS reply size (110 bytes)
Thu Aug 12 07:20:20 2021 INVERTER: QPIGS: 110 bytes read: (238.0 49.9 238.0 49.9 0382 0263 013 476 26.98 000 100 0620 0000 000.0 00.00 00000 10011101 00 04 00000 101
Thu Aug 12 07:20:20 2021 INVERTER: QPIGS query finished

Thu Aug 12 07:20:20 2021 INVERTER: Current CRC: F8 54
Thu Aug 12 07:20:20 2021 INVERTER: QPIRI reply size (104 bytes)
Thu Aug 12 07:20:20 2021 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 13.0 230.0 50.0 13.0 3000 3000 24.0 23.0 21.0 28.2 27.0 0 30 60 1 0 0 - 01 1 0 27.0 0 0qI

Thu Aug 12 07:20:20 2021 INVERTER: Current CRC: B4 DA
Thu Aug 12 07:20:23 2021 INVERTER: QPIWS read timeout
Thu Aug 12 07:20:23 2021 INVERTER: QPIWS reply too short (48 bytes)
```

GastonJ, 2022-02-13:
Try

run_interval=120
amperage_factor=1.0
watt_factor=1.01
qpiri=102
qpiws=38
qmod=5
qpigs=110

Works for mine. Next thing I'm after is how to read both MPPT's

HTH

snormonster, 2022-03-31:
@EduardTheThird Not sure if you had success or not, but the following worked for me

qpiri=97
qmod=5
qpigs=110
#qpiws=40

EduardTheThird, 2022-05-23:
Thanks for the suggestions. 
Closing issues.

closed 2022-05-23 by EduardTheThird