ned-kelly/docker-voltronic-homeassistant #57: No data from ttyUSB0 at all?

UV-PWRD, 2021-11-05:
I've managed to get this running, I think (RPi4, Debian, Supervised HA install, Portainer stand alone to manage docker, not the HA plug in). 

I can add entities the to dashboard, but they never have any value, always unknown. I assume that means the container is running correctly and talking to the MQTT broker at least?

When I try and query the inverter I am getting nothing back at all. 

docker exec -it voltronic-mqtt bash -c '/opt/inverter-cli/bin/inverter_poller -d -1'

Fri Nov  5 03:21:51 2021 INVERTER: Debug set
Fri Nov  5 03:21:51 2021 INVERTER: Current CRC: 49 C1
Fri Nov  5 03:21:54 2021 INVERTER: QMOD read timeout
Fri Nov  5 03:21:54 2021 INVERTER: QMOD reply too short (0 bytes)
Fri Nov  5 03:21:54 2021 INVERTER: Current CRC: B7 A9
Fri Nov  5 03:21:57 2021 INVERTER: QPIGS read timeout
Fri Nov  5 03:21:57 2021 INVERTER: QPIGS reply too short (0 bytes)
Fri Nov  5 03:21:57 2021 INVERTER: Current CRC: F8 54
Fri Nov  5 03:22:00 2021 INVERTER: QPIRI read timeout
Fri Nov  5 03:22:00 2021 INVERTER: QPIRI reply too short (0 bytes)
Fri Nov  5 03:22:00 2021 INVERTER: Current CRC: B4 DA

I grabbed tio also and it says connected but I never seen any send/receive when checking statistics. 

I did just grab a $10 usb/serial converter off eBay, is it worth be grabbing a better one? 

What else can I check to confirm that's the issue.

jpmeijers, 2021-11-05:
I have the same problem.

When I use inverter_poller from https://github.com/manio/skymax-demo it works fine. Same config with inverter_poller from this repo fails with no data read timeout.

UV-PWRD, 2021-11-22:
I am now seeing data within minicom and TIO after replacing the USB/RS232 adaptor, I can also use the windows App, so for me the no data issue was hardware.

1onord, 2022-01-12:
Hi,

i have the same Problem, can you Post a Link which Hardware i have to buy ?

Snubbeke, 2022-09-17:
In my case all voltronic entities are present in HA (latest version) but there are no values (updated). In the Log Viewer I can see that sensors are updated but without data. 
[image]

I have RPI4 with docker /portainer (all latest version) with following compose files running. No errors except "No log line matching the '' filter" in the log. inverter.conf has device=/dev/ttyUSB0 as well as the compose file. 
Mqtt is working fine because I see topics "homeassistant/sensor/voltronic*/config being updated. Strange enough the topics from the log viewer are not in the topics collector (I use MQTT.fx).
The (Mecer) Axpert 3KVA has an RJ45 connection thus I use RJ45 --> Serial --> USB and with WatchPower I see the data coming in.
Any tips?

peter97silva, 2022-10-04:
> I have the same issue with an usb 2 serial i have a Voltronic AXPERT MKS IV 5.6KW
> 
> pi@raspberrypi:~/ha-inverter-mqtt-agent/manual $ docker exec -it voltronic-mqtt bash -c '/opt/inverter-cli/bin/inverter_poller -d -1' Fri Apr 1 14:58:35 2022 INVERTER: Debug set Fri Apr 1 14:58:35 2022 INVERTER: Current CRC: 49 C1 Fri Apr 1 14:58:36 2022 INVERTER: QMOD reply size (5 bytes) configuration options for the actual inverter polling process...es. Buffer: (NAKs▒
> 
> # The device to read from...
> # Use: /dev/ttyS0 if you have a serial device,
> # /dev/ttyUSB0 if a USB<>Serial,
> # /dev/hidraw0 if you're connecting via the USB port on the inverter.
> device=/dev/ttyUSB0 #device=/dev/serial1 #device=/dev/hidraw0
> 
> # How many times per hour is the program going to run...
> # This is used to calculate the PV & Load Watt Hours between runs...
> # If unsure, leave as default - it will run every minute...
> # (120 = every 30 seconds)...
> run_interval=120
> 
> # This allows you to modify the amperage in case the inverter is giving an incorrect
> # reading compared to measurement tools. Normally this will remain '1'
> amperage_factor=1.0
> 
> # This allows you to modify the wattage in case the inverter is giving an incorrect
> # reading compared to measurement tools. Normally this will remain '1'
> watt_factor=1.01
> 
> # The following settings allow you to modify runtime buffers.
> # N.B. These values may not be applicable to al
> Fri Apr 1 14:58:36 2022 INVERTER: Current CRC: B7 A9 Fri Apr 1 14:58:39 2022 INVERTER: QPIGS read timeout Fri Apr 1 14:58:40 2022 INVERTER: QPIGS reply too short (7 bytes) Fri Apr 1 14:58:40 2022 INVERTER: Current CRC: F8 54 Fri Apr 1 14:58:43 2022 INVERTER: QPIRI read timeout Fri Apr 1 14:58:44 2022 INVERTER: QPIRI reply too short (7 bytes) Fri Apr 1 14:58:44 2022 INVERTER: Current CRC: B4 DA Fri Apr 1 14:58:47 2022 INVERTER: QPIWS read timeout Fri Apr 1 14:58:48 2022 INVERTER: QPIWS reply too short (7 bytes) Fri Apr 1 14:58:53 2022 INVERTER: Current CRC: 49 C1 Fri Apr 1 14:58:53 2022 INVERTER: QMOD reply size (5 bytes) configuration options for the actual inverter polling process...tes. Buffer: (NAKs▒

I have exactly this same happening to me. No values sent by mqtt. I think the orange pi is having trouble to communicate with these new inverters. ANyone has got a solution for this? 
Ive got the new voltronic 5.6kw inverter
Thanks a lot!!

byte4geek, 2022-10-05:
did you tried to set com port manually before start the docker?
see this tread https://community.home-assistant.io/t/programmatically-read-data-from-your-solar-inverter-voltronic-axpert-mppsolar-pip-voltacon-effekta-etc-and-interface-with-home-assistant-via-mqtt-works-with-rs232-usb/119053/499?u=byte4geek
and try to use catalinbordan fork:
https://github.com/catalinbordan/docker-voltronic-homeassistant

Snubbeke, 2022-10-10:
> > did you tried to set com port manually before start the docker? see this tread https://community.home-assistant.io/t/programmatically-read-data-from-your-solar-inverter-voltronic-axpert-mppsolar-pip-voltacon-effekta-etc-and-interface-with-home-assistant-via-mqtt-works-with-rs232-usb/119053/499?u=byte4geek and try to use catalinbordan fork: https://github.com/catalinbordan/docker-voltronic-homeassistant
> 
> Thanks a lot for your help. I havent seen the catalinbordan fork before. I just installed it as its in the readme :
> 
> ```
>    changed to serial usb adapter in the   inverter.conf  
>    changed the ip, username and password in the mqtt.json, the red circle only (maybe now i see that it is mistaken, the influx should be the orangepi address no? and the real mqtt server ip, username and password should be the other ones on the top of the file?) 
> ```
> 
> but now i get this QMOD READ TIMEOUT instead.
> 
> Tomorrow i will have to do more testing
> 
> [image]

Meanwhile I got it working by changing "timeout 60 /opt/inverter-cli/bin/inverter_poller -1" in /opt/ha-inverter-mqtt-agent/sources/inverter-mqtt/mqtt-push.sh 
In case the RPi goes down (I am in South Africa with power outages) I do have to run sudo stty -F /dev/ttyUSB0 sane
sudo stty -F /dev/ttyUSB0 2400 raw -echo cat /dev/ttyUSB0 and wait until data comes through. After this I have to restart the container and with some luck, it picks up the data and everything works well.

Also I tried uncommenting stty -F /dev/ttyUSB0 2400 raw in /opt/ha-inverter-mqtt-agent/sources/inverter-mqtt/entrypoint.sh and adding the other commands but that did not help.
I will have a look at the fork as well.
THANKS!

peter97silva, 2022-10-10:
Well I'm very happy. I just changed the connection, instead of using the serial to USB cable I used the mini USB connection on the inverter (and changed in the code) and it is working since 3 days!!

llbster, 2022-10-15:
Hi all but especially @peter97silva , I am having the same/similar issue to you (I think) I am running the voltronic-mqtt on Ubuntu in Docker and connecting to my inverter using the micro USB. I have set  the mode to /dev/hidraw0 in the config file. I am seeing the MQTT values in my HA Dashboard, but they are all 'unknown'.

When you say you 'changed the code' what did you have to do to get the values to show up?

Do you have any debug / trouble shooting advice that I can follow to get to the bottom of what is going wrong.
I have a feeling that the Linux box is not reading from the USB port, when I look in /dev I do not see any hidraw devices.

Any tips very much appreciated! :)

peter97silva, 2022-11-21:
> Hi all but especially @peter97silva , I am having the same/similar issue to you (I think) I am running the voltronic-mqtt on Ubuntu in Docker and connecting to my inverter using the micro USB. I have set the mode to /dev/hidraw0 in the config file. I am seeing the MQTT values in my HA Dashboard, but they are all 'unknown'.
> 
> When you say you 'changed the code' what did you have to do to get the values to show up?
> 
> Do you have any debug / trouble shooting advice that I can follow to get to the bottom of what is going wrong. I have a feeling that the Linux box is not reading from the USB port, when I look in /dev I do not see any hidraw devices.
> 
> Any tips very much appreciated! :)
> 
> Edit: when I execute the CLI command: bin/inverter_poller -d -1
> 
> I see only the following:
> 
> # bin/inverter_poller -d -1
> Sat Oct 15 17:55:51 2022 INVERTER: Debug set Sat Oct 15 17:55:51 2022 INVERTER: Unable to open device file (errno=2 No such file or directory) Sat Oct 15 17:55:56 2022 INVERTER: Unable to open device file (errno=2 No such file or directory)

Hey @llbster ,
I've had a problem with HA so I've had this off for quite some time. Now when i got back to this I suddenly have the same issue as you running the command to test. I havent found a solution yet. Let me know if you did!
Thanks

[image]

llbster, 2022-12-03:
I have not made any progress on this.
I ordered a USB To RJ45 Console Cable (RS232 Serial Adapter) and will try this instead of the USB. 
https://www.aliexpress.com/item/1005004651817501.html

When I was trying to get the USB to work, I could see the cable being connected, but the Invertor was not being recognised as a hidraw device. Hence me trying the serial port instead. 

I will not be able to test again until the new year (maybe Feb) but will update when I do.

frederickjh, 2023-01-12:
@llbster ]wrote:
> ``` bash
> $ bin/inverter_poller -d -1
> 
> Sat Oct 15 17:55:51 2022 INVERTER: Debug set 
> Sat Oct 15 17:55:51 2022 INVERTER: Unable to open device file (errno=2 No such file or directory) 
> Sat Oct 15 17:55:56 2022 INVERTER: Unable to open device file (errno=2 No such file or directory)
> ```

@llbster I think you first need to figure out what device the inverter is on. You can do this by opening a terminal on the computer you want to connect to it. First do this with the inverter not plugged into the computer:

```
ls /dev/
```
Then plug in the inverter and run that command again. Compare the results to find the difference and the device that has been added. This will be your inverter. Thenw adjust your configuration and test.

slash-bit, 2023-01-13:
@llbster Also try to run the command above as sudo

llbster, 2023-01-15:
@slash-bit, @frederickjh - will do thanks. I am remote at the moment, will be able to plug in the new cable and test in about a week. I will update this thread then. Cheers.

llbster, 2023-01-20:
Hi, 
One step forwards perhaps.... 
I have connected the new USB to RS232 (RJ45) cable and I now see the output below when I run:

inverter_poller -d -1

Fri Jan 20 12:19:13 2023 INVERTER: Debug set
Fri Jan 20 12:19:13 2023 INVERTER: Current CRC: 49 C1
Fri Jan 20 12:19:16 2023 INVERTER: QMOD read timeout
Fri Jan 20 12:19:48 2023 INVERTER: QMOD reply too short (0 bytes)
Fri Jan 20 12:19:48 2023 INVERTER: Current CRC: B7 A9
Fri Jan 20 12:19:51 2023 INVERTER: QPIGS read timeout
Fri Jan 20 12:20:23 2023 INVERTER: QPIGS reply too short (0 bytes)
Fri Jan 20 12:20:23 2023 INVERTER: Current CRC: F8 54
Fri Jan 20 12:20:26 2023 INVERTER: QPIRI read timeout
Fri Jan 20 12:20:57 2023 INVERTER: QPIRI reply too short (0 bytes)
Fri Jan 20 12:20:57 2023 INVERTER: Current CRC: B4 DA
Fri Jan 20 12:21:00 2023 INVERTER: QPIWS read timeout
Fri Jan 20 12:21:32 2023 INVERTER: QPIWS reply too short (0 bytes)

The first time I ran the inverter poller I also saw a value of qpiri as 97 returned from the Inverter (I think?) I changed qpri to this number but there was no change.

Am I now connected to the inverter? the 0 Bytes debug seems a bit suspicious? 

Any tips on next steps I can take?

Many thanks! 
@frederickjh

slash-bit, 2023-01-20:
One thing that got me was that if the Inverter is in deep standby mode it doesn't respond or sometimes sends incomplete data. May be ensure that iits fully running , with inverter button on.

llbster, 2023-01-20:
I just rebooted the inverter... same thing... I plugged the same RJ45 into the Battery Management Port on the inverter and got the same response (0 Bytes etc) so I suspect the inverter is not sending anything... 

Very frustrating :(

frederickjh, 2023-01-20:
@llbster check permissions on the USBtty device. Is your user in the group the device belongs to and are the groups permission correct?

llbster, 2023-01-22:
@frederickjh  I checked and inverter_poller is running as root. Just to be sure I temporarily set the USB permissions to read/write all... still not luck, empty responses.

I am now suspecting that it is the cable that is the issue. It is a USB -> RJ45 serial cable.  Bought 2 from AliExpress, one does not seem to work at all, the other shows up put with no data coming from the Inverter. In fact, I get the same errors in the Inverter_poller output regardless of whether the RJ45 end is plugged into the Inverter.

I have asked in another thread, but do you -- OR ANYBODY! ;) --  have tips for which cable to use?

llbster, 2023-01-24:
IT WORKS!!! :)

For anybody else with similar issues, there are many moving parts and lots to go wrong. Here is a summary of my issue and what eventually worked:

**No data returned from the Inverter along USB cable to a linux box** 
Using the _'hidraw0'  device=/dev/hidraw0_ config I was not getting any data from the inverter.

I decided that this might be a USB issue so I invested in an RJ45->USB Serial cable from AliExpress so that I could use the _device=/dev/ttyUSB0_ config entry instead.
**No data was returned from the serial to USB cable**. 
I am not sure if I bought the correct cable, it was tabled a Cisco compatible console cable, but although the inverter_poller seemed to be sending requests, there was never a response.

I went back to basics and plugged a USB C-micro USB cable between my Mac laptop and WatchPower.
**No data was returned.** (The cable was tested and functional)

At this point I was sure the Inverter was not sending data, but for a Hail Mary, I plugged a USB dock into the Mac laptop and used a USB2 to micro USB cable.... **and it worked!**

This was the first cable I used between the inverter and the linux box, so I set up the original setup and followed the advice of @frederickjh and checked the permissions on the '_hidraw0_' device and finally I was seeing data in Homeassistant and the inverter_poller -d 1 debug. Many thanks to everybody on this forum! So happy I got this working!

jpmeijers, 2023-01-24:
> Using the 'hidraw0' device=/dev/hidraw0 config I was not getting any data from the inverter.

My inverter is mounted as hidraw2 when I plug USB into it. You need to make sure with `dmesg` or `ls /dev/hidraw*` to see which one appears when the inverter is plugged in, and use that in the configs. Most likely the number of the hidraw device changes when you add a hub to the mix.