ned-kelly/docker-voltronic-homeassistant #59: Support of P17 InfiniSolar 5KW protocol

blacksocket, 2021-11-13:
Hello,

Will this software also support P17 protocol for 5KW Hybrid Inverters ?

[PI17_InfiniSolar-5KW-protocol-20160309.pdf](https://github.com/gch1p/inverter-tools/files/6612507/PI17_InfiniSolar-5KW-protocol-20160309.pdf)

vknmakes, 2023-07-07:
Hmm; I have what I believe is something based on this and mine is a P18, not P17:
https://github.com/jblance/mpp-solar/blob/master/docs/protocols/PI18_InfiniSolar-V-protocol-20170926.pdf

sudo ./inverterctl --device usb --verbose --raw "^P003PI"
voltronic::Device::send(): writing 10 bytes
0x0000: 5e 50 30 30 33 50 49 c3 2b 0d                    ^P003PI.+.
voltronic::USBDevice::write(): dataSize=10, writeSize=8
0x0000: 5e 50 30 30 33 50 49 c3 2b 0d                    ^P003PI.+.
voltronic::Device::writeLoop(): bytesWritten=8
voltronic::USBDevice::write(): dataSize=2, writeSize=2
0x0000: 2b 0d                                            +.
voltronic::Device::writeLoop(): bytesWritten=8
voltronic::Device::readLoop(): bytesRead=8
voltronic::Device::recv(): got 5 bytes
0x0000: 5e 30 1b e3 0d                                   ^0...
0x0000: 5e 30                                            ^0
^0

Spick79, 2024-12-18:
Al meu e un Infinisolar e5.5kw și cred ca p16 oare merge?