ned-kelly/docker-voltronic-homeassistant #60: 7.2 Voltronic Max with x2 MPPT

Sarelseekat, 2021-11-16:
Hello

I've managed to connect to my Voltronic 7.2, but then it doesn't seem to pull telemetry from the second Mppt. For some reason it also doesn't display the Watt output for the Mppt it does display, but that can be solved by a minor calculation in Home Assistant. Also not sure what's happening with the battery voltage, but need to sort out one thing at a time.

[image]

[image]

Does anyone know how I can get this fixed?

Sarelseekat, 2021-11-16:
Mqtt push looks like this

[image]

UV-PWRD, 2021-11-22:
I also have MPP 7.2kw, can you please share your inverter.conf settings?

Sarelseekat, 2021-11-26:
Hi UV - Sorry for the delay - It's been a nightmare week. Please see my config below.

device=/dev/hidraw0

run_interval=120

amperage_factor=1.0

watt_factor=1

# This allows you to modify the buffersize for the qpiri command
qpiri=108

# This allows you to modify the buffersize for the qpiws command
qpiws=45

# This allows you to modify the buffersize for the qmod command
qmod=5

# This allows you to modify the buffersize for the qpigs command
qpigs=110

Sarelseekat, 2021-11-26:
When I run the inverter poller I get the following.

Fri Nov 26 08:21:14 2021 INVERTER: Debug set
Fri Nov 26 08:21:14 2021 INVERTER: Current CRC: 49 C1
Fri Nov 26 08:21:14 2021 INVERTER: QMOD reply size (5 bytes)
Fri Nov 26 08:21:14 2021 INVERTER: QMOD: 5 bytes read: (B
Fri Nov 26 08:21:14 2021 INVERTER: QMOD query finished
Fri Nov 26 08:21:14 2021 INVERTER: Current CRC: B7 A9
Fri Nov 26 08:21:14 2021 INVERTER: QPIGS reply size (110 bytes)
Fri Nov 26 08:21:14 2021 INVERTER: QPIGS: 110 bytes read: (000.0 00.0 230.1 50.0 0184 0110 002 441 53.50 006                                                                 093 0044 00.6 345.5 00.00 00000 00010110 00 00 00235 010
Fri Nov 26 08:21:14 2021 INVERTER: QPIGS query finished
Fri Nov 26 08:21:14 2021 INVERTER: Current CRC: F8 54
Fri Nov 26 08:21:15 2021 INVERTER: QPIRI reply size (108 bytes)
Fri Nov 26 08:21:15 2021 INVERTER: QPIRI: 108 bytes read: (230.0 31.3 230.0 50.0 31.3 7200 7200 48.0 47.0 44                                                                .0 56.4 53.5 2 30 080 1 1 2 9 01 0 1 56.0 0 1 480 0 000
Fri Nov 26 08:21:15 2021 INVERTER: QPIRI query finished
Fri Nov 26 08:21:15 2021 INVERTER: Current CRC: B4 DA
INVERTER: ampfactor from config is 1.00
INVERTER: wattfactor from config is 1.00
{
  "Inverter_mode":4,
  "AC_grid_voltage":0.0,
  "AC_grid_frequency":0.0,
  "AC_out_voltage":230.1,
  "AC_out_frequency":50.0,
  "PV_in_voltage":345.5,
  "PV_in_current":0.6,
  "PV_in_watts":0.0,
  "PV_in_watthour":0.0000,
  "SCC_voltage":0.0000,
  "Load_pct":2,
  "Load_watt":110,
  "Load_watthour":3.6667,
  "Load_va":184,
  "Bus_voltage":441,
  "Heatsink_temperature":44,
  "Battery_capacity":93,
  "Battery_voltage":53.50,
  "Battery_charge_current":6,
  "Battery_discharge_current":0,
  "Load_status_on":1,
  "SCC_charge_on":1,
  "AC_charge_on":0,
  "Battery_recharge_voltage":47.0,
  "Battery_under_voltage":44.0,
  "Battery_bulk_voltage":56.4,
  "Battery_float_voltage":53.5,
  "Max_grid_charge_current":30,
  "Max_charge_current":80,
  "Out_source_priority":1,
  "Charger_source_priority":2,
  "Battery_redischarge_voltage":2482241852432648119030955047387136.0,
  "Warnings":""
}
Fri Nov 26 08:21:16 2021 INVERTER: All queries complete, exiting loop.
-d: line 1: docker: command not found

UV-PWRD, 2021-11-26:
Many thanks, I will test later and report back.

Sarelseekat, 2021-11-26:
Awesome

UV-PWRD, 2021-11-26:
ok so that worked, and I am ecstatic and extremely appreciative.

May I please ask how you worked out those buffer values? 108 and 45 that is.

Sarelseekat, 2021-11-26:
I didn't work it out. Not sure where, but I got it from somewhere online.

UV-PWRD, 2021-11-26:
Well I am very grateful. 

Is there something I can check in my setup to help you? 

I have 2x mppt also, do you want me to see if there's data on both or something?

Sarelseekat, 2021-11-26:
If you could - then yes please. Also check which firmware version you're running?

UV-PWRD, 2021-11-26:
The WiFi Module firmware is 3.0.9.1

I have no PV data as it's dark here so I only have battery data and inverter output data, tomorrow I will have history and can chaeck PV output watts/amps reported in HA vs what i see in SmartESS/WatchPower to tell if i am seeing both arrays. 

What other data can I get for you currently?

UV-PWRD, 2021-11-26:
Any idea why the run interval doesn't seem to be impacted by the inverter.conf?

MY updates every 30 seconds regardless of what I have it set to, I assume i have read it right and higher run interval means more times per hour.

Does the container have to be restarted for that to take effect?

Sarelseekat, 2021-11-26:
I'm still trying to figure it out. I know that changing this setting does little to the number of times per minute you get telemetry pulling to the HASS. There is another place where you can change the settings, but I haven't really seen that work ether

UV-PWRD, 2021-11-26:
Yes there are clearly issues with data from both MPPTs

I am only seeing 1 arrays output current, 0 PV output in watts, even though battery is charging etc.

Sarelseekat, 2021-11-27:
I Get the same - At least it's easy getting the watts - VxA=W
My two strings are the same size with no shading so I make the Total PV Wx2

Sarelseekat, 2021-11-27:
@ned-kelly Maybe you know of a way we can get this sorted?

UV-PWRD, 2021-11-27:
> 

Problem is I only see amps for 1 array. I have 2, same size, some shading but optimisers on everything.

tascaV, 2021-12-25:
I couldn't find 2'nd mptt data in output string. In de-compiled watchsolar app wasn't any reference of it as well, maybe I haven't tried hard enough. Anyway, in code the array output in watts is calculated from: `pv_input_watts = (scc_voltage * pv_input_current) * wattfactor`, for Voltronic Max 7.2 _scc_voltage_  is always 00.00. So basically you have 2 options, either change this line in your sources: `pv_input_watts = ((pv_input_voltage * pv_input_current) * wattfactor) * 2` OR compile data with Grafana. Hope this helps

UV-PWRD, 2021-12-25:
> I couldn't find 2'nd mptt data in output string. In de-compiled watchsolar app wasn't any reference of it as well, maybe I haven't tried hard enough. Anyway, in code the array output in watts is calculated from: `pv_input_watts = (scc_voltage * pv_input_current) * wattfactor`, for Voltronic Max 7.2 _scc_voltage_ is always 00.00. So basically you have 2 options, either change this line in your sources: `pv_input_watts = ((pv_input_voltage * pv_input_current) * wattfactor) * 2)` OR compile data with Grafana. Hope this helps I tested this with both mptt's connected and correlated results with solarpower output. In case of only 1 connected array the above math might be totally wrong.

Multiplying by 2 won't be correct for me as my arrays have different aspects.

Does Grafana allow anything different?

When using the windows Watchpower program, I canxsee total PV generation from both arrays but have no way of getting that data in.

Sarelseekat, 2021-12-26:
> I couldn't find 2'nd mptt data in output string. In de-compiled watchsolar app wasn't any reference of it as well, maybe I haven't tried hard enough. Anyway, in code the array output in watts is [...]

It might be overengineering, but with different String sizes on the two MPPT, using a Voltage and current meter on an ESP32, might be the only option. Not sure which sensor to use yet as many are only calibrated for AC. 

Maybe someone will find a way to pull this data directly from the inverter in the future or the manufactures will include this data in the output at some point.

tascaV, 2021-12-26:
My guess is that inverter returns mean() for mppt's volts & amps. If this assumption is true you should be fine multiplying by 2. Try this query with Influx DB explore dashboard: `SELECT ("pv_input_voltage_1" * "pv_input_current_for_battery") * 2 * 1.01 AS "pw_power_calculated" FROM "solar"."autogen"."solar" WHERE time > :dashboardTime: AND time < :upperDashboardTime: AND "device"='voltronic'` , set dashboard time to _Past 24h_. Crosscheck with solarpower/watchpower output. If readings are off by a little bit, take in consideration _wattfactor_ value = 1.01

Sarelseekat, 2021-12-27:
Really appreciate your input Tasca - It's a good idea to check the output is the mean root of the two MPPT's. I will test in a couple of days and reply here.

TheRoarman, 2022-01-03:
Hey guys, With my Kodak VM II there were extra output that included information such as PV_input_watt. I'm not sure if you have double checked the output from the inverter_cli yet to see what output is being produced. I've had to modify the cpp code to take the extra output into consideration. Maybe the same would apply to the extra MPPT charger.
I've been meaning to fork this project for a while but I just don't have the time. @Gadget78 and @shampeon have done some great additions to Ned-kelly's work. Eventually I would like to help them by adding in my code sometime for the VM II units.
Also with the change of the MQTT broker and the Energy metering specs change on HA, i need to update my code before I can upgrade HA to the latest verion of MQTT broker.

Enjoy.
P.S.
One of the lines of code that i changed to pick up the PV input watts straight from the inverter without trying use any calculations.
filename: main.cpp

sscanf(reply1->c_str(), "%f %f %f %f %d %d %d %d %f %d %d %d %f %f %f %d %s %d %d %d", &voltage_grid, &freq_grid, &voltage_out, &freq_out, &load_va, &load_watt, &load_percent, &voltage_bus, &voltage_batt, &batt_charge_current, &batt_capacity, &temp_heatsink, &pv_input_current, &pv_input_voltage, &scc_voltage, &batt_discharge_current, &device_status, &dummy1, &dummy2, &pv_input_watt);

jraydoran, 2022-02-04:
Hi Sarelseekat,

I would like to know if you have resolved the issue addressed above without doing extra calcs. I seem to be having the exact problem but have not managed to find a solution.

Thanks for the inverter setting posted above saved me days of work.

1) Battery Voltage Issue.
2) PV Input.

Appreciate to time and effort in advance?

Kind Regards
Jason

jraydoran, 2022-02-05:
I all have also found that

"load_watthour" topic value goes up and down. I would expect it to be a counter value going up and resetting at some point.

Sarelseekat, 2022-02-06:
Hi Jason - No I haven't yet. My supplier gave me new firmware that I'll try loading this week. Might solve the problem. Will post back on the group if it's the case.

jraydoran, 2022-02-06:
Thanks for you feedback.
Appreciated so much!

jraydoran, 2022-02-07:
May I ask if you have made any additional changes to your installation for the 7.2 Max?

The only changes I have made was to you use your settings in the inverter.config file

Do we need to apply the changes listing in Issue 40 as it is not clear if it is required for our model.

Once again appreciate your time and effort to help.

Kind Regards
Jason

UV-PWRD, 2022-03-03:
> Hey guys, With my Kodak VM II there were extra output that included information such as PV_input_watt. I'm not sure if you have double checked the output from the inverter_cli yet to see what output [...]

I tried this edit above and still no PV in watts for me, also tried changing the line where it calculates it and still nothing.

Sarelseekat, 2022-03-04:
Same here. I've resorted to using an ACS758 to measure the PV output from the second string. Can't find any other way around this.

LeafwaysWorks, 2022-03-18:
for changing the update interval , you have to change the entrypoint.sh file and change the 30 thats for the mqtt push to other number in sec, then you have to commit the docker container to the image or another image and load that in the compose file

LeafwaysWorks, 2022-03-18:
for the wattage also not working for me, a workaround inside home assisstant is:
sensor:
- platform: template
    sensors:
        homemaxinverter_pv_in_wattage:
            friendly_name: "HomeMaxInverter PV in wattage"
            unit_of_measurement: 'W'
            value_template: '{{ states("sensor.homemaxinverter_pv_in_voltage") |float * states("sensor.homemaxinverter_pv_in_current") | float }}'

nuvolare12, 2022-10-23:
Hello, sorry i'm new on home assitant, and i have a Voltronic MK IV 5.6, and i wish to connect it to my home assitant, from where i've to start? thnk you