ned-kelly/docker-voltronic-homeassistant #62: can it support phocos inverters?

test2a, 2022-01-09:
i own a phocos anygrid solar inverter that has a built in mppt solar charge controller. it has an rs-232 interface which i hope i can use woth a raspbery pi to get data into home assistant.
https://www.phocos.com/product/psw-h-hybrid-inverter-charger/

https://web.archive.org/web/20200919221625/https://www.phocos.com/wp-content/uploads/2019/10/Any-Grid_PSW-H_EN-manual_2020-07-29.pdf

can this docker support the inverter? is there anything special i need to check ?

ned-kelly, 2022-08-25:
Hi there,

Sorry you will need to test this one yourself - How did you go, any luck?

Cheers,

closed 2022-08-25 by ned-kelly

test2a, 2022-08-25:
@ned-kelly nah, i found solaranzeige.de and am using that.