ned-kelly/docker-voltronic-homeassistant #63: Help with setting up the right Values

1onord, 2022-01-13:
Hi all,

i setting up the Container, and MQTT, all works as aspected, but no values are send.

[image]

with qpiri 98:

Thu Jan 13 06:01:28 2022 INVERTER: Debug set
Thu Jan 13 06:01:29 2022 INVERTER: Current CRC: 49 C1
Thu Jan 13 06:01:32 2022 INVERTER: QMOD read timeout
Thu Jan 13 06:01:32 2022 INVERTER: QMOD reply too short (0 bytes)
Thu Jan 13 06:01:32 2022 INVERTER: Current CRC: B7 A9
Thu Jan 13 06:01:35 2022 INVERTER: QPIGS read timeout
Thu Jan 13 06:01:35 2022 INVERTER: QPIGS reply too short (0 bytes)
Thu Jan 13 06:01:35 2022 INVERTER: Current CRC: F8 54
Thu Jan 13 06:01:38 2022 INVERTER: QPIRI read timeout
Thu Jan 13 06:01:39 2022 INVERTER: QPIRI reply too short (0 bytes)

with 102:

Thu Jan 13 06:29:09 2022 INVERTER: Current CRC: F8 54
Thu Jan 13 06:29:11 2022 INVERTER: QPIRI reply size (102 bytes)
Thu Jan 13 06:29:11 2022 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 21.7 230.0 50.0 21.7 5000 4000 48.0 48.0 47.5 53.2 53.2 2 30 030 0 0 2 6 01 0 0 51.0 0 1�
(NAK0 110�&

Thu Jan 13 06:29:11 2022 INVERTER: Current CRC: B4 DA
Thu Jan 13 06:29:12 2022 INVERTER: QPIWS reply size (36 bytes)
Thu Jan 13 06:29:12 2022 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: (00000000000000000000000000000000��
0 48.0 48.0 47.5 53.2 53.2 2 30 030 0 0 2 6 01 0 0 51.0 0 1�
(NAK0 110�&

Thu Jan 13 06:29:17 2022 INVERTER: Current CRC: 49 C1
Thu Jan 13 06:29:18 2022 INVERTER: QMOD reply size (5 bytes)
Thu Jan 13 06:29:18 2022 INVERTER: QMOD: incorrect start/stop bytes.  Buffer: (L
00000000000000000000000000��
0 48.0 48.0 47.5 53.2 53.2 2 30 030 0 0 2 6 01 0 0 51.0 0 1�
(NAK0 110�&

Inverter = Effekta AX K 5000 48V / 2,4KWp-PV / AX-Serie 5KVA Last / Hybrid

Adapter = https://www.amazon.de/VTOP-USB-RS232-Seriell-Adapter/dp/B01LVYR1AB/ref=sr_1_1_sspa?keywords=rs232+usb+adapter&qid=1642055767&sprefix=rs%2Caps%2C85&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExTE1XUUZCMjlUNkVSJmVuY3J5cHRlZElkPUEwNjM5MTMwMVFBMDBRVzBGMkVDNSZlbmNyeXB0ZWRBZElkPUEwMDQxNTc5MzZQQVNBTU5RV0pSMSZ3aWRnZXROYW1lPXNwX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=

I read the whole Thread and try the most of the Things.

Did i miss something, or did i need another Adapter, can someone give me an Link to an Adapter that work ?

Any Help are welcome.

gelu-constantin, 2022-01-19:
I have the same problem with reading an easun SV II
I can't find exactly what settings to make for the 4 values
qpiri
qpiws
qmod
qpigs
--- now I have
102
36
5
110
but nothing comes out of the interface. Somebody knou settings for EASUN SV II 5KW?

pi@raspberrypi:~ $ /opt/ha-inverter-mqtt-agent/sources/inverter-cli/inverter_poller -d -1
Wed Jan 19 17:21:36 2022 INVERTER: Debug set
Wed Jan 19 17:21:36 2022 INVERTER: Current CRC: 49 C1
Wed Jan 19 17:21:37 2022 INVERTER: QMOD reply size (5 bytes)
Wed Jan 19 17:21:37 2022 INVERTER: QMOD: incorrect start/stop bytes.  Buffer: ^0
Wed Jan 19 17:21:37 2022 INVERTER: Current CRC: B7 A9
Wed Jan 19 17:21:37 2022 INVERTER: QPIGS reply size (110 bytes)
Wed Jan 19 17:21:37 2022 INVERTER: QPIGS: incorrect start/stop bytes.  Buffer:
Wed Jan 19 17:21:37 2022 INVERTER: Current CRC: F8 54
Wed Jan 19 17:21:38 2022 INVERTER: QPIRI reply size (102 bytes)
Wed Jan 19 17:21:38 2022 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: ^0
Wed Jan 19 17:21:38 2022 INVERTER: Current CRC: B4 DA
Wed Jan 19 17:21:39 2022 INVERTER: QPIWS reply size (36 bytes)
Wed Jan 19 17:21:39 2022 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer:
Wed Jan 19 17:21:44 2022 INVERTER: Current CRC: 49 C1
Wed Jan 19 17:21:45 2022 INVERTER: QMOD reply size (5 bytes)
Wed Jan 19 17:21:45 2022 INVERTER: QMOD: incorrect start/stop bytes.  Buffer:
Wed Jan 19 17:21:45 2022 INVERTER: Current CRC: B7 A9
Wed Jan 19 17:21:48 2022 INVERTER: QPIGS reply size (110 bytes)
Wed Jan 19 17:21:48 2022 INVERTER: QPIGS: incorrect start/stop bytes.  Buffer:
Wed Jan 19 17:21:48 2022 INVERTER: Current CRC: F8 54
Wed Jan 19 17:21:50 2022 INVERTER: QPIRI reply size (102 bytes)
Wed Jan 19 17:21:50 2022 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer:
Wed Jan 19 17:21:50 2022 INVERTER: Current CRC: B4 DA
Wed Jan 19 17:21:51 2022 INVERTER: QPIWS reply size (36 bytes)
Wed Jan 19 17:21:51 2022 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer:
Wed Jan 19 17:21:56 2022 INVERTER: Current CRC: 49 C1
Wed Jan 19 17:21:57 2022 INVERTER: QMOD reply size (5 bytes)
Wed Jan 19 17:21:57 2022 INVERTER: QMOD: incorrect start/stop bytes.  Buffer:
Wed Jan 19 17:21:57 2022 INVERTER: Current CRC: B7 A9
Wed Jan 19 17:21:58 2022 INVERTER: QPIGS reply size (110 bytes)
Wed Jan 19 17:21:58 2022 INVERTER: QPIGS: incorrect start/stop bytes.  Buffer:
Wed Jan 19 17:21:58 2022 INVERTER: Current CRC: F8 54
^C
pi@raspberrypi:~ $ cat /dev/usb/hiddev0
▒^▒D▒1▒0▒6▒2▒3▒5▒1▒,▒5▒0▒0▒,▒2▒3▒5▒1▒,▒5▒0▒0▒,▒0▒6▒1▒1▒,▒0▒4▒8▒0▒,▒0▒1▒2▒,▒5▒3▒2▒,▒0▒0▒0▒,▒0▒0▒0▒,▒0▒0▒1▒,▒0▒0▒0▒,▒0▒5▒0▒,▒0▒2▒7▒,▒0▒0▒0▒,▒0▒0▒0▒,▒0▒0▒0▒0▒,▒0▒0▒0▒0▒,▒1▒2▒0▒3▒,▒0▒0▒0▒0▒,▒0▒▒▒^▒D▒1▒0▒6▒2▒3▒4▒9▒,▒4▒9▒9▒,▒2▒3▒4▒9▒,▒4▒9▒9▒,▒0▒6▒1▒0▒,▒0▒4▒7▒6▒,▒0▒1▒2▒,▒5▒3▒2▒,▒0▒0▒0▒,▒0▒0▒0▒,▒0▒0▒1▒,▒0▒0▒0▒,▒0▒5▒0▒,▒0▒2▒7▒,▒0▒0▒0▒,▒0▒0▒0▒,▒0▒0▒0▒0▒,▒0▒0▒0▒0▒,▒1▒2▒0▒5▒,▒0▒0▒0▒0▒,▒0
pi@raspberrypi:~ $ cat /dev/hidraw0
^D1062348,499,2348,499,0^C3,0498,012,532,000,000,001,000,050,027,000,000,0000,0000,1199,0000,0,2,0,1,2,2,1,0m▒
pi@raspberrypi:~ $

closed 2022-07-07 by 1onord

eloquentvogon, 2022-11-09:
@1onord  did you get this working ? I am having the same issue with a MPP inverter, thanks

1onord, 2022-11-10:
unfortunately not I had switched to the Growatt SPF series in the meantime after this Inverter talked to the BMS of my lithium Battery. I think it was the Adapter in my Case. I'm very happy with the current solution I've never looked back ;)

eloquentvogon, 2022-11-10:
@1onord thanks for the follow up! I appreciate it

eloquentvogon, 2022-11-11:
Turns out from what I can see the LVX6048WP is running PI17

robwolff3, 2023-01-24:
@eloquentvogon is PI17 not compatible with docker-voltronic-homeassistant?

eloquentvogon, 2023-01-27:
I was not able to get to work, I went with another project that was super basic just reading inverter info and sending to a mqtt server. From there to homeassitant and grafana, not sure on protocol of dropping other projects in comments.

robwolff3, 2023-01-27:
I found the one you're talking about :) Got that one working for me too. And even wrote a bit for them on running their project in docker.

eloquentvogon, 2023-01-27:
@robwolff3 awesome! good to hear

carmelo42, 2023-11-23:
> @robwolff3 I was not able to get to work, I went with another project that was super basic just reading inverter info and sending to a mqtt server. From there to homeassitant and grafana, not sure on protocol of dropping other projects in comments.

Which project please ?

robwolff3, 2023-12-04:
> Which project please ?

mpp-solar