ned-kelly/docker-voltronic-homeassistant #66: Dual MPPT not showing PV2 at all

GastonJ, 2022-02-26:
Good afternoon, I have a 
Axpert Plus Duo 3000-24 from Voltacon

Using Watchpower I can see and monitor my 2nd array (PV2) and can see it is fully working from both Watchpower and the LCD panel.

However using this I am unable to view/monitory PV2

root@bf6f3302017b:/opt# /opt/inverter-cli/bin/inverter_poller -d -1
Sat Feb 26 12:27:51 2022 INVERTER: Debug set
Sat Feb 26 12:27:52 2022 INVERTER: Current CRC: 49 C1
Sat Feb 26 12:27:52 2022 INVERTER: QMOD reply size (5 bytes)
Sat Feb 26 12:27:52 2022 INVERTER: QMOD: 5 bytes read: (B
Sat Feb 26 12:27:52 2022 INVERTER: QMOD query finished
Sat Feb 26 12:27:52 2022 INVERTER: Current CRC: B7 A9
Sat Feb 26 12:27:52 2022 INVERTER: QPIGS reply size (110 bytes)
Sat Feb 26 12:27:52 2022 INVERTER: QPIGS: 110 bytes read: (246.5 50.0 230.8 50.0 0115 0049 003 386 27.15 001 100 0016 0000 000.0 00.00 00000 10010000 00 00 00000 110
Sat Feb 26 12:27:52 2022 INVERTER: QPIGS query finished
Sat Feb 26 12:27:52 2022 INVERTER: Current CRC: F8 54
Sat Feb 26 12:27:53 2022 INVERTER: QPIRI reply size (102 bytes)
Sat Feb 26 12:27:53 2022 INVERTER: QPIRI: 102 bytes read: (230.0 13.0 230.0 50.0 13.0 3000 2400 24.0 23.5 21.0 29.2 27.0 1 20 060 0 1 2 0 11 0 0 00.0 0 1 080
Sat Feb 26 12:27:53 2022 INVERTER: QPIRI query finished
Sat Feb 26 12:27:53 2022 INVERTER: Current CRC: B4 DA
Sat Feb 26 12:27:53 2022 INVERTER: QPIWS reply size (38 bytes)
Sat Feb 26 12:27:53 2022 INVERTER: QPIWS: 38 bytes read: (0000000000000000010000000000000000
Sat Feb 26 12:27:53 2022 INVERTER: QPIWS query finished
INVERTER: ampfactor from config is 1.00
INVERTER: wattfactor from config is 1.01
{
  "Inverter_mode":4,
  "AC_grid_voltage":246.5,
  "AC_grid_frequency":50.0,
  "AC_out_voltage":230.8,
  "AC_out_frequency":50.0,
  "PV_in_voltage":0.0,
  "PV_in_current":0.0,
  "PV_in_watts":0.0,
  "PV_in_watthour":0.0000,
  "SCC_voltage":0.0000,
  "Load_pct":3,
  "Load_watt":49,
  "Load_watthour":1.6333,
  "Load_va":115,
  "Bus_voltage":386,
  "Heatsink_temperature":16,
  "Battery_capacity":100,
  "Battery_voltage":27.15,
  "Battery_charge_current":1,
  "Battery_discharge_current":0,
  "Load_status_on":1,
  "SCC_charge_on":0,
  "AC_charge_on":0,
  "Battery_recharge_voltage":23.5,
  "Battery_under_voltage":21.0,
  "Battery_bulk_voltage":29.2,
  "Battery_float_voltage":27.0,
  "Max_grid_charge_current":20,
  "Max_charge_current":60,
  "Out_source_priority":1,
  "Charger_source_priority":2,
  "Battery_redischarge_voltage":2512665466838125624666875924316160.0,
  "Warnings":"0000000000000000010000000000000000"
}
Sat Feb 26 12:27:53 2022 INVERTER: All queries complete, exiting loop.

In the above I disconnected PV1 - since it will pick up and monitor PV1 without issue and it certainly doesn't pick up PV2.

For ref my invertor.conf below

device=/dev/hidraw0
run_interval=120
amperage_factor=1.0
watt_factor=1.01
qpiri=102
qpiws=38
qmod=5
qpigs=110

MQQT.json below

    "server": "xxx.xxx.xxx.xxx",
    "port": "1883",
    "topic": "solar",
    "devicename": "voltronic",
    "username": "user",
    "password": "xxxxxxxxx",
        "clientid": "voltronic_bd8041d0cdf131a6ba4e5b3360b8bc5a",
    "influx": {
        "enabled": "false",
        "host": "http://xxx.xxx.xxx.xxx:8086",
        "username": "user",
        "password": "xxxxxxxxxx",
        "device": "voltronic",
        "prefix": "solar",
        "database": "solar",
        "namingMap": {
            "Inverter_mode": "output_mode",
            "AC_grid_voltage":  "grid_voltage",
            "AC_grid_frequency":  "grid_frecuency",
            "AC_out_voltage":  "ac_output_voltage",
            "AC_out_frequency":  "ac_output_frecuency",
            "PV_in_voltage":  "pv_input_voltage_1",
            "PV_in_current":  "pv_input_current_for_battery",
            "PV_in_watts":  "pv_input_watts",
            "PV_in_watthour": "pv_input_watthour",
            "SCC_voltage": "battery_voltage_from_scc",
            "Load_pct": "output_load_percent",
            "Load_watt": "output_load_watt",
            "Load_watthour": "outpu_load_watthour",
            "Load_va": "ac_output_power_va",
            "Bus_voltage": "bus_voltage",
            "Heatsink_temperature": "inverter_heat_sink_temperature",
            "Battery_capacity": "battery_capacity",
            "Battery_voltage": "battery_voltage",
            "Battery_charge_current": "battery_charging_current",
            "Battery_discharge_current": "battery_discharge_current",
            "Load_status_on": "load_status_on",
            "SCC_charge_on": "scc_charge_on",
            "AC_charge_on": "ac_charge_on",
            "Battery_recharge_voltage":  "battery_recharge_voltage",
            "Battery_under_voltage":  "battery_under_voltage",
            "Battery_bulk_voltage":  "battery_bulk_voltage",
            "Battery_float_voltage":  "battery_float_voltage",
            "Max_grid_charge_current": "max_grid_charge_current",
            "Max_charge_current": "max_charge_current",
            "Out_source_priority": "output_source_priority",
            "Charger_source_priority": "charger_source_priority",
            "Battery_redischarge_voltage":  "battery_re_discharge_voltage"
        }
    }
}

While I know most people will have a single MPPT controller - has anyone any ideas on how to resolve this please. Any help appreciated please.

Regards,

Pete

GastonJ, 2022-11-10:
Hi Darzl,

No, got nowhere with it. I do get both feeds on Watchpower just not anywhere else, I've tried a few things so far all to no avail. That won't stop me looking and trying to work out how Watchpower sees it and yet this doesn't Perhaps it's just me. I will in all likelihood decompile the java into it's components and find out why, it's just a matter of time - which is never on my side.

Darzl, 2022-11-10:
I switched to https://github.com/catalinbordan/docker-voltronic-homeassistant and now I get my PV power as PV_Charging_Power1 in watts, probably only need to add another line PV_Charging_Power2 or something similar

GastonJ, 2022-11-11:
Thanks I'll take a look.

Regards

saschalenz, 2023-09-13:
This is a little bit old but I added support for a second mppt and created a PR to the repo above. https://github.com/catalinbordan/docker-voltronic-homeassistant/pull/2. The changes are in my fork in case anyone is interested https://github.com/saschalenz/docker-voltronic-homeassistant

GastonJ, 2024-04-05:
Thanks, I'll take a look, bit late I know.
Regards