ned-kelly/docker-voltronic-homeassistant #73: Always Receiving 0 byte responses

Jaysun11, 2022-03-16:
Hey guys, would anyone be able to assist me with this?

I am using an  Axpert VM III Inverter, I am currently plugged in to a Raspberry Pi 3b+ using a RJ45->USB using a RS232 Adapter.

When Running ./inverter_poller -d -1

I get the following output.. Never any response what so ever.

Thu Mar 17 09:36:19 2022 INVERTER: Debug set
Thu Mar 17 09:36:19 2022 INVERTER: Current CRC: 49 C1
Thu Mar 17 09:36:22 2022 INVERTER: QMOD read timeout
Thu Mar 17 09:36:23 2022 INVERTER: QMOD reply too short (0 bytes)
Thu Mar 17 09:36:23 2022 INVERTER: Current CRC: B7 A9
Thu Mar 17 09:36:26 2022 INVERTER: QPIGS read timeout
Thu Mar 17 09:36:27 2022 INVERTER: QPIGS reply too short (0 bytes)
Thu Mar 17 09:36:27 2022 INVERTER: Current CRC: F8 54
Thu Mar 17 09:36:30 2022 INVERTER: QPIRI read timeout
Thu Mar 17 09:36:31 2022 INVERTER: QPIRI reply too short (0 bytes)
Thu Mar 17 09:36:31 2022 INVERTER: Current CRC: B4 DA
Thu Mar 17 09:36:34 2022 INVERTER: QPIWS read timeout
Thu Mar 17 09:36:35 2022 INVERTER: QPIWS reply too short (0 bytes)
Thu Mar 17 09:36:40 2022 INVERTER: Current CRC: 49 C1

Any help would be hugely appreciated.

korsteeni, 2022-03-17:
I am using an Easun VM II Inverter, rs232 USB
and output

:~/inverter-cli# ./inverter_poller -d -1
Thu Mar 17 15:19:21 2022 INVERTER: Debug set
Thu Mar 17 15:19:21 2022 INVERTER: Current CRC: 49 C1
Thu Mar 17 15:19:24 2022 INVERTER: QMOD read timeout
Thu Mar 17 15:19:24 2022 INVERTER: QMOD reply too short (0 bytes)
Thu Mar 17 15:19:24 2022 INVERTER: Current CRC: B7 A9
Thu Mar 17 15:19:27 2022 INVERTER: QPIGS read timeout
Thu Mar 17 15:19:27 2022 INVERTER: QPIGS reply too short (0 bytes)
Thu Mar 17 15:19:27 2022 INVERTER: Current CRC: F8 54
Thu Mar 17 15:19:30 2022 INVERTER: QPIRI read timeout
Thu Mar 17 15:19:30 2022 INVERTER: QPIRI reply too short (0 bytes)
Thu Mar 17 15:19:30 2022 INVERTER: Current CRC: B4 DA
Thu Mar 17 15:19:33 2022 INVERTER: QPIWS read timeout
Thu Mar 17 15:19:33 2022 INVERTER: QPIWS reply too short (0 bytes)
Thu Mar 17 15:19:38 2022 INVERTER: Current CRC: 49 C1
Thu Mar 17 15:19:41 2022 INVERTER: QMOD read timeout
Thu Mar 17 15:19:41 2022 INVERTER: QMOD reply too short (0 bytes)
Thu Mar 17 15:19:41 2022 INVERTER: Current CRC: B7 A9
Thu Mar 17 15:19:44 2022 INVERTER: QPIGS read timeout
Thu Mar 17 15:19:44 2022 INVERTER: QPIGS reply too short (0 bytes)
Thu Mar 17 15:19:44 2022 INVERTER: Current CRC: F8 54
Thu Mar 17 15:19:47 2022 INVERTER: QPIRI read timeout
Thu Mar 17 15:19:47 2022 INVERTER: QPIRI reply too short (0 bytes)
Thu Mar 17 15:19:47 2022 INVERTER: Current CRC: B4 DA

korsteeni, 2022-03-17:
some clue on how to read data
EASUN POWER 5000w Hybrid Solar power inverter 48v 230vac Grid tied + off grid 80A MPPT
Axpert and PowMr works fine

tested 
qpiri=98.....110
qpiws=36...54
qmod=5
qpigs=110

and /dev/ttyUSB0 , /dev/ttyS0

if hint, i can try

Jaysun11, 2022-03-18:
I tried switching to the USB port, and changed my device to hidraw0, it worked straight away! Perhaps just a faulty serial port?

closed 2022-03-18 by Jaysun11