ned-kelly/docker-voltronic-homeassistant #74: Voltronik AXPERT MKS IV 5.6KW no data via USB2serial adapter

byte4geek, 2022-04-01:
I have the issue that non data was sent to mqtt, all entity was configured correctly on HA.
This what i see on inverter poller:

pi@raspberrypi:~/ha-inverter-mqtt-agent/manual $ docker exec -it voltronic-mqtt bash -c '/opt/inverter-cli/bin/inverter_poller -d -1'
Fri Apr 1 14:58:35 2022 INVERTER: Debug set
Fri Apr 1 14:58:35 2022 INVERTER: Current CRC: 49 C1
Fri Apr 1 14:58:36 2022 INVERTER: QMOD reply size (5 bytes)
configuration options for the actual inverter polling process...es. Buffer: (NAKs▒

The device to read from...
Use: /dev/ttyS0 if you have a serial device,
/dev/ttyUSB0 if a USB<>Serial,
/dev/hidraw0 if you're connecting via the USB port on the inverter.
device=/dev/ttyUSB0
#device=/dev/serial1
#device=/dev/hidraw0

How many times per hour is the program going to run...
This is used to calculate the PV & Load Watt Hours between runs...
If unsure, leave as default - it will run every minute...
(120 = every 30 seconds)...
run_interval=120

This allows you to modify the amperage in case the inverter is giving an incorrect
reading compared to measurement tools. Normally this will remain '1'
amperage_factor=1.0

This allows you to modify the wattage in case the inverter is giving an incorrect
reading compared to measurement tools. Normally this will remain '1'
watt_factor=1.01

The following settings allow you to modify runtime buffers.
N.B. These values may not be applicable to al
Fri Apr 1 14:58:36 2022 INVERTER: Current CRC: B7 A9
Fri Apr 1 14:58:39 2022 INVERTER: QPIGS read timeout
Fri Apr 1 14:58:40 2022 INVERTER: QPIGS reply too short (7 bytes)
Fri Apr 1 14:58:40 2022 INVERTER: Current CRC: F8 54
Fri Apr 1 14:58:43 2022 INVERTER: QPIRI read timeout
Fri Apr 1 14:58:44 2022 INVERTER: QPIRI reply too short (7 bytes)
Fri Apr 1 14:58:44 2022 INVERTER: Current CRC: B4 DA
Fri Apr 1 14:58:47 2022 INVERTER: QPIWS read timeout
Fri Apr 1 14:58:48 2022 INVERTER: QPIWS reply too short (7 bytes)
Fri Apr 1 14:58:53 2022 INVERTER: Current CRC: 49 C1
Fri Apr 1 14:58:53 2022 INVERTER: QMOD reply size (5 bytes)
configuration options for the actual inverter polling process...tes. Buffer: (NAKs▒

Fri Apr  1 15:05:28 2022 INVERTER: Current CRC: B7 A9
Fri Apr  1 15:05:31 2022 INVERTER: QPIGS read timeout
Fri Apr  1 15:05:32 2022 INVERTER: QPIGS reply too short (7 bytes)
Fri Apr  1 15:05:32 2022 INVERTER: Current CRC: F8 54
Fri Apr  1 15:05:35 2022 INVERTER: QPIRI read timeout
Fri Apr  1 15:05:36 2022 INVERTER: QPIRI reply too short (7 bytes)
Fri Apr  1 15:05:36 2022 INVERTER: Current CRC: B4 DA
Fri Apr  1 15:05:39 2022 INVERTER: QPIWS read timeout
Fri Apr  1 15:05:39 2022 INVERTER: QPIWS reply too short (16 bytes)
Fri Apr  1 15:05:45 2022 INVERTER: Current CRC: 49 C1
Fri Apr  1 15:05:45 2022 INVERTER: QMOD reply size (5 bytes)
Fri Apr  1 15:05:45 2022 INVERTER: QMOD: incorrect start/stop bytes.  Buffer: (L
s
(NAKss
000000010000000010000 48.0 46.0 42.0 56.4 54.0 0 030 060 0 2 2 9 00 0 0 54.0 0 1 0 00 00 00000

byte4geek, 2022-04-01:
Solved with:
# This allows you to modify the buffersize for the qpiri command
qpiri=103

byte4geek, 2022-06-10:
Sometime the /dev/hidraw0 disappear, i must to power off the inverter and power on again, someone have this issue?

ned-kelly, 2022-08-25:
Hi @byte4geek - Sorry I've changed inverters so I'm not as active here now - I also used to use a USB2serial adapter and found that I had to reboot the raspberry pi each night, there was no other fix, and I ended up dumping the usb2serial and just using the USB Port on the inverter directly

closed 2022-08-25 by ned-kelly

byte4geek, 2022-08-25:
I don't know why, but the docker can't set the serial port baud rate.

 i wrote a pyhton script that do this:

```
import serial

serialport = serial.Serial("/dev/ttyUSB0", 2400, timeout=0.5)
        
serialport.write(b"QMOD 4\r")
response = serialport.readlines(None)
print(response)

serialport.write(b"QPIGS 5\r")
response = serialport.readlines(None)
print(response)
```
        
and insterted into the crontab at boot:

```
@reboot python set_ttyUSB0.py > /home/pi/out.txt
```
and it works fine.