ned-kelly/docker-voltronic-homeassistant #79: Cannot c

Fabio72, 2022-05-04:


Fabio72, 2022-05-04:
Hi,
I'm using a Pi0 with an AXPERT VMIII

Everything is working fine, except changing max grid charge current.
If I send commands like MUCHGC030 nothing happens.
Rarely I can get an ACK, but most of the times nothing.
Read value sometimes jumps to strange values like 33 or 63 after a try, but eventually go back to the previous value.
I can change values with WatchPower app and bluetooth.

$ docker exec -it voltronic-mqtt bash -c '/opt/inverter-cli/bin/inverter_poller -d -1'
Wed May  4 16:46:54 2022 INVERTER: Debug set
Wed May  4 16:46:55 2022 INVERTER: Current CRC: 49 C1
Wed May  4 16:46:55 2022 INVERTER: QMOD reply size (5 bytes)
Wed May  4 16:46:55 2022 INVERTER: QMOD: 5 bytes read: (L
Wed May  4 16:46:55 2022 INVERTER: QMOD query finished
Wed May  4 16:46:55 2022 INVERTER: Current CRC: B7 A9
Wed May  4 16:46:55 2022 INVERTER: QPIGS reply size (110 bytes)
Wed May  4 16:46:55 2022 INVERTER: QPIGS: 110 bytes read: (237.6 49.9 237.6 49.9 0665 0478 013 346 49.80 002 087 0021 00.0 000.0 00.00 00000 00010100 00 00 00000 010
Wed May  4 16:46:55 2022 INVERTER: QPIGS query finished
Wed May  4 16:46:55 2022 INVERTER: Current CRC: F8 54
Wed May  4 16:46:56 2022 INVERTER: QPIRI reply size (98 bytes)
Wed May  4 16:46:56 2022 INVERTER: QPIRI: 98 bytes read: (230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 46.0 45.0 53.5 53.5 3 10 040 0 0 3 1 01 0 0 54.0 0 1
Wed May  4 16:46:56 2022 INVERTER: QPIRI query finished
Wed May  4 16:46:56 2022 INVERTER: Current CRC: B4 DA
Wed May  4 16:46:56 2022 INVERTER: QPIWS reply size (36 bytes)
Wed May  4 16:46:56 2022 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: (100000000000000000000000000000000000 48.0 46.0 45.0 53.5 53.5 3 10 040 0 0 3 1 01 0 0 54.0 0 1
INVERTER: ampfactor from config is 1.00
INVERTER: wattfactor from config is 1.01
{
  "Inverter_mode":3,
  "AC_grid_voltage":237.6,
  "AC_grid_frequency":49.9,
  "AC_out_voltage":237.6,
  "AC_out_frequency":49.9,
  "PV_in_voltage":0.0,
  "PV_in_current":0.0,
  "PV_in_watts":0.0,
  "PV_in_watthour":0.0000,
  "SCC_voltage":0.0000,
  "Load_pct":13,
  "Load_watt":478,
  "Load_watthour":15.9333,
  "Load_va":665,
  "Bus_voltage":346,
  "Heatsink_temperature":21,
  "Battery_capacity":87,
  "Battery_voltage":49.80,
  "Battery_charge_current":2,
  "Battery_discharge_current":0,
  "Load_status_on":1,
  "SCC_charge_on":0,
  "AC_charge_on":0,
  "Battery_recharge_voltage":46.0,
  "Battery_under_voltage":45.0,
  "Battery_bulk_voltage":53.5,
  "Battery_float_voltage":53.5,
  "Max_grid_charge_current":10,
  "Max_charge_current":40,
  "Out_source_priority":0,
  "Charger_source_priority":3,
  "Battery_redischarge_voltage":-0.0,
  "Warnings":""
}
Wed May  4 16:46:57 2022 INVERTER: All queries complete, exiting loop.

Fabio72, 2022-05-05:
Using   mpp-solar command I can change the parameter without problems

$ mpp-solar -p /dev/hidraw0 -c MUCHGC002
Command: MUCHGC002 - Set Utility Max Charging Current
------------------------------------------------------------
Parameter                       Value           Unit
muchgc                          ACK

SuperMaximus1984, 2022-11-16:
> 

I'm having the same problem. MQTT doesn't set any MUCHGC parameter, only polls the data from inverter.
Where did you find "mpp-solar" command? Cannot find it in the image/docker container.

SuperMaximus1984, 2022-11-16:
I decided to opt for shell_command via mpp-solar.
Using this method I can control the desired parameter

NextWork123, 2024-02-21:
with the fork of catalinbordan/docker-voltronic-homeassistant seem to work fine i used the mqtt :)

[image]