ned-kelly/docker-voltronic-homeassistant #85: Connecting 2 inverters using 1 Raspberry Pi

SuperMaximus1984, 2022-08-21:
How should I configure an `inverter.conf` if I need to use this container for two inverters?
Each unit connected to a separate port of Raspberry Pi.
Now my config is:

`# Basic configuration options for the actual inverter polling process...

# The device to read from...
# Use: /dev/ttyS0 if you have a serial device,
#      /dev/ttyUSB0 if a USB<>Serial,
#      /dev/hidraw0 if you're connecting via the USB port on the inverter.

device=/dev/hidraw0

# How many times per hour is the program going to run...
# This is used to calculate the PV & Load Watt Hours between runs...
# If unsure, leave as default - it will run every minute...

# (120 = every 30 seconds)...
run_interval=120

# This allows you to modify the amperage in case the inverter is giving an incorrect
# reading compared to measurement tools.  Normally this will remain '1'
amperage_factor=1.0

# This allows you to modify the wattage in case the inverter is giving an incorrect
# reading compared to measurement tools.  Normally this will remain '1'
watt_factor=1.01

# The following settings allow you to modify runtime buffers.
# N.B. These values may not be applicable to all inverter types, as such you will 
# need to run docker exec -it voltronic-mqtt bash -c '/opt/inverter-cli/bin/inverter_poller -d -1'
# or simply inverter_poller -d -1 to check for warnings or errors
# mentioned in https://github.com/ned-kelly/docker-voltronic-homeassistant/issues/5

# This allows you to modify the buffersize for the qpiri command
qpiri=108

# This allows you to modify the buffersize for the qpiws command
qpiws=45

# This allows you to modify the buffersize for the qmod command
qmod=5

# This allows you to modify the buffersize for the qpigs command
qpigs=110
`

ned-kelly, 2022-08-25:
Hi,
Please:
* Create two docker containers (one for each inverter)
* In the device/config for the first inverter specify `device=/dev/hidraw0` - and in the second container you may specify `device= /dev/hidraw1` ...
* Don't forget you may need to update your docker-compose file to map in both hidraw0 and hidraw1 (or whatever your USB devices are called)

closed 2022-08-25 by ned-kelly

SuperMaximus1984, 2022-08-25:
@ned-kelly 
Should I create a separate clone-container only for `voltronic-mqtt` or also a `watchtower`?
Thanks!