ned-kelly/docker-voltronic-homeassistant #88: HomeAssistant Auto Discovery Not Working

eloquentvogon, 2022-09-08:
I have an unraid set up using docker for HA and for MQTT, I have installed this docker on a raspberry pi and have it connected to my inverter.

I am getting the below messages in Home Assistant.

`2022-09-08 12:06:18.483 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Integration mpp_solar is not supported`

I am seeing this on my MQTT server

1662653178: New connection from 192.168.0.239 on port 1883.
1662653178: Client mpp_solar already connected, closing old connection.
1662653178: Client mpp_solar disconnected.
1662653178: New client connected from 192.168.0.239 as mpp_solar (c1, k60, u'').
1662653178: Sending CONNACK to mpp_solar (0, 0)
1662653178: Received PUBLISH from mpp_solar (d0, q0, r0, m0, 'homeassistant/mpp_solar/sensor/mpp_solar/config', ... (114 bytes))
1662653178: Sending PUBLISH to 276oORPCclJrLsdcVGqAks (d0, q0, r0, m0, 'homeassistant/mpp_solar/sensor/mpp_solar/config', ... (114 bytes))
1662653178: Received DISCONNECT from mpp_solar
1662653178: Client mpp_solar disconnected.
1662653179: New connection from 192.168.0.239 on port 1883.
1662653179: New client connected from 192.168.0.239 as mpp_solar (c1, k60, u'').
1662653179: Sending CONNACK to mpp_solar (0, 0)
1662653179: Received SUBSCRIBE from mpp_solar
1662653179:     homeassistant/mpp_solar/sensor/mpp_solar (QoS 1)
1662653179: mpp_solar 1 homeassistant/mpp_solar/sensor/mpp_solar
1662653179: Sending SUBACK to mpp_solar

I am not sure why it will not auto discover as my other mqtt device (tasmota) is being discovered and functioning.

Thanks

eloquentvogon, 2022-09-08:
not sure if it is related to this 

Thu Sep  8 17:41:27 2022 INVERTER: Current CRC: 49 C1
Thu Sep  8 17:41:27 2022 INVERTER: QMOD reply size (5 bytes)
▒hu Sep  8 17:41:27 2022 INVERTER: QMOD: incorrect start/stop bytes.  Buffer: ^0
Thu Sep  8 17:41:27 2022 INVERTER: Current CRC: B7 A9
Thu Sep  8 17:41:30 2022 INVERTER: QPIGS read timeout
Thu Sep  8 17:41:30 2022 INVERTER: QPIGS reply too short (8 bytes)
Thu Sep  8 17:41:30 2022 INVERTER: Current CRC: F8 54
Thu Sep  8 17:41:33 2022 INVERTER: QPIRI read timeout
Thu Sep  8 17:41:33 2022 INVERTER: QPIRI reply too short (8 bytes)
Thu Sep  8 17:41:33 2022 INVERTER: Current CRC: B4 DA
Thu Sep  8 17:41:36 2022 INVERTER: QPIWS read timeout
Thu Sep  8 17:41:36 2022 INVERTER: QPIWS reply too short (8 bytes)

eloquentvogon, 2022-09-09:
seems this is all related to the improper QMOD ,QPIGS,QPIRI,QPIWS  values and I have come up against a wall with this, any help is appreciated

saqib2050i, 2022-09-15:
its not because of qpigs etc
mosquitto_pub tool used to publish mqtt message by default disconnect after publishing message creating mess in mqtt broker logs

saqib2050i, 2022-09-15:
discovery function is bit tricky,, i modified entire publish message in mqtt_init.sh file to get auto discovery working

eloquentvogon, 2022-10-26:
update, tried it again and there is no discovery working, I am not using the standard home assistant mqtt integration I have a mosquitto docker running on its own 

I re-added an tasmota mqtt switch I had to the server and it was discovered by HA no problem, as you can see the inverter is not

1666807656: New client connected from 192.168.0.209:49236 as voltronic_bd8041d0cdf131a6ba4e5b3360b8bc5a (p1, c1, k60).
1666807659: Client 5T5X1QzF8dFHaAeWcXHSW9 has exceeded timeout, disconnecting.
1666807770: New connection from 192.168.0.28:58333 on port 1883.
1666807770: New client connected from 192.168.0.28:58333 as office_switch (p2, c1, k30, u'DVES_USER').

I hope this helps thx