ned-kelly/docker-voltronic-homeassistant #90: inverter_poller timeout too short?

degreal, 2022-09-18:
Easun ISolar SMV III 5kw inverter.
Windows 10 host -> Virtualbox -> Ubuntu -> Docker
Ultimately recognized as HID device /dev/hidraw1, where I don't have control over baud rate in windows host.

> Sat Aug 31 14:27:56 2019 INVERTER: QPIRI read timeout
> Sat Aug 31 14:27:56 2019 INVERTER: QPIRI reply too short (88 bytes)

All responses fail randomly between 60-90 symbol. I have feeling that timeout is too short, so I tried to increase timeout somehow but I really can't cpp.
```
        n = read(fd, (void*)buf+i, replysize-i);
        if (n < 0) {
            if (time(NULL) - started > 2) {
                lprintf("INVERTER: %s read timeout", cmd);
                break;
            } else {
                usleep(10);
                continue;
            }
        }
```
Does this >2 mean longer than 2 seconds? I increased it to 4 and now query succeeds every time.
Is this the solution or i'm missing the actual problem here?

Darzl, 2022-10-18:
Have you tried just plugging in a with Windows and using watchpower to make sure you have connection?

degreal, 2022-10-18:
Yes, watchpower in windows seems to work fine.

My proposed code fix resolves the cut issue, but USB connection dies in few dies and I must restart the VM.

Darzl, 2022-10-25:
I have switched over to https://github.com/catalinbordan/docker-voltronic-homeassistant no need to change any settings or try and work out  QPIRI or anything else, ned kelly wouldn't switch redischarge voltage for me but this does and so far it's stable other than a few times my automation mqtt send didn't work. I don't use the information from the configuration.yaml because its messes my automations section so nothing shows

michas79de, 2022-12-30:
I think also had issues to find the proper QPIRI, QPIWS, QMOD and QPIGS values.
After getting timeouts, too short replies and incorrect start/stop bytes I tried MultiSIBControl for Windows and could retrieve these answers:

> QPIRI
(230.0 15.2 230.0 50.0 15.2 3500 3500 24.0 22.5 20.0 28.2 27.0 2 30 060 0 2 1 1 01 0 0 27.0 0 1
QPIWS
(10000000000000000100000000000000
QMOD
(B
QPIGS
(236.6 50.0 230.0 50.0 0184 0151 005 392 25.70 000 100 0039 00.0 000.0 00.00 00008 00010000 00 00 00000 010

I've tried to count all chars (including spaces, dots and the "(" ) and use this number in the inverter.conf, but with no success.
Finding somewhere working examples with a value of my way of counting plus three, I tried adding 3 to mine - no.
(BTW MultiSIBControl works fine, so the RS-232<->USB cable should do.)

So I switched to the above mentioned fork of @catalinbordan 
But now I get also timeouts. At least the other errors seem to be gone:

> Fri Dec 30 20:33:24 2022 INVERTER: Debug set
Fri Dec 30 20:33:24 2022 DEBUG:  Current CRC: 49 C1
Fri Dec 30 20:33:24 2022 DEBUG:  Send buffer hex bytes:  ( 51 4d 4f 44 49 c1 0d )
Fri Dec 30 20:33:24 2022 DEBUG:  7 bytes written, 7 bytes sent, 0 bytes remaining
Fri Dec 30 20:33:24 2022 DEBUG:  5 bytes read, 5 total bytes:  00 00 00 28 42 e7 c9 0a
Fri Dec 30 20:33:30 2022 DEBUG:  QMOD read timeout
Fri Dec 30 20:33:30 2022 DEBUG:  Found reply <cr> at byte: -1987891007

> Fri Dec 30 20:48:22 2022 INVERTER: Debug set
Fri Dec 30 20:48:22 2022 DEBUG:  Current CRC: 49 C1
Fri Dec 30 20:48:22 2022 DEBUG:  Send buffer hex bytes:  ( 51 4d 4f 44 49 c1 0d )
Fri Dec 30 20:48:22 2022 DEBUG:  7 bytes written, 7 bytes sent, 0 bytes remaining
Fri Dec 30 20:48:22 2022 DEBUG:  5 bytes read, 5 total bytes:  00 00 00 28 42 e7 c9 0a
Fri Dec 30 20:48:28 2022 DEBUG:  QMOD read timeout
Fri Dec 30 20:48:28 2022 DEBUG:  Found reply <cr> at byte: 1766092993

Can anybody help me, please?
How do I have to count the answers to get the correct Q* values?
Or what to do to get the fork running fine?

Thanks in advance!

Darzl, 2022-12-30:
Did you wipe the original installation, I made the mistake originally you have to basically uninstall the ned kelly version

michas79de, 2022-12-30:
Because there stands in the readme "I have taking in consideration that you allready have installed and followed the intructions of original repository of ned-kelly." I just did the 11 steps for the fork.
But I can try to delete this and start new.
Because I'm new to linux/raspian, how to wipe/uninstall this?

Darzl, 2022-12-30:
Follow these steps exactly

When you will use this fork you need to do the following commands on your device:

cd /opt/ha-inverter-mqtt-agent
sudo docker-compose down
cd ..
sudo rm -rf ha-inverter-mqtt-agent
sudo git clone https://github.com/catalinbordan/docker-voltronic-homeassistant.git /opt/ha-inverter-mqtt-agent
cd /opt/ha-inverter-mqtt-agent
sudo nano config/inverter.conf (only edit this file if you don't have usb cable to inverter)
sudo nano config/mqtt.json (change it with your variables)
sudo docker-compose build
sudo docker-compose up -d
sudo docker exec -it voltronic-mqtt bash -c '/opt/inverter-cli/bin/inverter_poller -d -1' (this is to test to see is everything ok regarding connection between inverter and docker)

Darzl, 2022-12-30:
I didn't notice the first 4 lines and just installed as if I was starting from fresh

michas79de, 2022-12-31:
I copy pasted line for line including the first four without any errors until the last one results in errors.
I edited inverter.conf to ttyusb0 for my USB<->RS-232 cable.
And I changed some lines like the IP, vendor, serial in the mqtt.json.

Testing with the last line leads to the answer with  QMOD read timeout like cited above.
And in Home Assistant or in MQTT Explorer I can see the Sensors/entities but only as unknown without any values.

catalinbordan, 2022-12-31:
> I copy pasted line for line including the first four without any errors until the last one results in errors. I edited inverter.conf to ttyusb0 for my USB<->RS-232 cable. And I changed some lines [...]

Hi,
On forum it is a command to set the baundrate of serial connection. Try it before your start your docker.
I also had a issue with qpiri time out. It seems Inverter do not want to reply. I found that if I change a setting in Inverter app, problem it is solved(for example I changed to the appliance to ups mode)

michas79de, 2023-01-07:
I wasn't able to get it running.
After trying https://github.com/jblance/mpp-solar successfully, I'm using this now.
Thanks anyway!