ned-kelly/docker-voltronic-homeassistant #92: Incorrect start/stop size

dominikamann, 2022-10-03:
Hi all,

I try since hours to get the connection running. Do you have any ideas?

Thank you!!

[image]

```

# The device to read from...
# Use: /dev/ttyS0 if you have a serial device,
#      /dev/ttyUSB0 if a USB<>Serial,
#      /dev/hidraw0 if you're connecting via the USB port on the inverter.
 
device=/dev/hidraw0
 
# How many times per hour is the program going to run...
# This is used to calculate the PV & Load Watt Hours between runs...
# If unsure, leave as default - it will run every minute...
 
# (120 = every 30 seconds)...
run_interval=120
 
# This allows you to modify the amperage in case the inverter is giving an incorrect
# reading compared to measurement tools.  Normally this will remain '1'
amperage_factor=1.0
 
# This allos you to modify the wattage in case the inverter is giving an incorrect
# reading compared to measurement tools.  Normally this will remain '1'
watt_factor=1.01
 
qpiri=97
 
Mon Oct  3 16:28:34 2022 INVERTER: Current CRC: F8 54
Mon Oct  3 16:28:38 2022 INVERTER: QPIRI read timeout
Mon Oct  3 16:28:38 2022 INVERTER: QPIRI reply too short (24 bytes)
Mon Oct  3 16:28:38 2022 INVERTER: Current CRC: B4 DA
Mon Oct  3 16:28:38 2022 INVERTER: QPIWS reply size (0 bytes)
^0n Oct  3 16:28:38 2022 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: QPIWS��
Mon Oct  3 16:28:43 2022 INVERTER: Current CRC: 49 C1
Mon Oct  3 16:28:43 2022 INVERTER: QMOD reply size (0 bytes)
^0n Oct  3 16:28:43 2022 INVERTER: QMOD: incorrect start/stop bytes.  Buffer: QMODI�
Mon Oct  3 16:28:43 2022 INVERTER: Current CRC: B7 A9
Mon Oct  3 16:28:43 2022 INVERTER: QPIGS reply size (0 bytes)
^0n Oct  3 16:28:43 2022 INVERTER: QPIGS: incorrect start/stop bytes.  Buffer: QPIGS��
Mon Oct  3 16:28:43 2022 INVERTER: Current CRC: F8 54
Mon Oct  3 16:28:47 2022 INVERTER: QPIRI read timeout
Mon Oct  3 16:28:47 2022 INVERTER: QPIRI reply too short (24 bytes)
Mon Oct  3 16:28:47 2022 INVERTER: Current CRC: B4 DA
Mon Oct  3 16:28:47 2022 INVERTER: QPIWS reply size (0 bytes)
^0n Oct  3 16:28:47 2022 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: QPIWS��
Mon Oct  3 16:28:52 2022 INVERTER: Current CRC: 49 C1
Mon Oct  3 16:28:52 2022 INVERTER: QMOD reply size (0 bytes)
^0n Oct  3 16:28:52 2022 INVERTER: QMOD: incorrect start/stop bytes.  Buffer: QMODI�
Mon Oct  3 16:28:52 2022 INVERTER: Current CRC: B7 A9
Mon Oct  3 16:28:52 2022 INVERTER: QPIGS reply size (0 bytes)
^0n Oct  3 16:28:52 2022 INVERTER: QPIGS: incorrect start/stop bytes.  Buffer: QPIGS��
Mon Oct  3 16:28:52 2022 INVERTER: Current CRC: F8 54
Mon Oct  3 16:28:56 2022 INVERTER: QPIRI read timeout
Mon Oct  3 16:28:56 2022 INVERTER: QPIRI reply too short (40 bytes)
Mon Oct  3 16:28:56 2022 INVERTER: Current CRC: B4 DA
Mon Oct  3 16:28:56 2022 INVERTER: QPIWS reply size (0 bytes)
^0n Oct  3 16:28:56 2022 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: QPIWS��
Mon Oct  3 16:29:01 2022 INVERTER: Current CRC: 49 C1
Mon Oct  3 16:29:01 2022 INVERTER: QMOD reply size (0 bytes)
^0n Oct  3 16:29:01 2022 INVERTER: QMOD: incorrect start/stop bytes.  Buffer: QMODI�
Mon Oct  3 16:29:01 2022 INVERTER: Current CRC: B7 A9
Mon Oct  3 16:29:01 2022 INVERTER: QPIGS reply size (0 bytes)
^0n Oct  3 16:29:01 2022 INVERTER: QPIGS: incorrect start/stop bytes.  Buffer: QPIGS��
Mon Oct  3 16:29:01 2022 INVERTER: Current CRC: F8 54
Mon Oct  3 16:29:04 2022 INVERTER: QPIRI read timeout

```

Darzl, 2022-10-17:
I have the Axpert Max 8kw and these are my settings

109
40
5
112

I found that if you put the numbers high like 130 it will say that it's too high and expected 112 for instance so drop it to that and slowly lower, I was lucky with the 112 but the 109 took a few guesses because I did everything except and 40 was a pure guess :) 

D

Darzl, 2022-10-17:
Oh and I would really like your inverter :) the 900 pv open circuit would be good for me

gysief, 2022-11-06:
I has have an Axpert inverter. have the same problem I am updating the config/inverter.conf but the setting is not pulling through. Am I in the wrong location. I wanna try your setting. I have even try deleting the docker images. 

Kind Regards
G

54.00 000 100 0043 0000 000.0 00.00 00000 00010101 00 00 00000 110��
Sun Nov  6 20:12:46 2022 INVERTER: Current CRC: F8 54
Sun Nov  6 20:12:49 2022 INVERTER: QPIRI read timeout
Sun Nov  6 20:12:49 2022 INVERTER: QPIRI reply too short (104 bytes)
Sun Nov  6 20:12:49 2022 INVERTER: Current CRC: B4 DA
Sun Nov  6 20:12:49 2022 INVERTER: QPIWS reply size (40 bytes)
Sun Nov  6 20:12:49 2022 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: (00000000000000000000000000000000��
Sun Nov  6 20:12:54 2022 INVERTER: Current CRC: B7 A9
Sun Nov  6 20:12:54 2022 INVERTER: QPIGS reply size (112 bytes)
Sun Nov  6 20:12:54 2022 INVERTER: QPIGS: incorrect start/stop bytes.  Buffer: (225.9 49.8 225.9 49.8 0768 0753 015 438 54.00 000 100 0043 0000 000.0 00.00 00000 00010101 00 00 00000 110*
Sun Nov  6 20:12:54 2022 INVERTER: Current CRC: F8 54
Sun Nov  6 20:12:57 2022 INVERTER: QPIRI read timeout
Sun Nov  6 20:12:57 2022 INVERTER: QPIRI reply too short (104 bytes)
Sun Nov  6 20:12:57 2022 INVERTER: Current CRC: B4 DA
Sun Nov  6 20:12:57 2022 INVERTER: QPIWS reply size (40 bytes)
Sun Nov  6 20:12:57 2022 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: (00000000000000000000000000000000��

storrestw, 2023-01-10:
I have an Axpert 3kW inverter (24V) and these settings worked for me:

```
qpiri=99
qpiws=40
```

dominikamann, 2023-06-27:
Thanks for your responses. I was not able to find a solution.
But it works with MPP-SOLAR and Protocol PI17 (https://github.com/jblance/mpp-solar/blob/master/mppsolar/protocols/pi17.py). May it is possible to add it to docker-voltronic-homeassistant?

dominikamann, 2023-07-02:
Workarround until it may works with this integration.

-> https://github.com/dominikamann/InfiniSolar-Voltronic-Home-Assistant-Integration