ned-kelly/docker-voltronic-homeassistant #93: USB read working, but USB2Serial doesn't

avfoto, 2022-10-31:
I have a Voltronic Axpert VMIII 24V, when I read data trough hidraw0, my raspberry obtains data without any problem, but when I change the adquisition method to ttyUSB0 without changing the qpiri, qpiws, qmod and qpigs values, can't read data.
Need to change qpiri, qpiws, qmod and qpigs values or must read with ttyUSB0 without any changes?
Do I need to change anything else in the settings?

Boney35y, 2022-11-17:
Do you have /dev/ttyUSB0 under /dev?

I have a generic one of these and use hidraw1 (my ups is on hidraw0) why can you not use hidraw0?

avfoto, 2022-11-17:
How can I know if ttyUSB0 is under /dev/ ??

Yes, I can use hidraw0, now I'm reading data through hidraw0, but I want to read trough ttySUB0 too.

Boney35y, 2022-11-17:
if you cd to /dev on your pi and do an ls it will list them all.
Eg:
root@nas:/dev# ls
adsp             dvd        i2c-12      loop2           nvidia-uvm-tools  sdc1      stderr  tty2   tty36  tty52      ttyS1   ttyS26   userio