ned-kelly/docker-voltronic-homeassistant #96: 'run_interval' does not change polling interval

SuperMaximus1984, 2022-12-08:
Changing 'run_interval=120' to any other figure does not impact polling interval. It remains at default figure 30 seconds.
I need to make it 10 seconds. How?
Should I recreate containers, simple restart doesn't apply the new parameter.

SuperMaximus1984, 2023-01-14:
@ned-kelly , please help to sort this out. Thank you in advance!

dev-null2019, 2023-07-06:
Hello!
You can edit the last line in the file /opt/inverter-mqtt/entrypoint.sh inside the docker container.

`# execute exactly every 30 seconds...
watch -n 30 /opt/inverter-mqtt/mqtt-push.sh > /dev/null 2>&1`

the lowest that works is 2 seconds

`# execute exactly every 2 seconds...
watch -n 2 /opt/inverter-mqtt/mqtt-push.sh > /dev/null 2>&1`

With this I get updates around every 10s in Home ASsistant - have to find out where the delay comes from. It is probably my mosquitto mqtt ...