ned-kelly/docker-voltronic-homeassistant #103: Integration fail

Xcpro1, 2023-05-28:
Hi, trying to integrate with link...😭

<Integration ned-kelly/docker-voltronic-homeassistant> Repository structure for master is not compliant

DJBenson, 2023-06-13:
This isn't an integration in the normal sense, it's an external project which hooks into Home Assistant using MQTT - you can't just import the GitHub project into Home Assistant - follow the instructions in the readme file.

leondavibe, 2023-06-29:
you may have to change values in the inverter dot conf file

i could  not get mine to work changed the qpiri value and now it connects is it perfect i don't know, new to all this, just chiming in cause i could not get the connection to work, i also did the setup on a virtual machine (laptop)and the device ended up being hidraw1 not sure assume something else grabbed 0, only after probing usb a bit i figured this out