ned-kelly/docker-voltronic-homeassistant #117: mqtt output only presents .../config data but no values

sabom2d, 2024-02-13:
Hi

I have the docker isntallation running and MQTT is sending data, but all only config data, no values.
inverter_poller shows real data.
What do I need to configure to get real values for each topic via MQTT?

regards

emaklav, 2024-02-22:
same .. did you solve it?

sabom2d, 2024-02-22:
No, unfortunately not until now.

mprenditore, 2024-05-26:
same issue here

mprenditore, 2024-05-26:
I've solved it for my Inverter: `Axpert VM III TWIN 6000 watt 48V Off-Grid`

## TL;DR;
I've changed the following values in the `config/inverter.conf`
```ini
qpiri=99
qpiws=40
qmod=5
qpigs=120
```

## How to get the correct buffer values
Find out the proper values is a bit tricky but with roughly 5 minutes can be achieved.
It can all be done with a single terminal, but to speed up the process and reduce the amount of work needed to make changes and check them, I do recommend to have 2 terminals open, this can be achieved in various ways:
1. connect with 2 terminals via SSH to the same instance from a third instance (the easiest solution)
2. using TMUX session

### Procedure
```
cd /opt/ha-inverter-mqtt-agent
docker compose up -d
```
 1. We then need to connect to the running container to check if the values are correct
```
docker compose exec -ti voltronic-mqtt bash
```
 2. Inside the container, run:
```
stdbuf -oL /opt/inverter-cli/bin/inverter_poller -d -1 2>&1 | grep incorrect
```
 3. Check if it output anything. if it does, it will look something like this:
```
Sun May 26 13:38:12 2024 INVERTER: QPIGS: incorrect start/stop bytes.  Buffer: (226.2 49.9 230.0 49.9 0161 0096 002 363 51.60 003 054 0041 07.9 064.0 00.00 00000 00010110 00 00 00410 011 0 00 0000
```
 4. That means we need to change the `qpigs` (that is in the log line just after `INVERTER`) because the line is truncated.
 5. Exit the container with the command `exit`
 6. Open the file `config/inverter/conf` with your favourite editor and increase the parameter found on point **4** by `1`
```
qpigs=118
```
become
```
qpigs=119
```
 7. Save the file and start back from point **1** till you have no more lines with `incorrect start/stop bytes`
 8. To confirm all is fine, run:
```
/opt/inverter-cli/bin/inverter_poller -d -1
```
and you should have an output like
```
Sun May 26 13:48:07 2024 INVERTER: Debug set
Sun May 26 13:48:07 2024 INVERTER: Current CRC: 49 C1
Sun May 26 13:48:07 2024 INVERTER: QMOD reply size (5 bytes)
Sun May 26 13:48:07 2024 INVERTER: QMOD: 5 bytes read: (B
Sun May 26 13:48:07 2024 INVERTER: QMOD query finished
Sun May 26 13:48:07 2024 INVERTER: Current CRC: B7 A9
Sun May 26 13:48:07 2024 INVERTER: QPIGS reply size (120 bytes)
Sun May 26 13:48:07 2024 INVERTER: QPIGS: 120 bytes read: (231.3 50.0 229.9 50.0 0137 0090 002 371 52.80 001 066 0042 06.0 063.0 00.00 00000 00010110 00 00 00299 011 0 00 0000
Sun May 26 13:48:07 2024 INVERTER: QPIGS query finished
Sun May 26 13:48:07 2024 INVERTER: Current CRC: F8 54
Sun May 26 13:48:08 2024 INVERTER: QPIRI reply size (99 bytes)
Sun May 26 13:48:08 2024 INVERTER: QPIRI: 99 bytes read: (230.0 26.0 230.0 50.0 26.0 6000 6000 48.0 47.0 46.0 58.0 54.0 2 030 030 0 2 3 1 01 0 0 56.0 0 1
Sun May 26 13:48:08 2024 INVERTER: QPIRI query finished
Sun May 26 13:48:08 2024 INVERTER: Current CRC: B4 DA
Sun May 26 13:48:08 2024 INVERTER: QPIWS reply size (40 bytes)
Sun May 26 13:48:08 2024 INVERTER: QPIWS: 40 bytes read: (000000000000000000000000000000000000
Sun May 26 13:48:08 2024 INVERTER: QPIWS query finished
INVERTER: ampfactor from config is 1.00
INVERTER: wattfactor from config is 1.01
{
  "Inverter_mode":4,
  "AC_grid_voltage":231.3,
  "AC_grid_frequency":50.0,
  "AC_out_voltage":229.9,
  "AC_out_frequency":50.0,
  "PV_in_voltage":63.0,
  "PV_in_current":6.0,
  "PV_in_watts":0.0,
  "PV_in_watthour":0.0000,
  "SCC_voltage":0.0000,
  "Load_pct":2,
  "Load_watt":90,
  "Load_watthour":3.0000,
  "Load_va":137,
  "Bus_voltage":371,
  "Heatsink_temperature":42,
  "Battery_capacity":66,
  "Battery_voltage":52.80,
  "Battery_charge_current":1,
  "Battery_discharge_current":0,
  "Load_status_on":1,
  "SCC_charge_on":1,
  "AC_charge_on":0,
  "Battery_recharge_voltage":47.0,
  "Battery_under_voltage":46.0,
  "Battery_bulk_voltage":58.0,
  "Battery_float_voltage":54.0,
  "Max_grid_charge_current":30,
  "Max_charge_current":30,
  "Out_source_priority":2,
  "Charger_source_priority":3,
  "Battery_redischarge_voltage":0.0,
  "Warnings":"000000000000000000000000000000000000"
}
Sun May 26 13:48:09 2024 INVERTER: All queries complete, exiting loop.
```

Hope that helps anyone that had my same issues.