ned-kelly/docker-voltronic-homeassistant #119: Troubleshoot Port Settings

UV-PWRD, 2024-07-11:
Back after a few years with some new issues. I have successfully been using this for a while but have moved Rpi4 onto a more powerful x86 ProxMox environment and can only get this partially working, I’m convinced it’s the USB to Serial port settings. I would like to retire the RPi4 fully. 

New Environment is a Debian 12 VM, latest docker/portainer, using the same PL2303 adaptor as last time, same inverter, there is nothing else on this VM.

Used some config settings in inverter.conf and mqtt.json as last time and I can publish commands to the inverter all OK so MQTT is seemingly working? but the inverter poller command never finishes, just keeps looping with;

Thu Jul 11 04:14:27 2024 INVERTER: Current CRC: B4 DA
Thu Jul 11 04:14:30 2024 INVERTER: QPIWS read timeout
Thu Jul 11 04:14:30 2024 INVERTER: QPIWS reply too short (22 bytes)
Thu Jul 11 04:14:35 2024 INVERTER: Current CRC: B7 A9
Thu Jul 11 04:14:38 2024 INVERTER: QPIGS read timeout
Thu Jul 11 04:14:38 2024 INVERTER: QPIGS reply too short (9 bytes)
Thu Jul 11 04:14:38 2024 INVERTER: Current CRC: F8 54
Thu Jul 11 04:14:41 2024 INVERTER: QPIRI read timeout
Thu Jul 11 04:14:41 2024 INVERTER: QPIRI reply too short (13 bytes)
Thu Jul 11 04:14:41 2024 INVERTER: Current CRC: B4 DA
Thu Jul 11 04:14:44 2024 INVERTER: QPIWS read timeout
Thu Jul 11 04:14:44 2024 INVERTER: QPIWS reply too short (6 bytes)
Thu Jul 11 04:14:49 2024 INVERTER: Current CRC: B7 A9
Thu Jul 11 04:14:52 2024 INVERTER: QPIGS read timeout
Thu Jul 11 04:14:52 2024 INVERTER: QPIGS reply too short (76 bytes)
Thu Jul 11 04:14:52 2024 INVERTER: Current CRC: F8 54
Thu Jul 11 04:14:55 2024 INVERTER: QPIRI read timeout
Thu Jul 11 04:14:55 2024 INVERTER: QPIRI reply too short (77 bytes)
Thu Jul 11 04:14:55 2024 INVERTER: Current CRC: B4 DA
Thu Jul 11 04:14:58 2024 INVERTER: QPIWS read timeout
Thu Jul 11 04:14:58 2024 INVERTER: QPIWS reply too short (27 bytes)
Thu Jul 11 04:15:03 2024 INVERTER: Current CRC: B7 A9
Thu Jul 11 04:15:06 2024 INVERTER: QPIGS read timeout
Thu Jul 11 04:15:06 2024 INVERTER: QPIGS reply too short (81 bytes)
Thu Jul 11 04:15:06 2024 INVERTER: Current CRC: F8 54
Thu Jul 11 04:15:09 2024 INVERTER: QPIRI read timeout
Thu Jul 11 04:15:09 2024 INVERTER: QPIRI reply too short (70 bytes)
Thu Jul 11 04:15:09 2024 INVERTER: Current CRC: B4 DA
Thu Jul 11 04:15:12 2024 INVERTER: QPIWS read timeout
Thu Jul 11 04:15:12 2024 INVERTER: QPIWS reply too short (30 bytes)

Over and over.

If I run screen I can send and receive a QPIGS response, and the inverter.conf parameters are the same last time (last time was Debian 11 OS directly on RPi4).

I note that port settings default to 2400, if I change to 9600 with stty -F /dev/ttyUSB0 9600 the port will reset itself back to 2400 after a few seconds (checking with stty).

I feel like this is a port settings issue, hardware/software control, raw, something but do not know how to troubleshoot this.

closed 2024-07-11 by UV-PWRD

emuchogu, 2025-05-20:
Hi @UV-PWRD ,
How did you resolve the above issue?

UV-PWRD, 2025-05-20:
> Hi [@UV-PWRD](https://github.com/UV-PWRD) , How did you resolve the above issue?

Good morning, 

I think all came down to needing different settings for the buffer in each QPIRI/QPIGS etc in the end, but it's been so long I honestly can't remember fully. 

I can say I never regained the push functionality though, as in, I used to be able to submit an MQTT command to change the inverter ac charging amount, I can't do that anymore and I just gave up sinking time into it :(

emuchogu, 2025-05-21:
Thanks @UV-PWRD