ned-kelly/docker-voltronic-homeassistant #123: possible to convert to HA addon ?

spacecabbie, 2024-11-25:
Home Assistant is basically a docker server it is strange to have to create a seperate server with a docker engine on it to just get this to work.

jszbp, 2024-12-10:
I second that!
Here are some tutorials, how to do it. Unfortunatelly, I'm not so familiar with docker and most important: in linux architecture and its dependencies... But it seems to me that making a HA add-on for someone building doceer images as a daily routine should be quite easy.
https://developers.home-assistant.io/docs/add-ons/tutorial/
https://github.com/alexbelgium/hassio-addons/wiki/Building-an-addon

Currently I run HA in docker container instead of as Hassio - just because I have to run these voltronic docker image on the same machine as HA. So I run them both in separate docker images. But I wish I could just install HA as standalone OS and run this "voltronic-image" inside it.

Thats exactly what I hope for last few years

garethmo, 2025-06-25:
Thanks for this @GitGab19  ill give it a try this weekend and advise, i've just ported over from supervised ( being deprecated ) so was going to look into this